pellepl/spiffs #5: 30 out of 49 tests fail

benpicco, 2015-02-03:
I've checked out the repository and ran `make` which got me `linux_spiffs_test`.
Running this invokes the spiff test suite that prints:
```
Test report, 49 tests
19 succeeded
30 failed
  lu_check1
  page_cons1
  page_cons2
  page_cons3
  page_cons_final
  index_cons1
  index_cons2
  index_cons3
  index_cons4
  open_by_dirent
  write_big_file_chunks_page
  write_big_files_chunks_page
  write_big_file_chunks_index
  write_big_files_chunks_index
  write_big_file_chunks_huge
  write_big_files_chunks_huge
  truncate_big_file
  bad_index_1
  bad_index_2
  lseek_simple_modification
  lseek_modification_append
  lseek_modification_append_multi
  write_small_file_chunks_1
  write_small_files_chunks_1
  write_big_file_chunks_1
  write_big_files_chunks_1
  long_run_config_many_small_one_long
  long_run_config_many_medium
  long_run_config_many_small
  long_run
0 stopped
```

Does that mean the file system is still full of errors?

pellepl, 2015-02-03:
Odd. Did you make any modifications at all?

pellepl, 2015-02-03:
Ah! Did you run linux_spiffs_test directly? Try running "make test" instead.

2015-02-03 13:58 GMT+01:00 Peter Andersson <pelleplutt1976@gmail.com>:

benpicco, 2015-02-03:
Ah!
running `make test` gets me `ALL TESTS OK` :+1:

closed 2015-02-03 by benpicco

pellepl, 2015-02-03:
Great!