pellepl/spiffs #6: Lots of static analysis bugs

mkschreder, 2015-02-09:
spiffs_nucleus.c	spiffs_object_modify	line 1218 new_objix_pix is never initialized
spiffs_nucleus.c	spiffs_obj_lu_find_free_obj_id	line 1710	possible division by zero

state.compaction = (state.max_obj_id-state.min_obj_id) / ((SPIFFS_CFG_LOG_PAGE_SZ(fs) / sizeof(u8_t)));

Also lots of warnings for never read result value. Better to not store result at all otherwise written code has no effect. 

spiffs_nucleus.c	spiffs_object_modify	line 1135 value of res is never read
spiffs_hydrogen.c	SPIFFS_write	line 324 value of res never read
spiffs_hydrogen.c	SPIFFS_check line 	700 res never read
...

pellepl, 2015-02-10:
Could you please inform me of what compiler and with what flags you encounter this?

mkschreder, 2015-02-10:
Clang scan-build with arm-none-eabi-gcc with -Wall

pellepl, 2015-02-10:
Ok thanks. Seems plain linux gcc -Wall does not report as much as an arm cross-compiled gcc. Weird. I'll  look into it and fix it.

mkschreder, 2015-02-10:
For maximum effect: 

gcc:
-Wall -Wno-format-y2k -W -Wstrict-prototypes -Wmissing-prototypes \
-Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch \
-Wshadow -Wcast-align -Wchar-subscripts -Winline \
-Wnested-externs -Wredundant-decls

g++:
-Wall -Wno-format-y2k -W \
-Wpointer-arith -Wreturn-type -Wcast-qual -Wwrite-strings -Wswitch \
-Wcast-align -Wchar-subscripts -Wredundant-decls

pellepl, 2015-02-10:
Fixed. Phew. Found one bug amongst all warnings. Thanks!

closed 2015-02-10 by pellepl