pellepl/spiffs #15: Add a defrag method

ficeto, 2015-05-20:
Hello, I implemented your amazing piece of work into the latest Arduino framework for ESP8266 WiFi MCUs, and I run into some problems :)
Here is the current status of my partition
```c_cpp
==== SPIFFS Info ====
FS Mount: 1
FS Size: 7.00MB
FS Bytes: total: 27.45KB, used: 27.45KB
FS Blocks: total: 896, free: 335, size: 8.00KB
FS Pages: allocated: 112, deleted: 17276, size: 256B
FS File: /edit.htm, type: file, size: 19.43KB
FS File: /favicon.ico, type: file, size: 1.12KB
FS File: /graphs.js, type: file, size: 3.76KB
FS File: /index.htm, type: file, size: 1.51KB
```
Now onto the issues:
When written often onto, the FS generates lots of blocks with a few pages of data, or all deleted pages, which is part of the design, but when you get to the edges and have a larger file coming in (let's say I upload 900KB jpeg image), the time it takes to do a garbage cleanup for space triggers a watchdog reset. I do not know how familiar you are with the MCU, but the SPIFFS lives on the same flash as the firmware,so when it's accessed, all interrupts are stopped and CPU is locked.
When you have a larger partition like this 7MB one in my example, listing the root folder also takes way too long (i guess it goes to search for files in all blocks/pages), now I have tried using larger blocks but at some point the FS is unusable due to the mentioned watchdog resets (do not know the reason for those here).
I read that you run such a routine when a file is being written, and you check if there are enough free blocks to write the file, and if there are not enough or the free blocks are less than 3, garbage cleaner runs to free that space. I tried implementing that but had no luck on properly using the FS afterwards (could read fine but not write to it).
So I think that if you add such routine to the API, the user/firmware can decide when to run it and keep data aligned.

pellepl, 2015-05-20:
Thanks for the kind words! I'll look into it and get back to you. In the middle of a GL pipeline ;) In the meantime, what pagesizes and blocksizes have you tried? What errors did you get when writing to a borked fs?

ficeto, 2015-05-20:
I got "deleted" errors mostly. Regardless of if the file has been there before or not.
Have tried block sizes from 4K (hardware block size) to 64K (largest hardware block size).
It sometimes worked and sometimes it did not, resulting in WDT reset. Now our WDT is nasty and quick. There hardly ever is an exception address to point me in a direction.

pellepl, 2015-05-20:
Hmm.. well.. a quick fix would be to pet your dog in the hal erase
function. That would do it. Spiffs will still hog your CPU, but it
shouldn't crash. If that is not enough, then I blame the specs of the SPI
flash - and the watchdog timer should be increased.

ficeto, 2015-05-20:
hal erase is the method called also when SPIFFS-format is called, is it not? If so I have to say that there is no issue with erase, because format even though it takes forever on 8MB chip, runs and finishes fine.
Writing files is also fine(if you have enough free blocks), but reading and listing ... totally different story.

pellepl, 2015-05-20:
Ok, thats a bit weird. Formatting will not trigger the watchdog, but
reading will... Just to get it straight, a reading then takes longer than
formatting your chip?

pellepl, 2015-05-30:
Ping?

pellepl, 2015-06-05:
No life in this issue more. As a final remark: a garbagecollection is basically a defrag. If really necessary, you can always call spiffs_gc_check  yourself to trigger a gc.

closed 2015-06-05 by pellepl