pellepl/spiffs #17: 0 free blocks after mounting on a fresh fs

benpicco, 2015-05-28:
Hi,
I'm not sure if that's the right place to ask support questions, but then again I just followed your steps in `docs/INTEGRATION` so it might as well be a bug :wink: 

I pretty much [hooked up](https://gist.github.com/benpicco/c42cc73b893af41876be) spiffs as suggested, I didn't change the default config except for enabling `SPIFFS_USE_MAGIC` to see if the fs gets properly created at all - it appears to be.
But no matter what I do, `SPIFFS_mount` will always report 0 free blocks - and subsequently attempting to create a file obviously fails. The Flash is freshly formatted with `SPIFFS_format`, there shouldn't be anything on it.

What might I be missing?

```
2015-05-28 17:52:20,611 - INFO # INIT FLASH MEMORY (JEDEC format)
2015-05-28 17:52:20,616 - INFO #           MANUFACTURER:              0x44 (ADESTO)
2015-05-28 17:52:20,619 - INFO #           DEVICE-ID-1:               0x1f
2015-05-28 17:52:20,623 - INFO #           DEVICE-ID-2:               0x01
2015-05-28 17:52:20,625 - INFO #           -------------
2015-05-28 17:52:20,629 - INFO #           DENSITY:                   4MBit
2015-05-28 17:52:20,632 - INFO # -- init done, starting main --
2015-05-28 17:52:20,633 - INFO # erasing flash…
2015-05-28 17:52:23,363 - INFO # mounting…
2015-05-28 17:52:23,363 - INFO # mount res: -1
2015-05-28 17:52:23,364 - INFO # SPIFFS_format
2015-05-28 17:52:23,367 - INFO # erase 00000000:00001000
2015-05-28 17:52:23,404 - INFO # erase 00001000:00001000
2015-05-28 17:52:23,443 - INFO # erase 00002000:00001000
2015-05-28 17:52:23,479 - INFO # erase 00003000:00001000
2015-05-28 17:52:23,516 - INFO # erase 00004000:00001000
2015-05-28 17:52:23,553 - INFO # erase 00005000:00001000
2015-05-28 17:52:23,590 - INFO # erase 00006000:00001000
2015-05-28 17:52:23,628 - INFO # erase 00007000:00001000
2015-05-28 17:52:23,666 - INFO # erase 00008000:00001000
2015-05-28 17:52:23,704 - INFO # erase 00009000:00001000
2015-05-28 17:52:23,742 - INFO # erase 0000a000:00001000
2015-05-28 17:52:23,779 - INFO # erase 0000b000:00001000
2015-05-28 17:52:23,817 - INFO # erase 0000c000:00001000
2015-05-28 17:52:23,854 - INFO # erase 0000d000:00001000
2015-05-28 17:52:23,891 - INFO # erase 0000e000:00001000
2015-05-28 17:52:23,930 - INFO # erase 0000f000:00001000
2015-05-28 17:52:23,967 - INFO # erase 00010000:00001000
2015-05-28 17:52:24,004 - INFO # erase 00011000:00001000
2015-05-28 17:52:24,040 - INFO # erase 00012000:00001000
2015-05-28 17:52:24,079 - INFO # erase 00013000:00001000
2015-05-28 17:52:24,117 - INFO # erase 00014000:00001000
2015-05-28 17:52:24,154 - INFO # erase 00015000:00001000
2015-05-28 17:52:24,192 - INFO # erase 00016000:00001000
2015-05-28 17:52:24,229 - INFO # erase 00017000:00001000
2015-05-28 17:52:24,266 - INFO # erase 00018000:00001000
2015-05-28 17:52:24,304 - INFO # erase 00019000:00001000
2015-05-28 17:52:24,341 - INFO # erase 0001a000:00001000
2015-05-28 17:52:24,377 - INFO # erase 0001b000:00001000
2015-05-28 17:52:24,414 - INFO # erase 0001c000:00001000
2015-05-28 17:52:24,451 - INFO # erase 0001d000:00001000
2015-05-28 17:52:24,487 - INFO # erase 0001e000:00001000
2015-05-28 17:52:24,525 - INFO # erase 0001f000:00001000
2015-05-28 17:52:24,561 - INFO # erase 00020000:00001000
2015-05-28 17:52:24,597 - INFO # erase 00021000:00001000
2015-05-28 17:52:24,634 - INFO # erase 00022000:00001000
2015-05-28 17:52:24,671 - INFO # erase 00023000:00001000
2015-05-28 17:52:24,709 - INFO # erase 00024000:00001000
2015-05-28 17:52:24,746 - INFO # erase 00025000:00001000
2015-05-28 17:52:24,783 - INFO # erase 00026000:00001000
2015-05-28 17:52:24,820 - INFO # erase 00027000:00001000
2015-05-28 17:52:24,857 - INFO # erase 00028000:00001000
2015-05-28 17:52:24,894 - INFO # erase 00029000:00001000
2015-05-28 17:52:24,931 - INFO # erase 0002a000:00001000
2015-05-28 17:52:24,968 - INFO # erase 0002b000:00001000
2015-05-28 17:52:25,005 - INFO # erase 0002c000:00001000
2015-05-28 17:52:25,042 - INFO # erase 0002d000:00001000
2015-05-28 17:52:25,080 - INFO # erase 0002e000:00001000
2015-05-28 17:52:25,117 - INFO # erase 0002f000:00001000
2015-05-28 17:52:25,154 - INFO # erase 00030000:00001000
2015-05-28 17:52:25,191 - INFO # erase 00031000:00001000
[... 72 lines cut ...]
2015-05-28 17:52:27,917 - INFO # erase 0007a000:00001000
2015-05-28 17:52:27,954 - INFO # erase 0007b000:00001000
2015-05-28 17:52:27,992 - INFO # erase 0007c000:00001000
2015-05-28 17:52:28,029 - INFO # erase 0007d000:00001000
2015-05-28 17:52:28,066 - INFO # erase 0007e000:00001000
2015-05-28 17:52:28,194 - INFO # page index byte len:         128
2015-05-28 17:52:28,197 - INFO # object lookup pages:         1
2015-05-28 17:52:28,200 - INFO # page pages per block:        32
2015-05-28 17:52:28,203 - INFO # page header length:          5
2015-05-28 17:52:28,205 - INFO # object header index entries: 42
2015-05-28 17:52:28,211 - INFO # object index entries:        61
2015-05-28 17:52:28,211 - INFO # available file descriptors:  3
2015-05-28 17:52:28,214 - INFO # free blocks:                 0
2015-05-28 17:52:28,215 - INFO # mount res: 0
2015-05-28 17:52:28,216 - INFO # testing…
2015-05-28 17:52:28,303 - INFO # free_obj_id: COMP min:0001 max:0800 compact:15
2015-05-28 17:52:28,390 - INFO # free_obj_id: COMP select index:0 min_count:0 min:0001 max:0800 compact:15
2015-05-28 17:52:28,470 - INFO # erase 00000000:00001000
2015-05-28 17:52:28,588 - INFO # erase 00000000:00001000
2015-05-28 17:52:28,705 - INFO # erase 00000000:00001000
2015-05-28 17:52:28,817 - INFO # errno -10010
2015-05-28 17:52:28,901 - INFO # errno -10010
2015-05-28 17:52:28,902 - INFO # -->  <--

```

pellepl, 2015-05-28:
Weird. Does it work without SPIFFS_USE_MAGIC?

benpicco, 2015-05-28:
No, the results are the same

```
2015-05-28 18:44:43,918 - INFO # -- init done, starting main --
2015-05-28 18:44:43,919 - INFO # erasing flash…
2015-05-28 18:44:46,649 - INFO # mounting…
2015-05-28 18:44:46,738 - INFO # page index byte len:         128
2015-05-28 18:44:46,740 - INFO # object lookup pages:         1
2015-05-28 18:44:46,743 - INFO # page pages per block:        32
2015-05-28 18:44:46,746 - INFO # page header length:          5
2015-05-28 18:44:46,749 - INFO # object header index entries: 42
2015-05-28 18:44:46,751 - INFO # object index entries:        61
2015-05-28 18:44:46,754 - INFO # available file descriptors:  3
2015-05-28 18:44:46,757 - INFO # free blocks:                 0
2015-05-28 18:44:46,758 - INFO # mount res: 0
2015-05-28 18:44:46,759 - INFO # testing…
2015-05-28 18:44:46,847 - INFO # free_obj_id: COMP min:0001 max:0800 compact:15
2015-05-28 18:44:46,933 - INFO # free_obj_id: COMP select index:0 min_count:0 min:0001 max:0800 compact:15
2015-05-28 18:44:47,014 - INFO # erase 00000000:00001000
2015-05-28 18:44:47,131 - INFO # erase 00000000:00001000
2015-05-28 18:44:47,248 - INFO # erase 00000000:00001000
2015-05-28 18:44:47,361 - INFO # errno -10010
2015-05-28 18:44:47,445 - INFO # errno -10010
2015-05-28 18:44:47,446 - INFO # -->  <--
```

pellepl, 2015-05-28:
Ok. Hmmm...
What does SPIFFS_vis output look lite?
What logical block size are you using, and what's the physical erase size? Or rather, what does the config struct look like?

benpicco, 2015-05-28:
>  What does SPIFFS_vis output look lite?

```
2015-05-28 18:56:40,897 - INFO #    0 ///////////////////////////////	era_cnt: 3
2015-05-28 18:56:40,901 - INFO #    1 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,905 - INFO #    2 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,909 - INFO #    3 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,913 - INFO #    4 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,918 - INFO #    5 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,922 - INFO #    6 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,926 - INFO #    7 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,930 - INFO #    8 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,934 - INFO #    9 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,938 - INFO #   10 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,943 - INFO #   11 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,947 - INFO #   12 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,951 - INFO #   13 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,956 - INFO #   14 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,959 - INFO #   15 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,963 - INFO #   16 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,967 - INFO #   17 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,972 - INFO #   18 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,976 - INFO #   19 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,981 - INFO #   20 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,984 - INFO #   21 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,989 - INFO #   22 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,993 - INFO #   23 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:40,997 - INFO #   24 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,001 - INFO #   25 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,005 - INFO #   26 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,010 - INFO #   27 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,014 - INFO #   28 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,018 - INFO #   29 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,022 - INFO #   30 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,026 - INFO #   31 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,030 - INFO #   32 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,034 - INFO #   33 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,038 - INFO #   34 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,043 - INFO #   35 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,047 - INFO #   36 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,051 - INFO #   37 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,055 - INFO #   38 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,059 - INFO #   39 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,064 - INFO #   40 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,068 - INFO #   41 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,072 - INFO #   42 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,076 - INFO #   43 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,080 - INFO #   44 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,084 - INFO #   45 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,089 - INFO #   46 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,093 - INFO #   47 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,097 - INFO #   48 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,101 - INFO #   49 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,105 - INFO #   50 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,110 - INFO #   51 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,114 - INFO #   52 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,118 - INFO #   53 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,122 - INFO #   54 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,126 - INFO #   55 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,131 - INFO #   56 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,135 - INFO #   57 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,139 - INFO #   58 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,143 - INFO #   59 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,147 - INFO #   60 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,151 - INFO #   61 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,156 - INFO #   62 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,160 - INFO #   63 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,164 - INFO #   64 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,168 - INFO #   65 ///////////////////////////////	era_cnt: 0
2015-05-28 18:56:41,172 - INFO #   66 ///////////////////////////////	era_cnt: 0
[... 48 lines cut ...]
[... cut ...]
2015-05-28 18:56:41,429 - INFO # free_blocks: 0
2015-05-28 18:56:41,430 - INFO # page_alloc:  0
2015-05-28 18:56:41,432 - INFO # page_delet:  3937
2015-05-28 18:56:41,434 - INFO # used:        0 of 476748
```

> what does the config struct look like?

see [here](https://gist.github.com/benpicco/c42cc73b893af41876be#file-main-c-L32)
the smallest erase block size the chip supports is 4096 bytes so I set this for `phys_erase_block` and `log_block_size` as suggested. `log_page_size` is 128 as that's what I can read with one spi command.

pellepl, 2015-05-28:
Hmm again.. It should work (tm). It seems to me the erase function is a bit fishy. According to SPIFFS_vis, block zero has an erase count of 3,
<code>
0 ///////////////////////////////   era_cnt: 3
</code>
whilst all other blocks have an erase count of 0. If you've been running a full format a couple of times all erase counts would be the same.  Worse, all blocks only contain deleted pages "/". This is a very bad state. A format should have made all blocks contain free pages "_". Are you sure your erase function is setting everything to 0xff? To me it seems to be setting everything to 0x00 instead.

benpicco, 2015-05-29:
> Are you sure your erase function is setting everything to 0xff? To me it seems to be setting everything to 0x00 instead. 

You are right!
I did previously try a different filesystem that expected free bytes to be set to 0 (unix style), so I negated every byte written/read from flash.

Without that, it works!

closed 2015-05-29 by benpicco

pellepl, 2015-05-29:
Ah great!