pellepl/spiffs #18: SPIFFS_lseek does not create gaps in files

benpicco, 2015-06-09:
`SPIFFS_lseek` behaves different than POSIX `lseek`. When creating a file and putting the offset to a random position, data is written to that position and previous bytes in the file are filled with `\0`. With SPIFFS this doesn't work, the data is written to the start of the file, the offset is ignored.

This testcase illustrates the issue

```
TEST(lseek_create_file) {
  int res;
  spiffs_file fd;
  char *fname = "sparsefile";
  char *message = "Hello World!";
  int offset = 1024;

  fd = SPIFFS_open(FS, fname, SPIFFS_TRUNC | SPIFFS_CREAT | SPIFFS_RDWR, 0);
  TEST_CHECK(fd > 0);
  int pfd = open(make_test_fname(fname), O_TRUNC | O_CREAT | O_RDWR, S_IRUSR | S_IWUSR);

  SPIFFS_lseek(FS, fd, offset, SPIFFS_SEEK_SET);
  lseek(pfd, offset, SEEK_SET);

  res = SPIFFS_write(FS, fd, message, sizeof(message));
  TEST_CHECK(res >= 0);
  write(pfd, message, sizeof(message));
  res = read_and_verify(fname);
  TEST_CHECK(res >= 0);

  SPIFFS_close(FS, fd);
  close(pfd);

  return TEST_RES_OK;
}
TEST_END(lseek_create_file)
```

On Linux, the file starts with 1024 `\0` bytes, SPIFFS however writes `Hello World!` to position 0.

```
TEST 53/59 : running test lseek_create_file
file verification mismatch @ 0, 48 H != 00 
Hello Wo
........
data mismatch
```

pellepl, 2015-06-10:
I intentionally left that out.
Writing a bunch of zeroes which are meant to be overwritten later on a spi
flash is a bad idea wear-wise.
But i am open for arguments ;)

benpicco, 2015-06-10:
I'd be happy with a bunch of `0xff` instead, too.
What I'm trying to do is transfer files over a rather choppy wireless connection. Parts of the file get lost and are requested for retransmission while later parts arrive just fine - so of course I want to store them on flash and leave a gap for the missing chunks that will arrive later.

benpicco, 2015-06-10:
This very quick and dirty patch serves my needs
```
diff --git a/src/spiffs_hydrogen.c b/src/spiffs_hydrogen.c
index 13f72b7..5b4ce37 100644
--- a/src/spiffs_hydrogen.c
+++ b/src/spiffs_hydrogen.c
@@ -470,7 +470,11 @@ s32_t SPIFFS_lseek(spiffs *fs, spiffs_file fh, s32_t offs, int whence) {
   }
 
   if (offs > (s32_t)fd->size) {
-    res = SPIFFS_ERR_END_OF_OBJECT;
+    char buf = 0xff;
+    s32_t i;
+    for (i = fd->size; i < offs; ++i) {
+      SPIFFS_write(fs, fh, &buf, 1);
+    }
   }
   SPIFFS_API_CHECK_RES_UNLOCK(fs, res);
```

I'm sure you'll come up with something more efficient ;)

pellepl, 2015-06-11:
Hmm, now when looking at it I remember actually have been thinking about this. Adding "unwritten" regions within a file was .. cumbersome to say the least. There is no good place to store this metadata, as a page have no notion of what offset in the file it belongs to. 
I'm afraid the quick and dirty patch is perhaps not so dirty (apart from not checking results, and perhaps have a bigger buffer of 0xff:s). 
Perhaps you could store the chunks as you receive them in single files and then puzzle it all together? As long as they are sequential, I guess normal appends would do.

benpicco, 2015-06-12:
IMHO it would be acceptable to just resize the file and specify that the 
newly created areas are undefined - I think that would violate the spec 
less/be less surprising than the current behavior where the file is not 
resized at all.

Some other flash file systems expect the flash driver to negate all bits 
written/read from flash, so erased bits are 0x00 instead of 0xff

pellepl, 2015-06-14:
I finally got some time to look into it a bit further. I totally agree with you that the more POSIX like spiffs is, the better. Alas, to put it simple: implementing filegaps will be a major pain in the a**e. Negating bits will not help, as spiffs tries to save pages by writing to "unfilled" pages at the end. Having the concept of filegaps will clash deeply with that concept, and will cause a major rewrite. Also, the garbage collector would need to be rewritten from scratch to be able to deal with this. One could supposedly allocate pages with some new flag, e.g. not-yet-written flag, but finding the offsets within there would be tricky.

 It could be done of course, but I cannot allocate the time for digging into it for real at least this year.

Sorry for this. I still think it is a good idea though, whatever console that might give ;)

closed 2015-06-22 by pellepl