pellepl/spiffs #19: store checksum for files

benpicco, 2015-06-11:
Hi,
Have you thought about storing a checksum for files somewhere to make it possible to detect corruption?
I'm thinking about changing my file format for firmware binaries to include a checksum, but I'm wondering whether it wouldn't be better to do at file system level (mostly because my firmware files are currently just a ROM dump and I like to keep things simple :wink:)

vladant, 2015-06-17:
If you want a checksum then store it outside of file-system, either in the file or somewhere else.

pellepl, 2015-06-22:
I agree. Adding crc as a fs feature is very specific, and would also force each write to reread the whole file each time to be robust. Also, one crc (e.g. 16bit ccitt) might not suit some other, so I feel it best leaving this to the user.

closed 2015-06-22 by pellepl