pellepl/spiffs #20: SPIFFS_remove() gets stuck in spiffs_object_truncate()/spiffs_gc_clean()

benpicco, 2015-06-12:
Hi again ;)

I managed to get the fs to a state where attempting to remove a certain file (I haven't checked other files yet because I'm afraid I might not be able to reproduce this again) gets the system stuck - which is a bit of an unfortunate outcome.

Enabling the debug output suggests the culprit is `spiffs_gc_clean()` or rather `spiffs_obj_lu_find_entry_visitor()` where it gets stuck on [`_spiffs_rd`](https://github.com/pellepl/spiffs/blob/master/src/spiffs_nucleus.c#L161).

The output is

```
2015-06-12 13:13:44,767 - INFO # open: fd 1 is obj id 8071
2015-06-12 13:13:44,776 - INFO # gc_check #0: run gc free_blocks:3 pfree:59 pallo:2686 pdele:1161 [3906] len:0 of 7257
2015-06-12 13:13:44,779 - INFO # gc_check: bix:0 del:5 use:26 score:4449
2015-06-12 13:13:44,783 - INFO # gc_check: bix:1 del:6 use:25 score:4555
2015-06-12 13:13:44,786 - INFO # gc_check: bix:2 del:7 use:24 score:4411
2015-06-12 13:13:44,790 - INFO # gc_check: bix:3 del:8 use:23 score:4367
2015-06-12 13:13:44,793 - INFO # gc_check: bix:4 del:8 use:23 score:4317
2015-06-12 13:13:44,797 - INFO # gc_check: bix:5 del:7 use:24 score:4261
2015-06-12 13:13:44,800 - INFO # gc_check: bix:6 del:5 use:26 score:4199
2015-06-12 13:13:44,804 - INFO # gc_check: bix:7 del:7 use:24 score:4161
2015-06-12 13:13:44,807 - INFO # gc_check: bix:8 del:8 use:23 score:4117
2015-06-12 13:13:44,811 - INFO # gc_check: bix:9 del:6 use:25 score:4055
2015-06-12 13:13:44,814 - INFO # gc_check: bix:10 del:5 use:26 score:3999
2015-06-12 13:13:44,818 - INFO # gc_check: bix:11 del:5 use:26 score:3949
2015-06-12 13:13:44,821 - INFO # gc_check: bix:12 del:7 use:24 score:3911
2015-06-12 13:13:44,825 - INFO # gc_check: bix:13 del:6 use:25 score:3855
2015-06-12 13:13:44,829 - INFO # gc_check: bix:14 del:7 use:24 score:3811
2015-06-12 13:13:44,832 - INFO # gc_check: bix:15 del:12 use:19 score:3791
2015-06-12 13:13:44,836 - INFO # gc_check: bix:16 del:7 use:24 score:3711
2015-06-12 13:13:44,840 - INFO # gc_check: bix:17 del:3 use:28 score:3637
2015-06-12 13:13:44,843 - INFO # gc_check: bix:18 del:5 use:26 score:3599
2015-06-12 13:13:44,847 - INFO # gc_check: bix:19 del:4 use:27 score:3543
2015-06-12 13:13:44,850 - INFO # gc_check: bix:20 del:4 use:27 score:3493
2015-06-12 13:13:44,854 - INFO # gc_check: bix:21 del:4 use:27 score:3443
2015-06-12 13:13:44,858 - INFO # gc_check: bix:22 del:4 use:27 score:3393
2015-06-12 13:13:44,861 - INFO # gc_check: bix:23 del:11 use:20 score:3385
2015-06-12 13:13:44,865 - INFO # gc_check: bix:24 del:3 use:28 score:3287
2015-06-12 13:13:44,868 - INFO # gc_check: bix:25 del:5 use:26 score:3249
2015-06-12 13:13:44,872 - INFO # gc_check: bix:26 del:7 use:24 score:3211
2015-06-12 13:13:44,876 - INFO # gc_check: bix:27 del:10 use:21 score:3179
2015-06-12 13:13:44,879 - INFO # gc_check: bix:28 del:7 use:24 score:3111
2015-06-12 13:13:44,883 - INFO # gc_check: bix:29 del:12 use:19 score:3091
2015-06-12 13:13:44,887 - INFO # gc_check: bix:30 del:14 use:17 score:3053
2015-06-12 13:13:44,890 - INFO # gc_check: bix:31 del:11 use:20 score:2935
2015-06-12 13:13:44,894 - INFO # gc_check: bix:32 del:6 use:25 score:2955
2015-06-12 13:13:44,898 - INFO # gc_check: bix:33 del:6 use:25 score:2805
2015-06-12 13:13:44,901 - INFO # gc_check: bix:34 del:5 use:26 score:2849
2015-06-12 13:13:44,905 - INFO # gc_check: bix:35 del:6 use:25 score:2755
2015-06-12 13:13:44,908 - INFO # gc_check: bix:36 del:10 use:21 score:2729
2015-06-12 13:13:44,912 - INFO # gc_check: bix:37 del:12 use:19 score:2691
2015-06-12 13:13:44,916 - INFO # gc_check: bix:38 del:10 use:21 score:2629
2015-06-12 13:13:44,919 - INFO # gc_check: bix:39 del:12 use:19 score:2591
2015-06-12 13:13:44,923 - INFO # gc_check: bix:40 del:10 use:21 score:2529
2015-06-12 13:13:44,927 - INFO # gc_check: bix:41 del:11 use:20 score:2485
2015-06-12 13:13:44,931 - INFO # gc_check: bix:42 del:10 use:21 score:2429
2015-06-12 13:13:44,934 - INFO # gc_check: bix:43 del:11 use:20 score:2385
2015-06-12 13:13:44,938 - INFO # gc_check: bix:44 del:9 use:22 score:2323
2015-06-12 13:13:44,941 - INFO # gc_check: bix:45 del:8 use:23 score:2267
2015-06-12 13:13:44,945 - INFO # gc_check: bix:46 del:8 use:23 score:2217
2015-06-12 13:13:44,949 - INFO # gc_check: bix:47 del:10 use:21 score:2179
2015-06-12 13:13:44,952 - INFO # gc_check: bix:48 del:8 use:23 score:2117
2015-06-12 13:13:44,956 - INFO # gc_check: bix:49 del:10 use:21 score:2079
2015-06-12 13:13:44,960 - INFO # gc_check: bix:50 del:10 use:21 score:2029
2015-06-12 13:13:44,963 - INFO # gc_check: bix:51 del:11 use:20 score:1985
2015-06-12 13:13:44,967 - INFO # gc_check: bix:52 del:6 use:25 score:1905
2015-06-12 13:13:44,971 - INFO # gc_check: bix:53 del:10 use:21 score:1879
2015-06-12 13:13:44,974 - INFO # gc_check: bix:54 del:10 use:21 score:1829
2015-06-12 13:13:44,978 - INFO # gc_check: bix:55 del:9 use:22 score:1773
2015-06-12 13:13:44,982 - INFO # gc_check: bix:56 del:6 use:25 score:1705
2015-06-12 13:13:44,985 - INFO # gc_check: bix:57 del:9 use:22 score:1673
2015-06-12 13:13:44,989 - INFO # gc_check: bix:58 del:10 use:21 score:1629
[... 54 lines cut ...]
2015-06-12 13:13:45,188 - INFO # gc_check: bix:116 del:9 use:22 score:4973
2015-06-12 13:13:45,192 - INFO # gc_check: bix:117 del:8 use:23 score:4917
2015-06-12 13:13:45,195 - INFO # gc_check: bix:118 del:8 use:23 score:6367
2015-06-12 13:13:45,199 - INFO # gc_check: bix:119 del:18 use:13 score:6277
2015-06-12 13:13:45,203 - INFO # gc_check: bix:120 del:19 use:12 score:4933
2015-06-12 13:13:45,207 - INFO # gc_check: bix:121 del:6 use:25 score:4755
2015-06-12 13:13:45,210 - INFO # gc_check: bix:122 del:6 use:25 score:4805
2015-06-12 13:13:45,214 - INFO # gc_check: bix:123 del:7 use:24 score:6261
2015-06-12 13:13:45,218 - INFO # gc_check: bix:124 del:10 use:21 score:4729
2015-06-12 13:13:45,221 - INFO # gc_check: bix:125 del:7 use:24 score:4611
2015-06-12 13:13:45,225 - INFO # gc_check: bix:126 del:6 use:25 score:4655
2015-06-12 13:13:45,229 - INFO # gc_check: bix:127 del:7 use:24 score:4511
2015-06-12 13:13:45,231 - INFO # gc_clean: cleaning block 111
2015-06-12 13:13:45,234 - INFO # gc_clean: state = 0 entry:0
2015-06-12 13:13:45,238 - INFO # gc_clean: FIND_DATA state:0 - found obj id 0041
2015-06-12 13:13:45,242 - INFO # gc_clean: FIND_DATA find objix span_ix:0004
2015-06-12 13:13:45,245 - INFO # spiffs_obj_lu_find_entry_visitor:135 | 0
2015-06-12 13:13:45,249 - INFO # spiffs_obj_lu_find_entry_visitor:141 | 0
2015-06-12 13:13:45,253 - INFO # spiffs_obj_lu_find_entry_visitor:158 | 0
2015-06-12 13:13:45,258 - INFO # spiffs_obj_lu_find_entry_visitor:161 | entry_count = 3968
2015-06-12 13:13:45,263 - INFO # spiffs_obj_lu_find_entry_visitor:165 | obj_lookup_page = 0
```

Cache is disabled here as I use this in my bootloader where I wanted to save as much space as possible.
**edit:** just tried it with cache enabled and it gets stuck regardless

benpicco, 2015-06-12:
It hangs at `fs->cfg.hal_read_f(0, 128, 20005bac)` so that's my code

closed 2015-06-12 by benpicco

pellepl, 2015-06-12:
Phew ;)