pellepl/spiffs #22: two files with the same name

benpicco, 2015-06-16:
Hi again :wink: 
I just discovered this on one of my nodes and I find it a bit unsettling

```
2015-06-16 17:41:49,324 - INFO # > ls
2015-06-16 17:41:49,324 - INFO # 
2015-06-16 17:41:49,326 - INFO # 	4 >dummy.hex.tmp<
2015-06-16 17:41:49,328 - INFO # 	4096 >dummy.hex.tmp<
```

where `ls` is

```
int ls(void) {
	spiffs_DIR dir;
	struct spiffs_dirent file;

	SPIFFS_opendir(&spi_fs, 0, &dir);

	while (SPIFFS_readdir(&dir, &file)) {
		printf("\t%d >%s<\n", file.size, file.name);
	}

	SPIFFS_closedir(&dir);

	return 0;
}
```

I'm not exactly sure how I got there, but I think the fs shouldn't allow this.

pellepl, 2015-06-16:
Hmm. No this should not happen. Can you dump the entire fs to a file and attach it alongside with the fs configuration? Also, what does the obj_id in the spiffs_dirent struct tell you for the files? 
Could it have been a powerloss amidst a rename perhaps?

benpicco, 2015-06-17:
Here you go
[spiffs_config.h](https://gist.github.com/benpicco/b043999cf7d1acb6cc13)
[flash.dump](http://benpicco.de/foobar/flash.dump)

```
2015-06-17 17:38:31,565 - INFO # 	4 dummy.hex.tmp [obj_id = 33024, pix = 5]
2015-06-17 17:38:31,569 - INFO # 	4096 dummy.hex.tmp [obj_id = 32769, pix = 84]
```

pellepl, 2015-06-19:
On it. Sorry to say, things look bad. The fs seems totally borked, and when running an fs_check things hang. Digging on.

pellepl, 2015-06-19:
Hmm. Seems you can have up to 0x10000 pages (8M / 128).  But, in your spiffs_config.h spiffs_page_ix is typedeffed as

typedef u16_t spiffs_page_ix;

It should be 

typedef u32_t spiffs_page_ix;

or spiffs might/will run into endless loops.

benpicco, 2015-06-22:
Ah thank you! And I was wondering why fs_check took so long on the large flash :grin: 
Unfortunately if I make those 4 typedefs 32 bit, `SPIFFS_CHECK_MAGIC_POSSIBLE` will say I can't use magic.

pellepl, 2015-06-22:
Ah darn. Try using 256 byte pages instead. Or 8k blocks. By fiddling around
with those configs you'll sooner or later find a setup having place for the
magic in the LUTs.

benpicco, 2015-06-22:
I just discovered that I need to define `SPIFFS_LOCK`/`SPIFFS_UNLOCK` when using threads… I have a feeing this is somewhat related. I fixed my config and will report back if I see this behavior again, hope I won't :wink:

pellepl, 2015-06-22:
> and will report back if I see this behavior again, hope I won't [image:

That's good to hear. I've been pulling my beard trying to find a reason for
getting the structure in your dump without getting any wizer. The fs is
truly fubared. Guess it could be a simultaneous thing, I truly hope so.

benpicco, 2015-06-23:
Hm, I'm afraid of the integrity of your beard - now I'm using  
```
#define SPIFFS_LOCK(fs) xSemaphoreTakeRecursive(spiffs_mutex, portMAX_DELAY)
#define SPIFFS_UNLOCK(fs) xSemaphoreGiveRecursive(spiffs_mutex)
```
(recursive because of my `SPIFFS_lseek` hackery where `SPIFFS_lseek` calls `SPIFFS_write`) and I'm still getting
```
2015-06-23 15:54:04,565 - INFO # 	0 .volaWare
2015-06-23 15:54:04,566 - INFO # 	4 filetx.bin.tmp
2015-06-23 15:54:04,569 - INFO # 	13416 filetx.bin
2015-06-23 15:54:04,570 - INFO # 	4 filetx.bin.tmp
```
[flash.dump](http://benpicco.de/foobar/flash.dump.2)

Looks like it happens during this 
```
	// tx_id is the checksum of the file
	printf("receiving '%s'\n", packet->payload);

	fd = SPIFFS_open(&spi_fs, (char*) packet->payload, SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
	// generate .tmp file for current transfer
	file_from_buffer(strncat((char *) packet->payload, ".tmp", SPIFFS_OBJ_NAME_LEN), (uint8_t*) &packet->tx_id, sizeof(packet->tx_id));
	// remove previous .a32 file should it exist
	memcpy(packet->payload + strlen((char *) packet->payload) - strlen(".tmp"), ".a32", strlen(".a32"));
	SPIFFS_remove(&spi_fs, (char*) packet->payload);
```
```
uint32_t file_from_buffer(char* file, uint8_t* buf, size_t len) {
	spiffs_file fd = SPIFFS_open(&spi_fs, file, SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);

	if (fd <= 0)
		return 0;

	uint32_t res = SPIFFS_write(&spi_fs, fd, buf, len);

	SPIFFS_close(&spi_fs, fd);

	return res != len ? 0 : res;
}
```

pellepl, 2015-06-23:
*lol* It's pretty dense right now and could use some grooming.

I will try to reproduce in test bench. What does file_from_buffer do?

benpicco, 2015-06-23:
It creates a file with the content of what's behind the given pointer, it's code is attached below.
The corruption doesn't happen reliably, but it just did when the receiving side was slowed down because of `printf()` before each `SPIFFS_LOCK`/`SPIFFS_UNLOCK`.

pellepl, 2015-06-23:
Got it. Damn phone wouldn't show it. By the way, did you tinker with `SPIFFS_lseek`?

benpicco, 2015-06-23:
Yes, I didn't go further than [that](https://github.com/pellepl/spiffs/issues/18#issuecomment-110654025)

pellepl, 2015-07-05:
Just a pong: haven't forgotten about you, on a trip for a week more

pellepl, 2015-07-19:
Hey benpicco,
Been trying to reproduce this in numerous ways, all in vain. If possible, can you give me more info? Eg file sizes, debug outputs, frequency of the problem? Perhaps you can get a clue by looking at the result after each spiffs call, might be something that fails early.

pellepl, 2015-10-08:
Ping?

pellepl, 2015-11-03:
Hey benpicco,

I'll close this now. Feel free to reopen it if needed.

closed 2015-11-03 by pellepl

MaazSk, 2022-01-06:
@pellepl, I am facing the same issue like this 2 files with the same name are generated. I am using esp-idf 4.3.1. I have raised the same[ issue there](https://github.com/espressif/esp-idf/issues/8166). The files are generated when I am creating log files and writing in them for a long time

igrr, 2022-01-06:
FWIW, I found a similar once and reported it, and it got fixed: https://github.com/pellepl/spiffs/issues/74.

However that time I had a simple reproducer, which I don't have now. @MaazSk if you can narrow down the issue to some simple test case, I'm sure we could fix it as well.

MaazSk, 2022-01-14:
Hi @igrr, 

Just wanted to know if any updates on it?

igrr, 2022-01-14:
@MaazSk Thank you for trying to provide some details about the program which triggers this. Unfortunately based on your description I didn't manage to recreate the issue. If you can simplify the program while still reproducing the issue, and find a way how to reduce the time required to trigger the issue, it would help a lot!

MaazSk, 2022-01-14:
@igrr Right now I just did some minor changes in my code like I used the `remove()` whenever I am switching between files as the old file is no longer required. While doing this two files with the same name are not been created. But, then also I will check if I could reproduce the same issue and for sure give you the steps about how to reproduce them.

Thank You!