pellepl/spiffs #23: New feature - Controlled erase.

vladant, 2015-06-17:
Hi Peter,

I have tried spiffs as an alternative to commercial embedded FS. It is interesting how SPIFFS uses free space - it does not erase flash sectors until all of them are used after a format. This is good and bad at the same time, bad because erase is very slow and FS performance in the beginning will differ, good because you probably could distribute erase operations over time or even control them. Have you thought about making a function that will erase several unused sectors ? It also would be nice to have a mean to check whether format is needed or not too.

Vlad.

pellepl, 2015-06-22:
Hi,

ah that's interesting. I've never poked much in a commercial embedded FS, at least not the source ;)

Yes, spiffs is lazy - it only erases when needed. I might have misunderstood, but spiffs will try to erase more than one blocks if needed. It all depends on how many free pages are needed to be allocated. But, there is also a cap: https://github.com/pellepl/spiffs/blob/master/src/default/spiffs_config.h#L72
This cap defaults to that no more than maximum 5 blocks can be erased in one go.

Also, there is an internal gc that just erases blocks with only deleted pages. It is pretty experimental, but should work: https://github.com/pellepl/spiffs/blob/master/src/spiffs_gc.c#L30

Check if a format is needed - what do you mean by this? There is a config define, SPIFFS_USE_MAGIC, that might do what you want. It is somewhat cumbersome in its use (one must try to mount to check if format is needed). See https://github.com/pellepl/spiffs/blob/master/src/default/spiffs_config.h#L119 - hopefully it does what you want.

vladant, 2015-06-22:
Hi, thanks for the answer.

Yes, spiffs_gc_quick() could do this. I think it is an important feature because usually flash is used to store persistent data and it means most of the time flash either idle or read from and only rarely it is written on. It seems natural to erase free sectors during idle time to increase write speed later (if you have free space of course).

As for the commercial FS - it is a small general-purpose FS on top of NOR flash layer which sits on top of SPI-NOR layer. Not very efficient approach for small devices, luckily we do not have to save power.

By the way SPIFFS_check() takes a huge amount of time, in fact I never had enough patience to wait until it ends.

vladant, 2015-06-22:
I have an interesting question slightly related to GC: What will happen if to cut power to a flash memory in the middle of sector erase operation ? I suspect nothing good :). Sometimes I was unable to turn 1-s into 0-s in one particular sector after power-off.

pellepl, 2015-06-22:
Ok I will expose the SPIFFS_gc_quick to the common api and check it up.

Yes, SPIFFS_check does take forever. Given the amount of ram, the whole filesystem must be read several scans in order to check consistency.

Ah, the dreaded powerloss during erase! If SPIFFS_USE_MAGIC is enabled, spiffs will see that one sector not have been fully erased at mount time and erase it again. When SPIFFS_USE_MAGIC is disabled, all hell breaks loose. In theory, a SPIFFS_check can mend this, but I am pretty sure there are cornercases lurching beneath some rock I haven't thought of.

pellepl, 2015-07-13:
Try it out. It might even work! :)

closed 2015-07-13 by pellepl