pellepl/spiffs #26: Nonissue: ESP8266 Arduino examples

TheKitty, 2015-07-01:
With SPIFFS more integrated into the ESP8266 Arduino environment (there is a default 64K FS now in latest builds) and the sizes are adjustable by setting new targets), it would be extremely helpful to have a couple examples of how to use the calls and load files onto the system and read them off.

If you don't have time, not a problem but several ESP enthusiasts are eager to try your great work.

Thank you for your consideration, Mike

pellepl, 2015-07-05:
On a trip for a week more before I can pay full attention, but in the meantime:
what kind of examples are you looking for? Just plain C reading, writing and listing files?

TheKitty, 2015-07-05:
Storing on the device then reading them out when something like an http
request is received. I think that having C code for that use case would be
very popular. No hurries, appreciate your consideration. Mike

pellepl, 2015-07-30:
Not finished yet, but something like this? https://github.com/pellepl/spiffs/wiki/Using-spiffs

TheKitty, 2015-07-30:
Yes, that's great!  You're doing fine.  I don't think that many people will
be upset that the calls are not close enough to Posix, they're just glad
someone has done the hard work to implement a good flash file system.  Best
regards, Mike

pellepl, 2015-08-01:
Thanks ;) I'll be updating this over time (no way back now), so I close this issue.

closed 2015-08-01 by pellepl