pellepl/spiffs #34: User data in callbacks

rojer, 2015-07-28:
Currently callbacks are defined as follows:
```
typedef s32_t (*spiffs_read)(u32_t addr, u32_t size, u8_t *dst);
typedef s32_t (*spiffs_write)(u32_t addr, u32_t size, u8_t *src);
typedef s32_t (*spiffs_erase)(u32_t addr, u32_t size);
```
There is no way to pass any kind of user value here, which would let user distinguish between instances of SPIFFS mounts. Who in the world would need this, you ask?
Well, I won't bore you with details, but we have an application that would benefit from having more than one mount at a time.

I know it's hard to change API at this stage, but would you be open to having a compile-time option to have user data pointer in the callbacks? Perhaps something like this:

```
#ifdef SPIFFS_NEW_CALLBACKS
typedef s32_t (*spiffs_read)(u32_t addr, u32_t size, u8_t *dst, void *user_data);
typedef s32_t (*spiffs_write)(u32_t addr, u32_t size, u8_t *src, void *user_data);
typedef s32_t (*spiffs_erase)(u32_t addr, u32_t size, void *user_data);
#else
typedef s32_t (*spiffs_read)(u32_t addr, u32_t size, u8_t *dst);
typedef s32_t (*spiffs_write)(u32_t addr, u32_t size, u8_t *src);
typedef s32_t (*spiffs_erase)(u32_t addr, u32_t size);
#endif
```

that, plus a field in the `spiffs` struct should be all that's necessary.

WDYT?

pellepl, 2015-07-28:
Sure, should have thought of that myself. How about passing the spiffs struct down to the callbacks, and have a void * user_data in the spiffs struct instead?

rojer, 2015-07-28:
yep, that'll do

yihcdaso-yeskela, 2015-07-29:
> How about passing the spiffs struct down to the callbacks
My +1 for this

pellepl, 2015-07-29:
Ok, great - will do in a near future.

closed 2015-08-01 by pellepl