pellepl/spiffs #39: i2c version?

far5893, 2015-08-11:
Does this file system usable with i2c devices like 24lc1024?
They have 64byte page  and need to wait before send another page (i2c nack)

pellepl, 2015-08-11:
Hmm, at first I would say no, but after looking at the specs this looks
more like a spi flash with i2c bus. Though what i could see the data sheet
mentioned 256 byte pages, not 64. As for your question: maybe. It seems
very possible to write an i2c hal layer for this device. I cant say how
efficient it would be though.

pellepl, 2015-08-11:
Hold your horses, I googled the wrong device. Sorry, but I cannot recommend spiffs for i2c devices in general. The data throughput would be very slow considering the i2c bus speed. Also, i2c eeproms normally does not allow pulling down individual bits to zero which is used extensively in spiffs. E.g if i write 0xFE to some address (being 0xFF in the first place), and then 0x7F to same address, spiffs expects the byte on that address to be 0x7E.

closed 2015-08-12 by pellepl