pellepl/spiffs #40: Confused about padding in "spiffs_page_object_ix"

lishen2, 2015-08-13:
I was confused about padding in "spiffs_page_object_ix", do you mean
 ` u8_t _align[4 - (sizeof(spiffs_page_header)&3)==0 ? 4 : 4 - (sizeof(spiffs_page_header)&3)];`
 instead of
 ` u8_t _align[4 - (sizeof(spiffs_page_header)&3)==0 ? 4 : (sizeof(spiffs_page_header)&3)];` 
to make "spiffs_page_object_ix" 4-byte alignment ?

pellepl, 2015-08-13:
Regarding alignment after spiffs_page_header I want following behaviour:
```
sizeof(spiffs_phdr) | _align
                ... | ...
                  8 | 0
                  9 | 3
                 10 | 2
                 11 | 1
                 12 | 0
```
..and so on, so I think the latter is correct.

However, when looking into it I see that  #jmattson was indeed correct in #31.
It should be actually be
```
u8_t _align[4 - ((sizeof(spiffs_page_header)&3)==0 ? 4 : (sizeof(spiffs_page_header)&3))];
```
(extra parentheses)
Fix coming up.

lishen2, 2015-08-14:
The parentheses make sense.

closed 2015-08-14 by lishen2

lishen2, 2015-08-14:
I think, after modify the padding, the code will incompatible with previous disk image.

pellepl, 2015-08-14:
Yep, there's 50% risk of that, in cases of odd alignments. I am reluctant to break backward compatibility, but in cases of fixing errors there's not much else to do I'm afraid.