pellepl/spiffs #41: Two issues and one question.

lishen2, 2015-08-17:
Hi Peter, I got two issues and one question:

1, In function 'spiffs_obj_lu_find_free_obj_id', I think it shouldn't count object lookup pages when calculate 'max_objects'. So  tt's should be 
'u32_t max_objects = (fs->block_count * SPIFFS_OBJ_LOOKUP_MAX_ENTRIES(fs)) / 2;' instead of 
'u32_t max_objects = (SPIFFS_CFG_PHYS_SZ(fs) / (u32_t)SPIFFS_CFG_LOG_PAGE_SZ(fs)) / 2;'

2，In function 'spiffs_obj_lu_find_free' when not found free object id, 'spiffs_obj_lu_find_id' return 'SPIFFS_ERR_NOT_FOUND', not 'SPIFFS_VIS_END'.

Question:
About power loss. When should I perform SPIFFS_check? In other word, how to detect data corruption? 

thank you

pellepl, 2015-09-01:
Hi, sorry for belated answer - having summer vacation still :)

1) That's very observant of you. I will have to look into that, there are some cornercases that must be considered.

2) This is as intended - SPIFFS_VIS_END is more of an internal status code meaning the visitor reached the end. This need to be interpreted depending on the case the visitor is used. Here, as we try finding free object ids, coming to the end of the visitor means there were no free ids.

As for you question, this is a very good one and I'm a bit surprised no one asked it before.

When I wrote spiffs, it was intended for a board where I had a battery powered backup ram. Using the mutex macros made it easy to set a spiffs_busy bit in the backup ram when entering the mutex and reset it when leaving. Thus, on each powerup I just check this bit - if set, there was a powerloss, and I ran a check.

Now, if you do not have a battery backed ram it is a bit more tricky. Basically, if you get any of the error codes ```SPIFFS_ERR_NOT_FINALIZED```, ```SPIFFS_ERR_NOT_INDEX```, ```SPIFFS_ERR_IS_INDEX```, ```SPIFFS_ERR_IS_FREE```, ```SPIFFS_ERR_INDEX_SPAN_MISMATCH```, ```SPIFFS_ERR_DATA_SPAN_MISMATCH```, ```SPIFFS_ERR_INDEX_REF_FREE```, ```SPIFFS_ERR_INDEX_REF_LU```, ```SPIFFS_ERR_INDEX_REF_INVALID```, ```SPIFFS_ERR_INDEX_FREE```, ```SPIFFS_ERR_INDEX_LU```, ```SPIFFS_ERR_INDEX_INVALID``` you should run a check (I think this list is correct, must check more thoroughly). Also, you should check the ```SPIFFS_info``` function at mount.

I had some idea of reserving a "parity page" in the filesystem. When beginning a file operation spiffs clears a bit, when ending it it clears another bit. On mount, if parity pages does not exist or number of zeroed bits are odd, spiffs could report that a check should run. It'd be a bit tricky to implement this though, so I haven't come to do it. If enough people ask, I'll do it I guess :)

I'll look through the powerloss case more thoroughly and add above answer on the wiki FAQ.

Also, will add an issue for 1) so I won't forget to check it.

Cheers!

lishen2, 2015-09-06:
Hi, 
You have summer vacation? How wonderful. I envy you.

In question 2, I didn't make myself clear. I mean you still compare 'res' with 'SPIFFS_VIS_END' in function 'spiffs_obj_lu_find_free', which is not right. It should compare with 'SPIFFS_ERR_NOT_FOUND'.

About detection of power loss, stm32 MCUs has 'Backup registers' which is battery backed. I think it can be use to store the busy bit. For other MCUs, I think it should has some similar mechanism. At least you can use a insignificant bit in RTC registers, if the MCU has RTC and it is battery backed.

pellepl, 2015-09-14:
Hehe, yes, Sweden is nice in that respect ;)

Q2: Got it - you're right. I didn't read it thoroughly. Sorry.

Powerloss: exactly. This is however something I leave to implement for the user as it is it platform dependent. Will add wiki info.

closed 2015-09-14 by pellepl