pellepl/spiffs #44: Zero-size files cannot be read

rojer, 2015-09-03:
An empty file (zero bytes) can be successfully written but cannot be read.
The errno returned is -10015 (SPIFFS_ERR_INDEX_REF_FREE)

rojer, 2015-09-03:
code:

```c
    FILE *f = fopen("fs2.spiffs", "r");
    int n = fread(image, 1, image_size, f);
    printf("%d bytes\n", n);
    mem_spiffs_mount(); /* this mounts SPIFFS at global fs variable */

    char x[1000];
    int sfd = SPIFFS_open(&fs, "0.bytes", SPIFFS_RDONLY, 0);
    printf("sfd = %d\n", sfd);
    int nr = SPIFFS_read(&fs, sfd, x, 100);
    printf("nr = %d errno = %d\n", nr, SPIFFS_errno(&fs));
```

output:

```
65536 bytes
sfd = 1
nr = -1 errno = -10015
```

if i change it to read `2.bytes` instead (another file on the same filesystem), it works:

```
65536 bytes
sfd = 1
nr = 2 errno = 0
```

pellepl, 2015-09-03:
Ah, good finding! Hadn't thought of that. Will fix after next week, on a
trip.

Cheers!

closed 2015-09-14 by pellepl