pellepl/spiffs #50: Got 31 failed by running ./build/linux_spiffs_test

GallenLin, 2015-10-01:
I am making 'linux_spiffs_test' with revision 1aa4d63 (without any modifications) by following commands in my Mint17 64 bits Linux host : 
mkdir build
make 
./build/linux_spiffs_test

Got some report like this : 
...
...
Test report, 60 tests
29 succeeded
31 failed
  lu_check1
  page_cons1
  page_cons2
  page_cons3
  page_cons_final
  index_cons1
  index_cons2
  index_cons3
  index_cons4
  open_by_dirent
  write_big_file_chunks_page
  write_big_files_chunks_page
  write_big_file_chunks_index
  write_big_files_chunks_index
  write_big_file_chunks_huge
  write_big_files_chunks_huge
  truncate_big_file
  bad_index_1
  bad_index_2
  lseek_simple_modification
  lseek_modification_append
  lseek_modification_append_multi
  gc_quick
  write_small_file_chunks_1
  write_small_files_chunks_1
  write_big_file_chunks_1
  write_big_files_chunks_1
  long_run_config_many_small_one_long
  long_run_config_many_medium
  long_run_config_many_small
  long_run
0 stopped

FAILED

Is there any suggestion on these failed messages ?

pellepl, 2015-10-01:
Run 'make test' instead. Simple as that :)

pellepl, 2015-10-01:
Or, to be safe, run 'make all && make test'. The 'all' will create a needed
testfolder if missing. (I should really fix this someday)
Den 1 okt 2015 12:27 skrev "Peter Andersson" <pelleplutt1976@gmail.com>:

GallenLin, 2015-10-02:
Thanks pellepl , I got following messages after 'make all' . 

ran 60 tests
Test report, 60 tests
60 succeeded
0 failed
0 stopped

closed 2015-10-02 by GallenLin