pellepl/spiffs #52: 0.3.2 <-> 0.3.3 incompatibility

rojer, 2015-10-05:
it's not mentioned in the readme, so i thought i'd ask: is the incompatibility b/w 0.3.2 and 0.3.3 expected?
filesystems produced by them are mutually incompatible, though in a slightly different way.

ideally, of course, i'd like this fixed and compatibility restored, but i do realize we're at version 0.3 here, so i'll accept "yes, deal with it" as an answer, but i want to be sure it's expected. in which case it should be in readme.

fwiw, here's my test output. fs1.{old,new}.spiffs are filesystem images that contain two files" badger_badger.json and mushroom.html, created by 0.3.2-ish and current master respectively. dump_spiffs mounts and lists files on the filesystem image, old and new have been compiled with the same source respectively:

```
$ ./dump_spiffs.old fs1.old.spiffs 
mounted fs1.old.spiffs (65536 bytes)
   0 iddiddddddddddd    era_cnt: N/A
   1 ddddddddddddddd    era_cnt: N/A
   2 ddddddddddddddd    era_cnt: N/A
   3 ddddddddddddddd    era_cnt: N/A
   4 ddd____________    era_cnt: N/A
   5 _______________    era_cnt: N/A
   6 _______________    era_cnt: N/A
   7 _______________    era_cnt: N/A
   8 _______________    era_cnt: N/A
   9 _______________    era_cnt: N/A
  10 _______________    era_cnt: N/A
  11 _______________    era_cnt: N/A
  12 _______________    era_cnt: N/A
  13 _______________    era_cnt: N/A
  14 _______________    era_cnt: N/A
  15 _______________    era_cnt: N/A
era_cnt_max: 1
last_errno:  0
blocks:      16
free_blocks: 11
page_alloc:  63
page_delet:  0
used:        15813 of 52961
check result: 0
entry: |badger_badger.json|
entry: |mushroom.html|
```

new code reads old image with filenames truncated:
```
$ ./dump_spiffs.new fs1.old.spiffs                                                                                                                                                
...same stuff here...
check result: 0
entry: |dger_badger.json|
entry: |shroom.html|
```

old code just doesn't find any files at all:
```
$ ./dump_spiffs.old fs1.new.spiffs 
mounted fs1.new.spiffs (65536 bytes)
   0 iddiddddddddddd    era_cnt: N/A
   1 ddddddddddddddd    era_cnt: N/A
   2 ddddddddddddddd    era_cnt: N/A
   3 ddddddddddddddd    era_cnt: N/A
   4 ddd____________    era_cnt: N/A
   5 _______________    era_cnt: N/A
   6 _______________    era_cnt: N/A
   7 _______________    era_cnt: N/A
   8 _______________    era_cnt: N/A
   9 _______________    era_cnt: N/A
  10 _______________    era_cnt: N/A
  11 _______________    era_cnt: N/A
  12 _______________    era_cnt: N/A
  13 _______________    era_cnt: N/A
  14 _______________    era_cnt: N/A
  15 _______________    era_cnt: N/A
era_cnt_max: 1
last_errno:  0
blocks:      16
free_blocks: 11
page_alloc:  63
page_delet:  0
used:        15813 of 52961
check result: 0
```
(nothing after SPIFFS_check result, so SPIFFS_readdir must've returned NULL straight away)

pellepl, 2015-10-06:
Hi rojer,

Yes you're so right - and it is expected. I actually did not think of issuing a heads up, which of course I should have. The offending commit that cracked the compability is referenced in issue #40.

I strive for keeping compatibility as far as possible, at least by leaving the user with an ifdef choice if new functionality would break compat. But as this was an actual bug I had no choice.

I will add a warning in the readme. Thanks for letting me know.

Oh, and there might be a way to rescue this by yet another define if this poses a big problem for you.

rojer, 2015-10-06:
we have some field deployment of filesystems produced by old code, but we're at the alpha stage ourselves so blowing those won't be a huge deal. thanks for confirming, we'll bite the bullet and push through.

but i think a notice in release notes about things like these in the future is a must.

rojer, 2015-10-06:
just thought of something. would there be an easy way to detect this? can SPIFFS_check be made to fail on such a FS? it it's not too much effort, it'd be appreciated. i don't want to be ignoring errors and blowing away file systems forever just because this might happen to be a pre-0.3.3 FS.

pellepl, 2015-10-06:
Hmm, I'll look into it. If possible, it would only detect 0.3.2 -> 0.3.3 problems. Though I cannot promise anything I'm afraid, I think there aren't any safe patterns to look for to identify it 100%.

rojer, 2015-10-06:
yes, i'm only interested in 0.3.2 -> 0.3.3.

pellepl, 2015-10-06:
Btw, do you have SPIFFS_USE_MAGIC enabled?

rojer, 2015-10-06:
no, not currently. good idea to enable it for this upgrade, i'll do it.

pellepl, 2015-10-06:
Would that work as a fix for you? Knowing that older systems ( < 0.3.3) do not have magic, but the newer  ones have? If you'd try to mount a system not having magic with firmware having magic compiled in, it will return ERR_NOT_A_FS.

rojer, 2015-10-06:
yep, that's actually fortunate. that'll do for now. thanks!

closed 2015-10-06 by rojer

pellepl, 2015-10-06:
Sweet!