pellepl/spiffs #56: Fatal exception 9(LoadStoreAlignmentCause).

sguilly, 2015-11-01:
I try to use the lib with an ESP8266 module.

But, the code crash with this line :

 ((spiffs_page_ix*)((u8_t *)objix_hdr + sizeof(spiffs_page_object_ix_header)))[data_spix] = data_page;

Fatal exception 9(LoadStoreAlignmentCause).

Thanks for your help

Stéphane

Some trace :

1. mounting
Read flash 0x002581fe   Read flash 0x002681fe   Read flash 0x002781fe   Read flash 0x002881fe   Read flash 0x002981fe   Read flash 0x002a81fe   Read flash 0x002b81fe   Read flash 0x002c81fe       Read flash 0x002d81fe   Read flash 0x002e81fe   Read flash 0x002f81fe   Read flash 0x003081fe   Read flash 0x003181fe   Read flash 0x003281fe   Read flash 0x003381fe       Read flash 0x003481fe   Read flash 0x003581fe   Read flash 0x003681fe   Read flash 0x003781fe   Read flash 0x003881fe   Read flash 0x003981fe   Read flash 0x003a81fe       Read flash 0x003b81fe   Read flash 0x003c81fe   Read flash 0x003d81fe   Read flash 0x003e81fe   Read flash 0x003f81fe   Read flash 0x004081fe   Read flash 0x004181fe       Read flash 0x004281fe   Read flash 0x004381fe   Read flash 0x004481fe   Read flash 0x00258000   Read flash 0x00258100   Read flash 0x00268000   Read flash 0x00268100       Read flash 0x00278000   Read flash 0x00278100   Read flash 0x00288000   Read flash 0x00288100   Read flash 0x00298000   Read flash 0x00298100   Read flash 0x002a8000       Read flash 0x002a8100   Read flash 0x002b8000   Read flash 0x002b8100   Read flash 0x002c8000   Read flash 0x002c8100   Read flash 0x002d8000   Read flash 0x002d8100       Read flash 0x002e8000   Read flash 0x002e8100   Read flash 0x002f8000   Read flash 0x002f8100   Read flash 0x00308000   Read flash 0x00308100   Read flash 0x00318000       Read flash 0x00318100   Read flash 0x00328000   Read flash 0x00328100   Read flash 0x00338000   Read flash 0x00338100   Read flash 0x00348000   Read flash 0x00348100       Read flash 0x00358000   Read flash 0x00358100   Read flash 0x00368000   Read flash 0x00368100   Read flash 0x00378000   Read flash 0x00378100   Read flash 0x00388000       Read flash 0x00388100   Read flash 0x00398000   Read flash 0x00398100   Read flash 0x003a8000   Read flash 0x003a8100   Read flash 0x003b8000   Read flash 0x003b8100       Read flash 0x003c8000   Read flash 0x003c8100   Read flash 0x003d8000   Read flash 0x003d8100   Read flash 0x003e8000   Read flash 0x003e8100   Read flash 0x003f8000       Read flash 0x003f8100   Read flash 0x00408000   page index byte len:         256
object lookup pages:         2
page pages per block:        256
page header length:          5
object header index entries: 106
object index entries:        125
available file descriptors:  6
free blocks:                 26
mount res: 0

2. openning a file
Read flash 0x00258000   Read flash 0x00258100   Read flash 0x00268000   Read flash 0x00268100   Read flash 0x00278000   Read flash 0x00278100   Read flash 0x00288000   Read flash 0x00288100       Read flash 0x00298000   Read flash 0x00298100   Read flash 0x002a8000   Read flash 0x002a8100   Read flash 0x002b8000   Read flash 0x002b8100   Read flash 0x002c8000       Read flash 0x002c8100   Read flash 0x002d8000   Read flash 0x002d8100   Read flash 0x002e8000   Read flash 0x002e8100   Read flash 0x002f8000   Read flash 0x002f8100       Read flash 0x00308000   Read flash 0x00308100   Read flash 0x00318000   Read flash 0x00318100   Read flash 0x00328000   Read flash 0x00328100   Read flash 0x00338000       Read flash 0x00338100   Read flash 0x00348000   Read flash 0x00348100   Read flash 0x00358000   Read flash 0x00358100   Read flash 0x00368000   Read flash 0x00368100       Read flash 0x00378000   Read flash 0x00378100   Read flash 0x00388000   Read flash 0x00388100   Read flash 0x00398000   Read flash 0x00398100   Read flash 0x003a8000       Read flash 0x003a8100   Read flash 0x003b8000   Read flash 0x003b8100   Read flash 0x003c8000   Read flash 0x003c8100   Read flash 0x003d8000   Read flash 0x003d8100       Read flash 0x003e8000   Read flash 0x003e8a00   Read flash 0x003e8100   Read flash 0x003f8000   Read flash 0x003f8100   Read flash 0x00408000   free_obj_id: COMP min:0001 max:1001 compact:16
Read flash 0x00258000   Read flash 0x00258100   Read flash 0x00268000   Read flash 0x00268100   Read flash 0x00278000   Read flash 0x00278100   Read flash 0x00288000   Read flash 0x00288100       Read flash 0x00298000   Read flash 0x00298100   Read flash 0x002a8000   Read flash 0x002a8100   Read flash 0x002b8000   Read flash 0x002b8100   Read flash 0x002c8000       Read flash 0x002c8100   Read flash 0x002d8000   Read flash 0x002d8100   Read flash 0x002e8000   Read flash 0x002e8100   Read flash 0x002f8000   Read flash 0x002f8100       Read flash 0x00308000   Read flash 0x00308100   Read flash 0x00318000   Read flash 0x00318100   Read flash 0x00328000   Read flash 0x00328100   Read flash 0x00338000       Read flash 0x00338100   Read flash 0x00348000   Read flash 0x00348100   Read flash 0x00358000   Read flash 0x00358100   Read flash 0x00368000   Read flash 0x00368100       Read flash 0x00378000   Read flash 0x00378100   Read flash 0x00388000   Read flash 0x00388100   Read flash 0x00398000   Read flash 0x00398100   Read flash 0x003a8000       Read flash 0x003a8100   Read flash 0x003b8000   Read flash 0x003b8100   Read flash 0x003c8000   Read flash 0x003c8100   Read flash 0x003d8000   Read flash 0x003d8100       Read flash 0x003e8000   Read flash 0x003e8a00   Read flash 0x003e8100   Read flash 0x003f8000   Read flash 0x003f8100   Read flash 0x00408000   free_obj_id: COMP select index:0 min_count:0 min:0001 max:1001 compact:16
Read flash 0x00258000   create: found free page @ 0002 bix:0 entry:0

spiffs write 0x  258000 [▒▒?] [2] bytes

spiffs write 0x  258200 [▒] [43] bytes
Read flash 0x00258200   open: fd 1 is obj id 8001

3. writing to a file
append: 12 bytes @ offs 0 of size -1
append: 8001 load objixhdr page 0002:0000

spiffs write 0x  258002 [] [2] bytes

spiffs write 0x  258300 [] [5] bytes

spiffs write 0x  258305 [Hello world] [12] bytes
append: 8001 store new data page, 0003:0000 offset:0, len 12, written 0
append: 8001 wrote page 0003 to objix_hdr entry 00 in mem

Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x4028a888, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3ffeb4a7, depc=0x00000000

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

pellepl, 2015-11-01:
>    1. writing to a file append: 12 bytes @ offs 0 of size -1 append: 8001

Try enabling SPIFFS_ALIGNED_OBJECT_INDEX_TABLES.

Cheers / Peter

sguilly, 2015-11-02:
Big thank,

This fix my issue.

pellepl, 2015-11-02:
Great!

closed 2015-11-02 by pellepl

danielpazeto, 2016-04-26:
Hi @pellepl,
I'm new here and I'm having the same issue, but I dont know how enable SPIFFS_ALIGNED_OBJECT_INDEX_TABLES.
Can u explain how I do it?

pellepl, 2016-04-26:
Look for the file spiffs_config.h - in there are all the build time config defines. Define SPIFFS_ALIGNED_OBJECT_INDEX_TABLES to 1.