pellepl/spiffs #57: Zero-size file cannot be seek

lishen2, 2015-11-03:
An empty file (zero bytes) cannot be seek to the end or beginning
The errno returned is SPIFFS_ERR_END_OF_OBJECT

Bug is in 'SPIFFS_lseek'
```
  if (offs > (s32_t)fd->size) {
    res = SPIFFS_ERR_END_OF_OBJECT;
  }
```
When seeking empty file, fd->size is SPIFFS_UNDEFINED_LEN , the statement (offs > (s32_t)fd->size) fail(zero is bigger than negative number).
The following modification will fix:
```
  if ((offs > (s32_t)fd->size) && (SPIFFS_UNDEFINED_LEN != fd->size)) {
    res = SPIFFS_ERR_END_OF_OBJECT;
  }
```

pellepl, 2015-11-03:
Ah yes, you're right.  Good spottning,  thanks!

lishen2, 2015-11-03:
I feel like I should thank you for create spiffs and maintain it.

pellepl, 2015-11-03:
Thanks! Always fun when people find use of something home-tinkered ;) I'll fix the issue this evening.

closed 2015-11-03 by pellepl