pellepl/spiffs #58: SPIFFS_write function does not work when SPIFFS_APPEND is specified

zdrc, 2015-11-16:
I'm on Mac OSX 10.11.1, and I'm using the spiffs framework with sming to program an ESP201

I ran into a problem where I was unable to write to a file (using SPIFFS_write) I had opened with the `SPIFFS_APPEND` flag called. I was getting the error `SPIFFS_ERR_NOT_WRITABLE` with the error code -10021. After some testing I discovered that the root of the problem seems to be this section of code,

```
if ((fd->flags & SPIFFS_WRONLY) == 0) {
  res = SPIFFS_ERR_NOT_WRITABLE;
  SPIFFS_API_CHECK_RES_UNLOCK(fs, res);
}
```

I believe this code dictates that SPIFFS_write will only work if opened when the flag is set to `SPIFFS_WRONLY` this becomes a problem when calling SPIFFS_write when a file is opened in `SPIFFS_APPEND`

If I change the bitwise operation in the if clause's condition to 
`(fd-> flags & (SPIFFS_WRONLY | SPIFFS_APPEND)) == 0`
I am able to append to the file I opened, and my code behaves as it should.

pellepl, 2015-11-17:
Hmm.. The code looks correct to me (though I have not slept enough ;))

Did you open the file with only SPIFFS_APPEND? You probably want to open it with (SPIFFS_RDWR | SPIFFS_APPEND) or (SPIFFS_WRONLY | SPIFFS_APPEND). Specifying only SPIFFS_APPEND when opening the file would lead to a file that cannot be read nor written.

zdrc, 2015-11-17:
Ah I see. In the Sming repository code only SPIFFS_APPEND is used. It sounds like there was a miscommunication on how to set the flags, I'll discuss with them.

pellepl, 2015-11-17:
Yes, I tried opening a file on linux with only O_APPEND and only got a bad
filedesc in return.
But thanks for the heads up!

closed 2015-11-18 by pellepl