pellepl/spiffs #63: SPIFFS_USE_MAGIC and erase race problem

dobryj, 2016-01-08:
Hello,

I found one issue with SPIFFS_USE_MAGIC. Problem is simple: erase a sector, and reset/turn off the device before a magic write. In this case the sector is erased, but lost for SPIFFS, because does not contain the magic on its begining.

Fix for this can be simple, but incomplete (see below): write the magic into the sector, when it is detected as erased during a mount.

But switch off during an undesirable moment and flash content is more complicated problem. It can happen during erasing flash in physical layer. After this, flash sector will not be completely erased and can't be used for SPIFFS (independent if the magic is used or not).

Therefore here is better solution:
- use magic every time
- when sector does not contain magic, erase it and write the magic on mount
- when sector contain invalid non-SPIFFS data, erase it and write the magic on mount

pellepl, 2016-01-08:
It is implemented the way you describe it. If a block is found without magic, it is erased. And spiffs always write magic directly after erasing a block. Also, if spiffs finds more than one bad block the entire fs is formatted. 

Have you found something wierd it should be some other bug. Do you have more details?

dobryj, 2016-01-08:
Opps. I has more than one block without magic. If I understand correctly, in this case mount function returns -1 and I must call format function. It is not done automatically in mount. Correct?

In this case, spiffs_obj_lu_scan bail out for more than one unerased block. And return code bail out from mount.

Thanks for the information and sorry for invalid issue. Have a nice day.

closed 2016-01-08 by dobryj

pellepl, 2016-01-10:
You're right, no automatic formatting. Spiffs_errno should return a SPIFFS_ERR_NOT_A_FS.

Cheers!