pellepl/spiffs #64: creating a file fails with different errors

benpicco-tmp, 2016-01-12:
Hi,
again, I'm having a strange issue - which might well be on my part, but I'm not sure where to look.
We changed our MCU on the new board but kept all the code and also the flash chip, so after implementing my [interface for spi](https://gist.github.com/benpicco-tmp/a2ce0cdc1aac3dfd0e31), spiffs should work again happily, right?
The flash driver remained untouched and it does write and read back the raw data, at least in my test cases so far - so spiffs shouldn't behave any different than on the old hardware. ([old hardware](http://flauschlabor.de/foobar/vt4x4_dump_fresh.hex) -[new hardware](http://flauschlabor.de/foobar/vt8x8_dump_fresh.hex))

Except for when it does - in interesting ways.

I've narrowed it down to one byte being different after initializing the fs (at least that's consistent), what happens afterwards (especially the error values that I pull out of spiffs) seems to be rather random:

```
> ls
2016-01-12 18:20:56,238 - INFO # 	0 .volaWare
> dump_rom hello_cli.hex
2016-01-12 18:21:06,240 - INFO # writing firmware to hello_cli.hex…
2016-01-12 18:21:07,752 - INFO # 0 bytes written
2016-01-12 18:21:07,758 - INFO # error: -10072
> ls
2016-01-12 18:21:16,956 - INFO # 	0 .volaWare
2016-01-12 18:21:16,956 - INFO # 	0 hello_cli.hex
2016-01-12 18:21:16,962 - INFO # 	4 hello_cli.hex.a32
> dump_rom hello_cli.hex
2016-01-12 18:23:20,202 - INFO # writing firmware to hello_cli.hex…
2016-01-12 18:23:20,202 - INFO # 0 bytes written
2016-01-12 18:23:20,203 - INFO # error: -10072
> dump_rom hello_cli.hex
2016-01-12 18:23:22,243 - INFO # writing firmware to hello_cli.hex…
2016-01-12 18:23:23,754 - INFO # 0 bytes written
2016-01-12 18:23:23,760 - INFO # error: -10002
> dump_rom hello_cli.hex
2016-01-12 18:23:25,800 - INFO # writing firmware to hello_cli.hex…
2016-01-12 18:23:25,801 - INFO # 0 bytes written
2016-01-12 18:23:25,806 - INFO # error: -10015
> dump_rom hello_cli.hex
2016-01-12 18:23:26,628 - INFO # writing firmware to hello_cli.hex…
2016-01-12 18:23:26,629 - INFO # 0 bytes written
2016-01-12 18:23:26,630 - INFO # error: -10003
> dump_rom hello_cli.hex
2016-01-12 18:23:27,150 - INFO # writing firmware to hello_cli.hex…
2016-01-12 18:23:27,151 - INFO # 0 bytes written
2016-01-12 18:23:27,156 - INFO # error: -10015
> ls
2016-01-12 18:23:32,869 - INFO # 	0 .volaWare
2016-01-12 18:23:32,874 - INFO # 	0 hello_cli.hex
2016-01-12 18:23:32,875 - INFO # 	0 hello_cli.hex.a32
2016-01-12 18:23:32,880 - INFO # 	4 hello_cli.hex.a32
```

(that `dump_rom` fails at `SPIFFS_open(&spi_fs, file, SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);` already, I'm not sure why a later `open` that creates the checksum file does through)

The initialization looks like this

```
#define LOG_PAGE_SIZE 128
static u8_t spiffs_work_buf[LOG_PAGE_SIZE*2];
static u8_t spiffs_fds[32*4];
static u8_t spiffs_cache_buf[(LOG_PAGE_SIZE+32)*4];

spiffs spi_fs;

void my_spiffs_mount(bool create_magic) {
    spiffs_config cfg;
    cfg.phys_addr        = 0;           // start spiffs at start of spi flash
    cfg.phys_size        = flash_get_size() - cfg.phys_addr; // use all spi flash
    cfg.phys_erase_block = BLOCK_4K;    // according to datasheet
    cfg.log_block_size   = BLOCK_4K;    // let us not complicate things
    cfg.log_page_size    = LOG_PAGE_SIZE;
    cfg.hal_read_f  = (spiffs_read)  flash_read_page;
    cfg.hal_write_f = (spiffs_write) flash_write_page;
    cfg.hal_erase_f = (spiffs_erase) flash_erase_block;
    int res = SPIFFS_mount(&spi_fs,
        &cfg,
        spiffs_work_buf,
        spiffs_fds,
        sizeof(spiffs_fds),
        spiffs_cache_buf,
        sizeof(spiffs_cache_buf),
        0);

    if (res == 0) {
        // 'magic' file
        res = SPIFFS_open(&spi_fs, ".volaWare", create_magic ? SPIFFS_CREAT : SPIFFS_RDONLY, 0);
        if (res > 0)
            SPIFFS_close(&spi_fs, res);
    }
    if (res < 0) {
        printf("mount failed (%d), formatting device…\n", res);
        flash_erase_block(0, BLOCK_ALL);
        printf("done.\n");

        my_spiffs_mount(true);
    }
}
```

Do you have any idea what that might be, when that byte is written and what might happen here?
There is no OS in the background and only the uart interrupt triggering - just to be sure I also disable all interrupts during an spi transfer.

Thanks again!

pellepl, 2016-01-12:
Well, my first guess would be that you switched to a faster processor/using quicker spi bus. Writes to a spi flash takes some time, and issuing subsequent writes too soon will make them silently fail. In you spi flash write function, do you check the BUSY bit in the spi flash status register, or do you have the busy pin mapped to the processor in some way? One quick thing to do for verifying this hypothesis would be to simply add a ~4ms delay after each spi flash write.

benpicco-tmp, 2016-01-13:
I did indeed switch to a faster processor, I moved from a tm4c123 to stm32f446, but delays didn't help, they only made everything slower, still got the same errors (I also formatted the flash every time to avoid errors due to a corrupted fs). I already do check the busy bit in the spi flash status register 

I'm writing to spi like so:

```
static inline uint8_t spi_send(SPI_TypeDef* SPI, uint8_t data) {
    // wait until SPI is ready
    while ((SPI->SR & SPI_FLAG_BSY));

    // Setting the Data Register (DR) transmits the byte of data on MOSI.
    SPI->DR = data;

    // Wait until the data has been transmitted.
    while (!(SPI->SR & SPI_FLAG_TXE));

    // Wait for any data on MISO pin to be received.
    while (!(SPI->SR & SPI_FLAG_RXNE));

    // Return the data received on MISO pin.
    return SPI->DR;
}

// responses to cmd_buffer are ignored, responses to data_buffer are stored in data_buffer
void spi_tx(const spi_cfg_t cfg, const uint8_t *cmd_buffer, int cmd_buffer_len, uint8_t *data_buffer, int data_buffer_len, uint8_t flags) {

  // just to be safe
  master_int_disable();

  // CS low
  ((GPIO_TypeDef *) cfg.port_cs)->ODR &= !cfg.pin_cs;

  for (int i = 0; i < cmd_buffer_len; ++i) {
    spi_send((SPI_TypeDef*) cfg.base, cmd_buffer[i]);
  }

  uint8_t data;
  for (int i = 0; i < data_buffer_len; ++i) {
    data = spi_send((SPI_TypeDef*) cfg.base, data_buffer[i]);

    if (!(flags & SPI_DATA_IMMUTABLE))
      data_buffer[i] = data;
  }

  // CS high
  ((GPIO_TypeDef *) cfg.port_cs)->ODR |= cfg.pin_cs;

  master_int_enable();
}
```

for the flash driver itself I'm using [this](https://gist.github.com/benpicco-tmp/3c18eb4ad40600801d26), spiffs is configured with

```
    cfg.hal_read_f  = (spiffs_read)  flash_read_page;
    cfg.hal_write_f = (spiffs_write) flash_write_page;
    cfg.hal_erase_f = (spiffs_erase) flash_erase_block;
```

I don't know where that error comes from, it's the content of `spi_fs.err_code` after `SPIFFS_open` returns -1.

pellepl, 2016-01-14:
Hmm. I cannot see any obvious problem with the spi flash driver.

How do you reproduce the error exactly? After format, you seem to add a "volaware" file, and then the dump_rom does something - what?

What confuses me is that you get different error codes when listing files. This might indicate that something else is using same memory as spiffs e.g. you've passed stack content when it should be static buffer data or something like that. 

It might also be a good idea to check your linker setup to see if any regions overlaps or are undefined.

Found out the error code -10072, it is the internal ```SPIFFS_VIS_END ```. It is a "positive" error meaning ```SPIFFS_readdir``` reached end of fs. I probably should fix this so it is reported as a real error code, something like ```SPIFFS_ERR_READDIR_END``` or something.

Getting  ```-10015 SPIFFS_ERR_INDEX_REF_FREE``` is bad, indicates a corrupt fs.

What version are you using? Could you please post your spiffs_config.h? Also, enable ```SPIFFS_DBG``` and see what the output is.

benpicco-tmp, 2016-01-14:
Looking into `dump_rom_to_file` I indeed found the bug (although I'm still wondering why it produced those errors)

What the function does is this

```
static uint32_t dump_rom_to_file(char *file) {
	uint32_t *flash = (uint32_t*) APP_START_ADDRESS;
	uint8_t empty = 0;

	spiffs_file fd = SPIFFS_open(&spi_fs, file, SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);

	if (fd <= 0)
		return 0;

	while ((uint32_t) flash - APP_START_ADDRESS < get_rom_size() && !(empty > 2 && (int) flash % 16 == 0)) {
		SPIFFS_write(&spi_fs, fd, (uint8_t*) flash, sizeof(uint32_t))

		if (*flash == 0xffffffff)
			++empty;
		else
			empty = 0;

		++flash;
	}

	SPIFFS_close(&spi_fs, fd);

	return (uint32_t) flash - APP_START_ADDRESS;
}
```

Well what it actually did was to compare `(uint32_t) flash < get_rom_size()` - this worked on tiva where flash starts at 0x0, but always fails on stm32 where flash is at 0x08000000 so it would always write 0 bytes.

So those errors are probably just a leftover of some previous internal operation?

Sorry to have bothered you again :sweat_smile:

closed 2016-01-14 by benpicco-tmp

pellepl, 2016-01-14:
Ok, good to hear you found it. Yes, those errors could be leftovers as you say - still, I am a bit confused over that behaviour. I'll look into it eventually.