pellepl/spiffs #65: Implementation of "defragmentation" method needed

me-no-dev, 2016-01-21:
Hello :) this is a cry out for a defragmentation function.
The case is as follows:
We are using 4MB ESP8266 mcus and run web servers on them.
We often edit, write and change files on the system and all is fine and dandy until we get to the point where all of our free blocks are gone, we are full of deleted pages and each new write requires to go around, relocate pages to free blocks and so on. While that is usually not a big deal for small files, it's really noticeable on larger ones that require many blocks to be cleared. And since recent changes in Espressifs SDK, while running asynchronous network and writing an upload to a file, the stack loses retransmitted packets and data is lost. I know that fact has nothing to do with you, but I still find in many cases the need to speed up the file system.
As an example, I have this 100K test file that get's uploaded for 440ms when SPIFFS has enough free blocks and for 2000ms when there are none (if the upload succeeds). And I have filled actually less than 10% of the available space with data. The rest is all deleted pages.

me-no-dev, 2016-01-21:
yeah... I see methods here that are not implemented on our side. Sorry for bothering with false alarms.

closed 2016-01-21 by me-no-dev

pellepl, 2016-01-22:
Just for the record: going from 440ms on a clean system to 2000ms on a dirty one is by no means weird. Have a peek on a common spi flash datasheet, [winbond](https://www.winbond.com/resource-files/w25q128fv%20rev.l%2008242015.pdf) for instance. 

A page program takes typically 0.7ms, max 3 ms. A 100k file would correspond to 400 pages, giving 280ms to 1200ms. 

A 64k block erase takes typically 150ms, max 2000ms. A 100k file spans at minimum 2 blocks, probably more - 300ms to 6000ms. 

This is spi flash characteristics, not counting in any code execution.

I might be missing something, bit a defrag would not achieve anything. You still need to erase blocks. Sure, one might try to make files more linear in order to be able to erase less blocks, but then the wear leveling would break, reducing the life time of the flash.

Regarding reading files there is no boost what so ever having them linearly written in the spi flash. Spiffs uses a first order indirection. A file written linearly over pages will be read just as quick (or slow) as a file being scattered over the flash.

me-no-dev, 2016-01-22:
I ended up writing a clean function deal with deleted pages when they fill the space and have no empty blocks left (apart from the are minimum). It worked well and I very well understand the consequences of using such function, but... having the FS full of deleted pages and wasting time to deal with it in a network ISR is not something that I like doing and proves problematic. I have the understanding that while writing new files, this exact same thing is happening, but just block by block as they are needed, so I am deleting blocks and putting wear on the flash anyway. The purpose is to not spend that same time while writing the new files and do it in the loop where it can take it's time to optimize once it hits certain level, thus not putting more wear on the flash than it would have been otherwise.
And it worked charms! I get successive uploads and they take a fifth of the time. I imagine I put much more wear on the flash trying to upload a file and missing some data from it, then uploading again and again until success :)
The fact that you exposed the gc methods is enough for someone to hack a similar routine and get the job done if in a similar situation. Cheers :)

pellepl, 2016-01-22:
Good to hear! Yes, the SPIFFS_gc* was exposed for this case. Using gc_quick will affect leveling to the worse, but SPIFFS_gc is exactly what is called from within spiffs. Whichever, I have yet to see a flash fail due to erases, and I've been running spiffs daily on the same for some years now.  😉