pellepl/spiffs #69: File systems configuration and performance depend on static build-time configuration

JBlackArty, 2016-02-19:
Types spiffs_block_ix, spiffs_page_ix, spiffs_obj_id, spiffs_span_ix determine structure of all file systems supported by given application build variant. These sizes not only are being skipped by magic verification (SPIFFS_USE_MAGIC=1), but also restrict support of multiple instances (SPIFFS_SINGLETON=0).
Also there are some configuration macros, which values are dependent on logical structure (such as SPIFFS_COPY_BUFFER_STACK having comment "...no meaning setting this bigger than the logical page size...").
I suggest these macros/types/sizes must be dynamically used and included in spiffs_config for SPIFFS_SINGLETON=0 variant.

pellepl, 2016-02-23:
Yes, this is all correct. Regarding running multiple instances, one must see to that all build time sizes and types are configured as largest common denominator.  The trade-off is simpler and smaller code versus possibly unused bits in the flash.

Rewriting this would be quite a substantial thing to do, and nothing that I've planned for.

ArtemPisarenko, 2016-02-24:
But typedefs cannot be configured in such way. In code I saw macros where sizeof()'s were used in calculations of offsets in flash. So they must exactly match underlying file system. Please, correct me if I wrong.

pellepl, 2016-02-24:
No, you're fully correct - the typedefs must match up with the underlying structure. What I meant was that, for example, if you need to run two filesystems A and B, and A must have a spiffs_obj_id of at least 16 bits, and B needs a spiffs_obj_id of 32 bits, you declare spiffs_obj_id to 32 bits. For both. In A case, you will waste bits, but both A and B will be compatible for the spiffs binary you are compiling. And can live happily together in a non singleton build.

When it comes to SPIFFS_USE_MAGIC, the position of the magic will move around depending on the size of the typedefs. In, if not all, at least (pretty?) many cases the verification will work. Perhaps :)

I might be missing something here, but I don't really feel I understand the source of the question. When I wrote spiffs I took it for granted that the implementer knew the configuration of the spiffs structure.

ArtemPisarenko, 2016-02-25:
Ok. This is not question. I've just filed issue pointing to frustrating limitation. Given spiffs binary cannot work with already existing multiple filesystems, where each of them were previously formatted by binary with it's own (possibly even unknown!) typedefs.
Maybe it's very uncommon situation and you will never fix that, but documentation should explicitly warn users about that, because it's not obvious ;)

pellepl, 2016-02-25:
I'm sorry, but I just cannot see it as a bigger limitation. Please elaborate if  you have a specific use case. It might well be that I'm missing something.

In my little world, spi flash contents are very "contained" within their application, so to speak. Would spiffs images commonly be shared between different targets somehow then yes, this would be an obstacle. Also, in such a case there is another problem as spiffs just dumps structs directly to flash from memory, meaning target endiannes (and possibly even compiler) adds to the dependency chain.

Spiffs was written for small cpu:s, so i took these shortcuts to keep the binary at a minimum.

ArtemPisarenko, 2016-02-25:
I cannot imagine very common use cases right now... Maybe something like following:
- initial version of application uses one filesystem occupying part of flash;
- developer releases new version, which extends functionality with additional filesystem, requiring different config and occupyng different part of flash;
- developer deploys firmware upgrade, which should keep working with already existing data, stored in first filesystem.

Anyway it's just my personal opinion, that any _non-obvious_ interchangeability limitation should be documented. Yes, you clearly documented design objectives of spiffs, but such limitation may be left unnoticed by some users.

pellepl, 2016-03-01:
Yes, you do have a point. I'll add a wiki entry somewhere appropriate.

closed 2016-04-06 by pellepl