pellepl/spiffs #70: It's not clear how to handle mount failure in formatting procedure

ArtemPisarenko, 2016-02-19:
For SPIFFS_USE_MAGIC=1 [wiki page](../wiki/Using-spiffs#formatting) says "If SPIFFS_mount fails with SPIFFS_ERR_NOT_A_FS, keep going. Otherwise, call SPIFFS_unmount", but [SPIFFS_format()](../blob/1d9f6707407ea2c9486d1cf12bbc2ba8a877839c/src/spiffs.h#L569) documentation says that unmount must not be called if it simply fails with any error.
Which of these correct ?

pellepl, 2016-02-23:
To me, they both mean the same thing. Then again, one should never let the programmer explain the system..

The critical points are:
1) we must have tried to mount once to init spiffs with correct config.
2) we are NOT mounted when calling format().
If mounting with an error we have tried mounting, but are not mounted. If the error is anything other than SPIFFS_ERR_NOT_A_FS, it could be safe to proceed but that depends on the error.

E.g, if you get error codes originating from your own spi flash HAL layer, you should probably not proceed to formatting before checking things.

Suggestions for words more comprehensible are always welcome :)

closed 2016-02-23 by pellepl

ArtemPisarenko, 2016-02-24:
Now I've got the point. Confusion was caused by fact that you mentioned things going beyond a scope of just formatting procedure. ;)