pellepl/spiffs #71: It's not clear how SPIFFS_check powerloss rollback works.

ad7718, 2016-02-24:
I read the wiki and it's unclear for me what happens in the following situation: there is a file in FS, for example, 4x bigger than logical page size, I call SPIFFS_write once to rewrite whole file contents, 2 pages written successfully, and powerloss happened on the third. Will SPIFSS_check rollback only third page, or all of them?

pellepl, 2016-02-24:
Only third page. spiffs works by pages only. The successfully written pages are "finalized" and committed, it is only just the page currently being written that is dirty and will be rollbacked.

ad7718, 2016-02-25:
If I'll set logical page size to for example 4096 and work with file that is smaller than it, will SPIFFS_check rollback my file correctly in all powerloss cases?

pellepl, 2016-02-25:
Hmm, in theory yes. But I never tried spiffs with that large page sizes. Make sure you have appropriately large blocks.

And I cannot give my word on every

pellepl, 2016-03-01:
ok 2 close?

ad7718, 2016-03-01:
ok, thanks.

closed 2016-03-01 by ad7718