pellepl/spiffs #72: SPIFFS_eof() returns wrong value

ArtemPisarenko, 2016-02-29:
Two different cases.
1. When SPIFFS_eof() called on just opened empty file, it returns 0.
Debugging shows that fd->size = 0xFFFFFFFF at [here](../blob/1d9f6707407ea2c9486d1cf12bbc2ba8a877839c/src/spiffs_hydrogen.c#L1056), looking abnormal for me.
2. When file opened with SPIFFS_APPEND, seeked to the middle and being written, calling SPIFFS_eof() returns 0.

pellepl, 2016-03-01:
Ok, thanks for reporting - I'll look into it. 

Regarding length = 0xffffffff - this is valid for newly created files without content. Regard 0xffffffffff as 0. It is a minor optimization when actually writing something in the future. Instead of needing to allocate a new page,I just fill in the length. However, eof probably does not take this special case in account.

I remember I had to make something not so beautiful with file offsets when caches was created, I hope I can fix tell to be posix-compliant.

Anyhow, thanks again for reporting.

ArtemPisarenko, 2016-03-01:
Glad to help you with improvements. And thank YOU for such great work ! :)

closed 2016-03-07 by pellepl