pellepl/spiffs #73: Spiffs fails with SPIFFS_ERR_IS_INDEX with increased block size

mozturk, 2016-03-02:
Hi @pellepl,

Was hoping you had some insight into an issue I've been trying to solve all day. I've been using spiffs with a logical page size of 256 successfully for a while, but haven't had any luck moving to a page size of 512 (or greater). Formatting, file creation and writing seem to work fine, but reading back from the filesystem fails with `SPIFFS_ERR_IS_INDEX`. I am using the following configuration:
```
#define SPIFFS_ERASE_BLOCK_SIZE  (FLASH_SECTOR_SIZE)         // use 4kB sectors to erase
#define SPIFFS_LOG_BLOCK_SIZE    (FLASH_SECTOR_SIZE * 16)  // use 64kB logical sectors
#define SPIFFS_PAGE_SIZE         (512) 
```
I suspected a bug in the HAL write function (flash page size is 256) but the page wrapping seems to be working fine :confused: 

I was hoping you could point me in the right direction, or tell me more about what could cause a `SPIFFS_ERR_IS_INDEX`. 

Thanks!

mozturk, 2016-03-03:
Bug was in the HAL write function all along :)

pellepl, 2016-03-03:
Sorry for not looking earlier, but great you found it! Cheers ;)

mozturk, 2016-03-04:
Thanks for following up!

I'm actually having a problem with optimizing spiffs, I'd love your insight
on it.

About once every 5 minutes i generate ~3500 bytes of data to be written to
the filesystem. These writes start off taking around 5-7 seconds, but
eventually jump up and start taking 60-70 seconds. It looks like the
filesystem is thrashing to me:

[image: Inline image 1]

The number of algorithm iterations it takes before the filesystem starts to
thrash is dependent on the logical page size (the output of spiffs debug
statements during one of these long writes is below). Do you have any tips
on what I could do to reliably achieve ~5-7 second write times to the
filesystem? My spiffs configuration is below. I allow for at most 8 cached
pages in the work buffer.

Let me know if there is any other data that would be useful!

Thanks!
Meric

Spiffs Configuration:
#ifndef SPIFFS_CFG_PHYS_SZ
#define SPIFFS_CFG_PHYS_SZ(ignore)        (FLASH_TOTAL_SIZE)
#endif
#ifndef SPIFFS_CFG_PHYS_ERASE_SZ
#define SPIFFS_CFG_PHYS_ERASE_SZ(ignore)  (FLASH_BLOCK64_SIZE)
#endif
#ifndef SPIFFS_CFG_PHYS_ADDR
#define SPIFFS_CFG_PHYS_ADDR(ignore)      (0)
#endif
#ifndef SPIFFS_CFG_LOG_PAGE_SZ
#define SPIFFS_CFG_LOG_PAGE_SZ(ignore)    (1024)
#endif
#ifndef SPIFFS_CFG_LOG_BLOCK_SZ
#define SPIFFS_CFG_LOG_BLOCK_SZ(ignore)   (FLASH_BLOCK64_SIZE)
#endif
#endif

Spiffs Log:
[03-03-16 17:57:07.366] modify: load objixhdr page 19a2:0000
[03-03-16 17:57:07.366] modify: store to existing data page, src:19a1,
dst:19e7:0003 offset:555, len 208, written 0
[03-03-16 17:57:07.366] modify: wrote page 19e7 to objix_hdr entry 03 in mem
[03-03-16 17:57:07.531]        callback: setting fd 13:8000005b
objix_hdr_pix to 19e8, size:3808
[03-03-16 17:57:07.531]        callback: setting fd 13:8000005b span:0000
objix_pix to 19e8
[03-03-16 17:57:07.531]        callback: setting fd 14:8000005b
objix_hdr_pix to 19e8, size:3808
[03-03-16 17:57:07.531]        callback: setting fd 14:8000005b span:0000
objix_pix to 19e8
[03-03-16 17:57:07.531] modify: store modified objix_hdr page, 19e8:0000,
written 208
[03-03-16 17:57:07.531] modify: load objixhdr page 19e8:0000
[03-03-16 17:57:07.531] modify: store to existing data page, src:199a,
dst:19e9:0000 offset:0, len 528, written 0
[03-03-16 17:57:07.688] modify: wrote page 19e9 to objix_hdr entry 00 in mem
[03-03-16 17:57:07.688]        callback: setting fd 13:8000005b
objix_hdr_pix to 19ea, size:3808
[03-03-16 17:57:07.688]        callback: setting fd 13:8000005b span:0000
objix_pix to 19ea
[03-03-16 17:57:07.688]        callback: setting fd 14:8000005b
objix_hdr_pix to 19ea, size:3808
[03-03-16 17:57:07.688]        callback: setting fd 14:8000005b span:0000
objix_pix to 19ea
[03-03-16 17:57:07.688] modify: store modified objix_hdr page, 19ea:0000,
[... 697 lines cut ...]
[03-03-16 17:58:19.191] modify: wrote page 1a34 to objix_hdr entry 00 in mem
[03-03-16 17:58:19.351]        callback: setting fd 33:800000f1
objix_hdr_pix to 1a35, size:992
[03-03-16 17:58:19.351]        callback: setting fd 33:800000f1 span:0000
objix_pix to 1a35
[03-03-16 17:58:19.351]        callback: setting fd 34:800000f1
objix_hdr_pix to 1a35, size:992
[03-03-16 17:58:19.351]        callback: setting fd 34:800000f1 span:0000
objix_pix to 1a35
[03-03-16 17:58:19.351] modify: store modified objix_hdr page, 1a35:0000,
written 32
[03-03-16 17:58:19.351] modify: load objixhdr page 1a35:0000
[03-03-16 17:58:19.351] modify: store to existing data page, src:1a34,
dst:1a36:0000 offset:800, len 96, written 0
[03-03-16 17:58:19.457] modify: wrote page 1a36 to objix_hdr entry 00 in mem
[03-03-16 17:58:19.457]        callback: setting fd 33:800000f1
objix_hdr_pix to 1a37, size:992
[03-03-16 17:58:19.566]        callback: setting fd 33:800000f1 span:0000
objix_pix to 1a37
[03-03-16 17:58:19.566]        callback: setting fd 34:800000f1
objix_hdr_pix to 1a37, size:992
[03-03-16 17:58:19.566]        callback: setting fd 34:800000f1 span:0000
objix_pix to 1a37
[03-03-16 17:58:19.566] modify: store modified objix_hdr page, 1a37:0000,
written 96

someburner, 2016-03-05:
5-7 seconds for 3500 sounds like quite a long time for a SPIFFS write. What platform is this on? I wrote a websocket uploader that chunks files into 256 bytes and uploads to SPIFFS and it only takes maybe 5 seconds to write ~100KBytes of data. This is using the ESP8266 clocked at 160 MHz, 4MB BergMicro chip in QIO mode.

mozturk, 2016-03-05:
Sorry, I mischaracterized the system. This is a low-priority task
performing a low-priority activity running inside MQX (K24 platform,
arm-eabi-gcc), so I am okay with the 5 second write (there are external
events that can preempt this data save/contend for filesystem access). The
flash chip is the N25Q26A, single SPI, 30MHz.

The 3500 byte happen every 15 minutes, and over time the write time
increases from 5 to 60 seconds. The jump from 5 to 60 seconds is immediate,
it looks like the filesystem starts thrashing, constantly updating page
headers or swapping pages or something. The thrashing starts happening
later with increased page sizes, but eventually the writes jump from 5 to
60 seconds. Maybe I need more cached pages, or need to use a separate
SPIFFS instance or something? I'm not sure where to go but I can supply
whatever info you need.

Thanks!
Meric

someburner, 2016-03-06:
I couldn't find N25Q26A with a google search. Do you have the datasheet? I noticed you have SPIFFS_CFG_LOG_PAGE_SZ set to 1024. I've always used 256 or 128.

pellepl may know what's going on, but I don't know enough of the internals to offer any advice from the output you provided. I think that it might be more helpful to see how you call SPIFFS_write and what you do before/after. Do you open the file and close? Or just leave it open?

pellepl, 2016-03-06:
Well - going from x seconds to 10x instantly is most probably due to that you start with a fully formatted system. All blocks are clean. Then you write and delete numerous times - when all blocks become dirty, a garbage collection commences, cleaning a block for new data. This takes time and is normal. I would not call it thrashing though, as it does not depend very much on how files are written in memory. Sort of anyway.

Could you measure how long time you spend in your ```hal_erase``` function? I bet it is not even called in the beginning, but when things start to go slow you'll see that much of the time lies here.

Perhaps the erase function is not optimal, perhaps you could poll the spiflash more often for busy flag, or something. Doing a bit of measuring here on where time is spent will provide good info. Perhaps you can boost your file thread prior to garbage collection.

Unfortunately, garbage collection _must_ be performed at some stage. In spiffs, garbage collection is performed on demand, i.e. when you call write and spiffs finds that there are no clean pages left.
What you can do is to call ```SPIFFS_gc``` yourself, if you know a time when the system is idle with regard to file accesses. This at least let you have control over when to do it. Calling a ```SPIFFS_gc``` prior to ```SPIFFS_write``` would, if my presumptions are correct, make the write go down to the 5 second delay instead. As the other 55 seconds would be spent in the gc call.

The time in gc is normally much spent in the ```hal_erase``` function. Depending on how full the system is (i.e. lots of written pages, few deleted pages) the gc can take more or less time. It is of course faster just to move a few written pages before erasing, compared to moving many pages and perhaps just freeing x/2 bytes, meaning spiffs will have to find another block again to gc before reaching x free bytes.

pellepl, 2016-03-06:
@someburner  "it only takes maybe 5 seconds to write ~100KBytes of data" - that sounds more normal. Do you know if this was on a clean fs or a very "cycled" one?

someburner, 2016-03-06:
@pellepl Pretty clean. I'm still debugging stuff so I format it every couple days, sometimes more if it's SPIFFS-related stuff. But I've done quite a few erases and writes of large files one after the other. I've allocated a bit under 1MB for the FS, might be interesting to try filling it up and seeing the performance.

mozturk, 2016-03-07:
Hi Peter,

Thanks for the response! The HAL erase function was my first thought too,
but it doesn't look like SPIFFS is doing any garbage collection when we
jump from 6 second writes to 60. I don't see any GC debug messages printed
and the filesystem is still too small to have cycled through all available
formatted blocks. On my latest test the "thrashing" (lack of a better term)
started about 15 hours into the test when the filesystem size was:

 Used :       582610 bytes
 Total:     32612965 bytes

Which looks too small to have needed to garbage collect. The spiffs
visualization also points to free blocks still being available. I will do
more testing of the abstraction layer and see where I'm spending the
majority of my time as you suggested.

The "thrashing" also starts sooner with smaller page sizes, though the
amount of data being generated remains the same.

Is it at all possible the slowing is a result of the way I am using the
filesystem? I have ~20 files, and I open read- and write-only handles to
each of them during boot. The file handles remain open until the system is
shut down. Most of the files grow slowly, reach a size limit, and cycle
over to the beginning of the file. There are two files which grow much
larger than the others, one of them being the file that is causing the
"thrashing" (this file also grows the fastest). However I don't know enough
about what spiffs is doing internally to know if this could be an issue or
not. I was thinking it might be better to have these files in their own
spiffs instance, but it sounds like I need to better characterize the
problem first.

Thanks!
Meric

pellepl, 2016-03-07:
Hi Meric,

whoa! You're using a 32mbyte spiflash? I think I safely can say you're on somewhat uncharted waters :)

Ok, from your description I think we can count out the GC - it does not seem to be the culprit here.

I would suggest to keep a large page size since you're having such a big flash. You were using 1024 bytes, right? I think that is a good size. If possible, you could even try larger. Regarding block size you might want to try 128K or 256K, or in that manner. Spiffs handles the case when the logical block size is n x physical_page_size, so no need for worry here.

One other thought, have you looked over the typedefs in spiffs_config.h (@ the end)? You could try setting all these to u32_t with respect to the sizes you're running.

Cache, well - you cannot have more than 32 pages anyway. Since you're keeping your files open, your cache could be set to <max_nbr_of_open_files> + 2 or something. More would not speed things up much I think.

I cannot think of much else to try right now. The way you run the fs sounds very sensible (keeping the filehandles open). This way, spiffs will keep track of the files in a better way and won't need to rescan the system for each open. So keep this implementation. 

Optimally, if there's any possibility at all to profile the app I think this would be the safest bet. It is very odd that you all of the sudden get a 10x increase in time. I have never seen this on smaller configs. Try to poke around a bit and tell me what you find. 

Good luck!

mozturk, 2016-03-07:
> keeping your files open, your cache could be set to + 2 or something. More

Thanks for the info! I am using a logical page of 1024 and a physical and
logical block size of 64kB. Maybe I should increase my logical block size?
I think I have enough work buffer right now for 8 cached pages max. I could
get away with going to a logical page size of 2048 and a logical block size
of 128K/256K I think. Based on my testing larger page sizes work better :)

I am profiling the HAL layer right now, I will let you know if I find
anything interesting.

Thanks!

pellepl, 2016-03-07:
Great, keep me posted. Again, also have a peek on those typedefs. When
being too small in number of bits, pretty much anything can happen.

mozturk, 2016-03-07:
Ahh yeah, forgot to mention I typedefed everything to u32 :)

mozturk, 2016-03-08:
Hi,

Just wanted to follow up with you on this. With page size 2048, logical
block 256K, physical block 64K the "thrashing" is completely gone (see the
bottom line in the attached graph). At the end of the test the filesystem
started garbage collecting; the write times still look way better than with
smaller page/block sizes. In the previous tests the filesystem never got
large enough to start garbage collecting.

Looks like there is some relationship between filesystem size and
page/block size that can produce this "thrashing" behavior. I still have no
idea what was going on :)

Cheers,
Meric

pellepl, 2016-03-09:
Great you got it working! Unfortunately, I'm also lost why this is
happening on other configurations. I'll try to reproduce it in the test
framework. Beware of future questions 😉

benpicco-tmp, 2016-05-18:
I have the same issue with log_page_size = 256; log_block_size = 4096; phys_erase_block = 4096; on a 8 MiB Flash 

On a fresh fs I get about 13kb/s when writing.
On a more used fs I often have files with stretches where I can only read ~100 byte every second.

quadrupling the spi frequency only yields an 18% speed increase in the best case.

benpicco-tmp, 2016-05-18:
I've looked at coffeefs and while it performs much worse in the default configuration 

```
2016-05-18 16:54:21,509 - INFO # dumping rom to file…
2016-05-18 16:54:21,631 - INFO # Coffee: Reserved 5 pages starting from 9 for file fw.bin
2016-05-18 16:55:24,613 - INFO # 33984 bytes written.
```
-> ~500 byte/s

There is an optimization when knowing the size of the file in advance. Using `cfs_coffee_reserve` yields to

```
2016-05-18 17:04:37,375 - INFO # dumping rom to file…
2016-05-18 17:04:37,684 - INFO # Coffee: Reserved 2049 pages starting from 5 for file fw.bin
2016-05-18 17:04:37,909 - INFO # 34176 bytes written.
```
-> 56kb/s

Maybe a similar optimization would be possible for SPIFFS? 

https://github.com/contiki-os/contiki/wiki/File-systems#CFS_Interface_Extensions

pellepl, 2016-05-18:
Thanks for info! Is it possible to get a dump of the fs in a "quick" state and a "slow" state?

There is something similar in SPIFFS, ```SPIFFS_gc(spiffs *fs, u32_t size)```. This will reserve the given size, or rather, free up blocks so a subsequent write won't need to gc. Though, the time won in your write is probably more or less lost in the previous gc call - mayhap with a small gain in performance. I would suspect the same goes for coffeefs though.

pellepl, 2016-06-07:
@benpicco-tmp, in the image I got from you in #90, I saw you're using a 128-byte page size in an 8 megabyte flash. This calls for using u32_t in typedefs for spiffs_page_ix, spiffs_obj_id and spiffs_span_ix in the config. Perhaps this can explain the weird read behaviour on a full fs.

pellepl, 2016-06-07:
@mozturk, I know this is very long ago, but ...
> There are two files which grow much larger than the others, one of them being the file that is causing the "thrashing" (this file also grows the fastest). 

Can you remember roughly what size that file reached when things started to get slow? I actually got a hunch here, it might be when the indirection tables starts spanning different pages. That might explain why the problem disappeared when you increased the page size, since then all indirection info on your data fits into one header page.

phlemoine, 2016-09-29:
i am facing a similar issue. 
Observations:
writing to a file takes longer as the file system nears 70% usage. that happens with a 512kB and 1024kB file system. while files seem to be written properly, my system does not get a chance to schedule the other taska and eventually resets

Configuration:
i am not resizing or deleting files, only adding up new files, so it does not look like the file system gets fragmented.
i hav tried to write chunks of 127 bytes at a time, and chunks of 512 bytes 
my flash devicec has sectors of 4kB.
i am on the tip of the branch as of sep-28.

any recommendation on what to look for or optimize writing files ?

pellepl, 2016-09-29:
@phlemoine Are you sure it isn't the garbage collector kicking in? This issue is about things getting slow before the system starts erasing blocks. Even if you're only creating and appending to files, there will be some "fragmentation" - i.e. some pages need to be deleted - as the file headers need to be updated on a file change. 

I presume this is on the ESP8266. Are you running this on an rtos of some sort? If so, it is a bit odd that you get a reset (watchdog reset?). Even if you're executing from the same spi flash you're erasing blocks on, the rtos should reschedule the file system thread to give slots to other tasks and feed the watchdog. Perhaps you could feed the watchdog in the hal layer of spiffs, e.g. before and/or after the block erase. While writing spiffs, I didn't know of the ESP so I never figured we ever would keep executable and filesystem data on the same flash :)

Then again, when spiffs gets crammed it does hog up resources when GCing, by design, unfortunately. Given the ram we have to work with, it does take time. I do have plans for optimising this when I find the time. This will however be a major rewrite of some stuff.

Sorry for not giving more concrete tips.

phlemoine, 2016-09-29:
Thanks! Any clue will help. Yes, it's on the ESP. I'm getting Timer Stop Failed or Timer Failed Stop errors, that sometimes lead to a watchdog reset-not all times tho. It's running with the freertos port from esp-open-tos

pellepl, 2017-03-02:
Been a while now - issue dead? Feel free to reopen if necessary.

closed 2017-03-02 by pellepl

phlemoine, 2017-03-02:
Issue is still there. It's probably GC running when file system is 'used'.
I workaround by reformating when it reaches 80% utilization to avoid WDT
issues. Not ideal, but I'm lacking a better option

c10ud, 2018-04-20:
Hello, I encountered the same issue with an AT25DF321A running at 4Mhz, here's my config:

#ifndef SPIFFS_CFG_PHYS_SZ
#define SPIFFS_CFG_PHYS_SZ(ignore)        (64 * 64 * 1024)
#endif
#ifndef SPIFFS_CFG_PHYS_ERASE_SZ
#define SPIFFS_CFG_PHYS_ERASE_SZ(ignore)  (4 * 1024)
#endif
#ifndef SPIFFS_CFG_PHYS_ADDR
#define SPIFFS_CFG_PHYS_ADDR(ignore)      (0)
#endif
#ifndef SPIFFS_CFG_LOG_PAGE_SZ
#define SPIFFS_CFG_LOG_PAGE_SZ(ignore)    (256)
#endif
#ifndef SPIFFS_CFG_LOG_BLOCK_SZ
#define SPIFFS_CFG_LOG_BLOCK_SZ(ignore)   (4 * 1024)
#endif
#endif

Do you think increasing log page size to 1024 or so would help?