pellepl/spiffs #74: Duplicate file names

igrr, 2016-03-07:
Hi Peter,
One of our users has reported an issue with file name duplication (https://github.com/esp8266/Arduino/issues/1685). The sample posted there is pretty long, but it boils down to the following sequence of actions (pseudo-code):
```
    createFileWithContent("/config", "some text");
    createFileWithContent("/data", "");
    readFile("/config");
    openFile("/data");
    listFiles("/");
```
Expected behavior is that we have two files, `/config` and `/data`. However, directory listing contains two files called `/config`.

Here is the test case which reproduces the behaviour without relying on C++ wrapper used in our project.
```c++
TEST(spiffs_dup_file) {
    fs_reset_specific(0, 0, 64*1024, 4096, 4096*2, 256);
    {
        spiffs_file fd = SPIFFS_open(FS, "/config", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_WRONLY, 0);
        TEST_CHECK(fd >= 0);
        char buf[5];
        strncpy(buf, "test", sizeof(buf));
        SPIFFS_write(FS, fd, buf, 4);
        SPIFFS_close(FS, fd);
    }
    {
        spiffs_file fd = SPIFFS_open(FS, "/data", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_WRONLY, 0);
        TEST_CHECK(fd >= 0);
        SPIFFS_close(FS, fd);
    }
    {
        spiffs_file fd = SPIFFS_open(FS, "/config", SPIFFS_RDONLY, 0);
        TEST_CHECK(fd >= 0);
        char buf[5];
        int cb = SPIFFS_read(FS, fd, buf, sizeof(buf));
        TEST_CHECK(cb > 0 && cb < sizeof(buf));
        TEST_CHECK(strncmp("test", buf, cb) == 0);
        SPIFFS_close(FS, fd);
    }
    {
        spiffs_file fd = SPIFFS_open(FS, "/data", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_WRONLY, 0);
        TEST_CHECK(fd >= 0);
        spiffs_stat stat;
        SPIFFS_fstat(FS, fd, &stat);
        if (strcmp((const char*) stat.name, "/data") != 0) {
            // oops! lets check the list of files...
            spiffs_DIR dir;
            SPIFFS_opendir(FS, "/", &dir);
            struct spiffs_dirent dirent;
            while (SPIFFS_readdir(&dir, &dirent)) {
                printf("%s\n", dirent.name);
            }
            // this will print "/config" two times
            TEST_CHECK(0);
        }
        SPIFFS_close(FS, fd);
    }
    return TEST_RES_OK;
} TEST_END(spiffs_dup_file)
```

Test output follows:
```
[igrokhotkov@igrokhot-mbp ~/e/spiffs]$ ./build/linux_spiffs_test -f spiffs_dup_file
adding suites...
adding test spiffs_dup_file
1 tests added
running tests...
TEST 1/1 : running test spiffs_dup_file
/config
/config
  TEST FAIL src/test/test_bugreports_my.c:66
  free blocks     : 7 of 8
  pages allocated : 3
  pages deleted   : 1
  gc runs         : 0
  cache hits      : 48 (sum 48)
  cache misses    : 28 (sum 28)
  cache utiliz    : 0.631579
  RD:         21 reads        5165 bytes        245 avg bytes/read
  WR:         13 writes        623 bytes         47 avg bytes/write
  fs consistency check:
   check: LOOKUP FIX INDEX 8001:0000
   check: LOOKUP DELETE PAGE 0004
  .. FAILED
  spiffs errno:-10072
   0 id//i__________________________	era_cnt: 0
   1 _______________________________	era_cnt: 0
   2 _______________________________	era_cnt: 0
   3 _______________________________	era_cnt: 0
   4 _______________________________	era_cnt: 0
   5 _______________________________	era_cnt: 0
   6 _______________________________	era_cnt: 0
   7 _______________________________	era_cnt: 0
era_cnt_max: 1
last_errno:  -10072
blocks:      8
free_blocks: 7
page_alloc:  3
page_delet:  2
used:        753 of 46937
ran 1 tests
Test report, 1 tests
0 succeeded
1 failed
  spiffs_dup_file
0 stopped

FAILED
```

I might not be doing sufficient error checking here. I noticed that at some point while this test is being executed, FS->err_code equals SPIFFS_ERR_DELETED. I haven't found documentation about this error code, and i couldn't infer what it means from the test cases (none of the test cases seem to check for it).

pellepl, 2016-03-07:
Oh, that's not good. I'll look into it as soon as I can, a bit busy this week though. @igrr, you do get a star for the best bug report so far, billions of thanks for taking the trouble of writing a testcase for me. This helps a lot! 

The SPIFFS_ERR_DELETED is simply that we're trying to open a file that is deleted. Or really a page that has been deleted. This is very odd. Regarding the error checking, well yeah, one _should_ check a bit more, but no-one does. The test-case does not do anything complicated really. I'll have to adapt to reality ;)

pellepl, 2016-03-07:
Thanks again Ivan for reporting this. It was a sneaky one, will only happen when opening a only-created-never-written file with SPIFFS_TRUNC flag. Fix coming up.

closed 2016-03-07 by pellepl

igrr, 2016-03-10:
Thanks for the quick fix!