pellepl/spiffs #75: Need help

Ondreyk, 2016-03-13:
Can you help me? I think I find flash memory leaks. Вut can't find the cause.
I see growing count of "page_delet" when I use open and close one file (one file name) several times with. I use "w" flag option for opening. Not "w+". "used" parametr is stable. What should be the conversion process from "page_delet" to "free_blocks"? Why garbage collector do nothing? 
```
era_cnt_max: 1
last_errno:  0
blocks:      136
free_blocks: 136
page_alloc:  0
page_delet:  0
used:        0 of 504761

era_cnt_max: 1
last_errno:  0
blocks:      136
free_blocks: 135
page_alloc:  2
page_delet:  5
used:        502 of 504761

era_cnt_max: 1
last_errno:  0
blocks:      136
free_blocks: 135
page_alloc:  2
page_delet:  12
used:        502 of 504761

era_cnt_max: 1
last_errno:  0
blocks:      136
free_blocks: 134
page_alloc:  2
page_delet:  19
used:        502 of 504761
```

Ondreyk, 2016-03-13:
Project - fork from nodemcu-firmware
Device - esp8266 - esp-12e

pellepl, 2016-03-14:
Hi there,

well - it looks like you're opening the file, write something on the same offset, and then close it. Repeatedly. Am I correct?

If so, everything looks ok. You can regard deleted pages as free space in a way. The garbage collector kicks in when all blocks - 2 are dirty.

Ondreyk, 2016-03-14:
Yes. Thanks! But i have exception when free_blocks < 3. And spiffs falls in the next.

pellepl, 2016-03-14:
Ok. Where in the spiffs stack do you get the exception?

Ondreyk, 2016-03-14:
It happened when i try open any file ("w" or "w+" - it does not matter). 

(More detailed information I can get for you in the evening (sorry)).

pellepl, 2016-03-14:
Ok, my gut feeling tells me there's something wrong in your erase_hal function. If you try to write, and the system fully dirty (i.e. all blocks minus two), spiffs will try to GC. From your description I think you have a problem in the erase function, as it seems it is this one that make it crash. Check the implementation. If I recall correctly, the ESP executes from the same flash that you're running spiffs on, so double checking addresses cannot hurt. Also, it might be the watchdog that kicks in. As we're executing from flash, and at the same time erase it, I'd gather the ESP will be fully stalled until erases are finished , meaning no one will caress the dog.

Ondreyk, 2016-03-14:
I found your solution: https://github.com/pellepl/spiffs/commit/7c5493732da5b43e09fa4d81be6c8a713c2be79c

Thanks!!!

pellepl, 2016-03-14:
> I found your solution: 7c54937

Ah great! Never run old versions :)

Cheers / Peter

Ondreyk, 2016-03-14:
Confirm for pull request: https://github.com/nodemcu/nodemcu-firmware/issues/1147

closed 2016-03-14 by Ondreyk