pellepl/spiffs #76: SPIFFS is not working on my target

kikuVT, 2016-03-14:
Hi,
I started exploring SPIFFS for the target based on PIC32MZ processor interfaced with SQI-Flash(serial NOR Flash, nibble wide interface) SST26V0F032B.

I am running into issue with SPIFFS usage. Here is snapshot of where things are
1. I have integrated SPIFFS into my simple project which is meant to validate SPIFFS on my target.
2. SQI-Flash driver has been validated and its functions(read, write and erase) are tied to SPIFFS.
3. I used the example code from this GitHub.  Modified the 'spiffs_sqi_wrapper.c' , see code below.
4. I see successful, mounting of file-system based on its return value, yet, I cannot open a new file; fd(file descriptior) from SPIFFS_open() is negative number. 

Stepping into code through debugger, I see that failure is happening in spiffs_nucleus.c file and in function spiffs_object_open_by_page() and at line, 
SPIFFS_VALIDATE_OBJIX(oix_hdr.p_hdr, fd->obj_id, 0);

During this validation, it is failing w/ SPIFFS_ERR_NOT_INDEX 

Any timely help to resolve this issue?
============================== spiffs_sqi_wrapper.c================================

[spiffs_sqi_wrapper.txt](https://github.com/pellepl/spiffs/files/171923/spiffs_sqi_wrapper.txt)

pellepl, 2016-03-14:
Hmm, that error indicates that you read something unexpected from the fs. Which may meaning
1. read is not working
2. write is not working
3. erase is not working

Or any combination of above. Your configuration should pose no problems, so my bet would be the hal layer. You might want to printf what you write and what you read. The pic32 is a new platform as far as i know, it could be some alignment thing also. Make 100% sure the hal is ok, then we take it from there.

Cheers

pellepl, 2016-03-15:
Btw, come think of it, did you try enabling SPIFFS_ALIGNED_OBJECT_INDEX_TABLES?

kikuVT, 2016-03-16:
It was not before, now I enabled it and , at least, I see successful file create, write and read operation !!. This is huge considering I have spent so much hours chasing the issue.  Thanks for your quick response and fix.

I did read about SPIFFS_ALIGNED_OBJECT_INDEX_TABLES in config wiki but could you elaborate more on why this matters in my case ?

pellepl, 2016-03-16:
Great you got it working!
Well, it seems the PIC32 cannot access memory unaligned. The file header comprises a struct containing file name, flags, length, etc. This size of this struct can become an uneven number of bytes depending on the configuration. After the header, a first-level indirection table follows. If the header size is uneven, this table will be unaligned. Meaning, you processor will try to read e.g. 16-bit numbers from a table being at an uneven address. Normally, you'd get a bus error here, but it seems the PIC allows it and instead returns something that spiffs do not recognize.

Spiffs was originally constructed for cortex processors - these can access memory unaligned, if configured so. Then came all ESP8266 users, whose processor cannot - hence came the SPIFFS_ALIGNED_OBJECT_INDEX_TABLES flag.

Cheers!

closed 2016-03-19 by pellepl