pellepl/spiffs #79: Problem with file names that are long

pjsg, 2016-03-17:
If the length of the filename is equal to the SPIFFS_OBJ_NAME_LEN then there appear to be problems. It looks as though there are several strcpy/strcmp calls in the code -- I suspect that they should all be bounded. This test case just verifies that directory listings return the right thing. The alternative is that the null should be included in the SPIFFS_OBJ_NAME_LEN size -- in which case, the open of the longest file should fail. 

This is a very simple test case:
```
TEST(long_filename_75) {
  fs_reset();

  int len;
  char c;
  int limit = SPIFFS_OBJ_NAME_LEN;
  for (c = 'A'; c <= 'D'; c++) {
      for (len = 1; len <= limit; len++) {
          char namebuff[SPIFFS_OBJ_NAME_LEN + 3];
          memset(namebuff, 0, sizeof(namebuff));
          memset(namebuff, c, len);

          spiffs_file fd = SPIFFS_open(FS, namebuff, SPIFFS_RDWR | SPIFFS_CREAT, 0);
          TEST_CHECK_GE(fd, 0);

          int res = SPIFFS_close(FS, fd);
          TEST_CHECK_GE(res, 0);
      }
  }

  spiffs_DIR d;
  struct spiffs_dirent e;
  struct spiffs_dirent *pe = &e;

  SPIFFS_opendir(FS, "/", &d);

  unsigned char counts[SPIFFS_OBJ_NAME_LEN + 3];
  memset(counts, 0, sizeof(counts));

  while ((pe = SPIFFS_readdir(&d, pe))) {
    counts[strlen(pe->name)]++;
  }
  SPIFFS_closedir(&d);

  for (len = 1; len <= limit; len++) {
    TEST_CHECK(counts[len] == 4);
  }
  TEST_CHECK(counts[limit+1] == 0);

  return TEST_RES_OK;
} TEST_END(long_filename_75)
```

pellepl, 2016-03-17:
Yes, this is an old sin of mine, that I've been too lazy to fix. I left it
to the users to make sure they did not have names longer than MAX-1.

I'll look into it.

pjsg, 2016-03-17:
The back story is that I want to move nodemcu-firmware onto the stock spiffs code with none of our changes in it. This is (AFAIK) the last thing that we changed..... There were a number of alignment related things, but you seemed to have fixed these. Given that you are also running on an ESP module, you will have faced all of those problems.

pellepl, 2016-03-17:
Yes, bringing up your version to latest would solve a bunch of problems. Saw in you issue tracking you've had a great deal of trouble wrt spiffs - sorry for that. 

I just got started on my ESP project here, so you perhaps shouldn't rely on my pretty shallow experience with spiffs on ESP. So far no problems though. The only alignment fixes I can think of is #10 (config flag ```SPIFFS_ALIGNED_OBJECT_INDEX_TABLES```) and #29. Does this correlate with your fixes?

As for the ```SPIFFS_OBJ_NAME_LEN``` related fix, this config will be commented to indicate that this length includes the zero. Plus changing ```strcmp``` and ```strcpy``` to their size limited variants. Unfortunately, I cannot add the extra zero-byte in the patch, as that would move the indirection table in file headers, making previous versions incompatible.

pjsg, 2016-03-17:
Can you reject filenames that are too long (in open)?

pellepl, 2016-03-17:
Ah yes of course. That'd be more posix aligned. Good idea. Thanks!

pellepl, 2016-03-18:
something like that?

pjsg, 2016-03-19:
Yes -- just like that!

pellepl, 2016-03-19:
Great! Cheers for all input.

closed 2016-03-19 by pellepl