pellepl/spiffs #81: Minor tweaks

Nails-, 2016-03-24:
For these macros:
#define SPIFFS_API_CHECK_MOUNT(fs) \
  if (!SPIFFS_CHECK_MOUNT((fs))) { \
    (fs)->err_code = SPIFFS_ERR_NOT_MOUNTED; \
    return -1; \
  }

#define SPIFFS_API_CHECK_CFG(fs) \
  if (!SPIFFS_CHECK_CFG((fs))) { \
    (fs)->err_code = SPIFFS_ERR_NOT_CONFIGURED; \
    return -1; \
  }

#define SPIFFS_API_CHECK_RES(fs, res) \
  if ((res) < SPIFFS_OK) { \
    (fs)->err_code = (res); \
    return -1; \
  }

#define SPIFFS_API_CHECK_RES_UNLOCK(fs, res) \
  if ((res) < SPIFFS_OK) { \
    (fs)->err_code = (res); \
    SPIFFS_UNLOCK(fs); \
    return -1; \
  }

Instead of returning -1, can you please change this to return the actual error codes instead of -1? The issue that might arise is that you may not get the correct error code if you have to rely upon the SPIFFS_errno function in a multi-threaded environment.  For example, when one thread encounters some error on a spiffs filesystem, but gets switched out for another thread that uses SPIFFS with the same filesystem, and it then encounters some other type of error.  When the original thread returns and reads the error_code, won't it get the second thread's error number?

I am proposing a simple change to these for functions to return these values:
SPIFFS_API_CHECK_MOUNT   --  return SPIFFS_ERR_NOT_MOUNTED;
SPIFFS_API_CHECK_CFG -- return SPIFFS_ERR_NOT_CONFIGURED;
 SPIFFS_API_CHECK_RES -- return res;
SPIFFS_API_CHECK_RES_UNLOCK -- return res;

Also my compiler issues annoying remarks (but still compiles fine) that could easily be fixed.  They are:
spiffs-master/src/spiffs.h", line 107: remark #230-D: trailing comma is nonstandard                                                               
spiffs-master/src/spiffs.h", line 126: remark #230-D: trailing comma is nonstandard                                                               
spiffs-master/src/spiffs_cache.c", line 17: remark #270-D: declaration may not appear after executable statement in block                         
spiffs-master/src/spiffs_cache.c", line 69: remark #270-D: declaration may not appear after executable statement in block                         
spiffs-master/src/spiffs_cache.c", line 70: remark #270-D: declaration may not appear after executable statement in block                         
spiffs-master/src/spiffs_cache.c", line 71: remark #270-D: declaration may not appear after executable statement in block                         
spiffs-master/src/spiffs_cache.c", line 95: remark #270-D: declaration may not appear after executable statement in block                         
spiffs-master/src/spiffs_cache.c", line 128: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 129: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 130: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 151: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 159: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 173: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 174: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 175: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 188: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 216: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 235: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 250: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 251: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 268: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 269: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 270: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 271: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 280: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_cache.c", line 289: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 62: remark #270-D: declaration may not appear after executable statement in block                         
spiffs-master/src/spiffs_check.c", line 104: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 117: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 164: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 177: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 178: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 179: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 180: remark #270-D: declaration may not appear after executable statement in block                        
spiffs-master/src/spiffs_check.c", line 314: remark #270-D: declaration may not appear after executable statement in block                        
[... 166 lines cut ...]
spiffs-master/src/spiffs_nucleus.c", line 1246: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1312: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1313: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1398: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1435: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1436: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1437: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1509: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1510: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1511: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1512: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1513: remark #270-D: declaration may not appear after executable statement in block                     
[... cut ...]
spiffs-master/src/spiffs_nucleus.c", line 1986: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1987: remark #270-D: declaration may not appear after executable statement in block                     
spiffs-master/src/spiffs_nucleus.c", line 1999: remark #270-D: declaration may not appear after executable statement in block                     

Thanks                                                                                                                                                  
James Lockwood

pellepl, 2016-03-24:
Hi James,

regarding return codes, spiffs tries to be posix compliant. Though the errnos themselves are not. In any case, I do agree with you. Returning ```-x``` instead of ```-1``` where ```-x``` is the proper error code should not make a big difference as I've never seen anything else then ```if (SPIFFS_something(..) < 0) error```, and it allows for some smaller code. I personally find checking errno something of a nuisance - I'll look into it.

As for the huge list of warnings, I must admit I'm not jumping of joy of fixing all that ;). I guess this is not gcc or clang? Most of them seem to be ```declaration may not appear after executable statement in block```. But declaring variables anywhere in scope is C99 standard, and should be ok on most modern compilers. Yes, it may be regarded as sloppy coding, but if so then I am a sloppy coder. I would not dare to think of what would happen if I passed the spiffs code to Lint...

And of course I understand, having your compiler puking at you this way each time you build cannot be fun. Isn't there any way to disable those warnings? I could probable fix the non 270-D:s though.

Nails-, 2016-03-24:
Peter,

Thanks for such a quick response. The compiler is from TI, for a Tiva chip (Arm M4F).

Just the non-270-D's would be the most helpful.  Once the library is built it doesn't have to do it anymore, but any other code that needs to import the .h header file to use spiffs would have thrown the nag warnings.

Thanks!

I'd clean up all the others if you want me to.

pellepl, 2016-04-06:
Sorry for taking such time, busy here. See #83. now at least the headers should not puke at you.

pellepl, 2016-04-06:
And for the return codes: #84

pellepl, 2016-04-06:
> I'd clean up all the others if you want me to.

Thanks, but I actually prefer having it as is. It is sort of part of what I think makes the code look pretty. But thanks for the offer!

closed 2016-04-06 by pellepl