pellepl/spiffs #82: Unable to recover from "filesystem full"

rojer, 2016-03-31:
I filled up my fs (while testing) and got into a situation where i cannot free it: even `remove` returns `SPIFFS_ERR_FULL`. opening with truncate doesn't work either.
so... what to do in this case? :)

pellepl, 2016-03-31:
That cannot happen(tm) :)

Darn, that's not good. Is this the latest version? What does SPIFFS_info
tell you? Is it possible to get a SPIFFS_vis dump? Aaand, is it possible to
get a full fs dump? Oh, and is this on the ESP8266? Any clues on how you
wound up in this situation? I'll stop asking now, sorry.

rojer, 2016-03-31:
it was on ESP8266, i was testing file upload error handling code - file system full in particular.
it actually worked a couple times - file would be deleted, free up space, i'd do it again... 
but then file cleanup (deletion) began to fail, and here we are.

below if `SPIFFS_vis`
[fz.dump.zip](https://github.com/pellepl/spiffs/files/197953/fz.dump.zip)

 output, followed by file list (opendir/readdir*N/closedir) and then an attempt to delete a particular file ("v7.gcno") which returns -10001.
file system image attached.

```$ ./unspiffs -l -v fs.dump 
   0 /ddiddddiddddid    era_cnt: 306
   1 ddddddddddddddd    era_cnt: 0
   2 idd/dddd/ddddid    era_cnt: 296
   3 d/dddddddddd/di    era_cnt: 294
   4 /dddddd/ddddidd    era_cnt: 297
   5 ddd/dddddd/dddd    era_cnt: 267
   6 ddddddddddddddd    era_cnt: 0
   7 dddd/ddddddddd/    era_cnt: 272
   8 dddd/dddd/ddidd    era_cnt: 295
   9 ddd/dddddd/dddd    era_cnt: 273
  10 dd/dddddd/ddddd    era_cnt: 275
  11 d/dd/dddddddddd    era_cnt: 277
  12 /dddidddd/ddddd    era_cnt: 298
  13 d/dd/dddddddddd    era_cnt: 279
  14 _______________    era_cnt: 308
  15 d/dd/ddddddddid    era_cnt: 300
  16 /ddddddd/dddddd    era_cnt: 283
  17 /diddd/didddddd    era_cnt: 284
  18 dddddd/dddddddd    era_cnt: 285
  19 dddd/ddddddddd/    era_cnt: 286
  20 d/dddddddd/dddd    era_cnt: 301
  21 dddddddd/dddddd    era_cnt: 288
  22 ddd/dddd/dddddd    era_cnt: 289
  23 i______________    era_cnt: 307
  24 /dddd/dddd/dddd    era_cnt: 305
  25 d/dddddddddddd/    era_cnt: 274
  26 dddd/ddd/ddddd/    era_cnt: 292
  27 dd/ddd/ddd/ddid    era_cnt: 293
  28 dd/ddddddd/dddd    era_cnt: 268
  29 d/ddddddddddd/d    era_cnt: 265
  30 dddd/dddddddd/d    era_cnt: 276
  31 /dddddddddddd/d    era_cnt: 266
era_cnt_max: 309
last_errno:  0
blocks:      32
free_blocks: 1
page_alloc:  395
page_delet:  56
used:        99145 of 113201
test.mk
index.html
conf_sys_defaults.json
demo.js
conf_sys_schema.json
sys_init.js
I2C.js
conflib.js
conf_app_schema.json
conf_app_defaults.json
v7.gcno
conf.json

Remove v7.gcno: -1 -10001
```

rojer, 2016-03-31:
[fz.dump.zip](https://github.com/pellepl/spiffs/files/197955/fz.dump.zip)

pellepl, 2016-03-31:
Thanks for all info! Finally, what is the config of your system (page size, fs size, log block size etc)?  The system looks really crammed from what I can see, you're in "panic mode" - less than two blocks free. Did you interrupt the gc somehow, crash or power failure? Any case, spiffs should handle this. I'll look into it asap when I have your conf.
Cheers!

pellepl, 2016-03-31:
Looks like 128kB size, page size 256 bytes, block size 4096 bytes. Correct?

rojer, 2016-03-31:
sorry - yes, 128K / 4K / 256 are the params (side note - would be great to have them stored somehow, carrying this metadata along is inconvenient).

the FS is crammed, that's true. there are a few files there that have not changed, and i filled it up with "v7.gcno", which is a big file that i knew would not fit.
as i said, i was testing my write error handling code, and i thought that the easiest way would be to fill up the fs :)
so yes, FS is crammed, it is supposed to be. the problem is that v7.gcno cannot be deleted or truncated to free space.

pellepl, 2016-03-31:
As for your side note - yes, that'd be wonderful. I have no idea where to put it though, in some cases we cannot even fit the magic. Planning to do some kind of dump function or something in the future.

pellepl, 2016-03-31:
Ok, try this. Now you should be able to remove the file. Cheers for finding this!

rojer, 2016-03-31:
indeed i can, thank you!
does it only apply to remove? truncation will not work?

pellepl, 2016-03-31:
No, unfortunately not. For trunc, you need a free page for the file header
with the new size, and in your image you're all out of free pages. I got
some ideas though, it might happen.

closed 2016-04-01 by pellepl