pellepl/spiffs #85: Write operations randomly stops working

arjun789, 2016-04-18:
All write operations (delete, create new file) sometimes stops working (returns -10002). I can still list the files and read them. Reformatting seems to be the only way to fix it.

pellepl, 2016-04-18:
Nothing I've heard of before. What platform is it? Is it latest spiffs version? Can you provide your config and a spiffs_vis output?

arjun789, 2016-04-19:
>What platform is it

Mbed on k22f

>Is it latest spiffs version

Either 0.3.4 or master. Don't remember which one i used

arjun789, 2016-04-19:
`#include <iostream>

#include "hal.h"

#include "mbed.h"
#include "spiffs.h"

int main()
{
        Serial pc(USBTX, USBRX);
        pc.baud(115200);

        spiffs fs;
        u8_t spiffs_work_buf[SPIFFS_CFG_LOG_PAGE_SZ(0) * 2];
        u8_t spiffs_fds[32 * 4];
        u8_t spiffs_cache_buf[(SPIFFS_CFG_LOG_PAGE_SZ(0) + 32) * 4];

        spiffs_config config;

        config.hal_read_f = spiRead;
        config.hal_write_f = spiWrite;
        config.hal_erase_f = spiErase;

        const auto mountResult = SPIFFS_mount(&fs, &config, spiffs_work_buf,
                                              spiffs_fds, sizeof(spiffs_fds),
                                              spiffs_cache_buf, sizeof(spiffs_cache_buf), 0);
        if (mountResult != 0) {
                std::cerr << "Failed to mount";
                return EXIT_FAILURE;
        }

        uint32_t total, used;
        SPIFFS_info(&fs, &total, &used);

        std::cout << "Total: " << total << ", used: " << used << std::endl;

        const auto fd1 = SPIFFS_open(&fs, "my_file", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);

        std::cout << "Create: " << fd1 << " " << fs.err_code << std::endl;

        spiffs_DIR d;
        struct spiffs_dirent e;
        struct spiffs_dirent *pe = &e;

        SPIFFS_opendir(&fs, "/", &d);

        int count = 0;
        unsigned long totalSize = 0;
	
        while ((pe = SPIFFS_readdir(&d, pe))) {
                ++count;
                totalSize += pe->size;
                std::cout << pe->obj_id << " " << SPIFFS_remove(&fs, reinterpret_cast<char *>(pe->name)) << " " << fs.err_code << std::endl;
        }
        SPIFFS_closedir(&d);

        std::cout << "Found " << count << " files using up " << totalSize << " bytes\n";
        
        SPIFFS_vis(&fs);
}
`

arjun789, 2016-04-19:
https://gist.github.com/arjun789/935e33905c557c6cfbdbdfa2ae3c7eb1

https://gist.github.com/arjun789/fdf0a57af9578b843e2f0c854da863fa

pellepl, 2016-04-19:
Thanks! Hmm, this is very odd. I cannot see anything obviously wrong with
the structure. SPIFFS_open should not return SPIFFS_ERR_NOT_FOUND.

Is it possible for you to give me a full spi flash dump of the spiffs area,
and the spiffs_config.h file?

I need to debug it here to see why things behave this way.

arjun789, 2016-04-25:
I think i've fixed the issue. I should have used u32_t instead of u16_t for `spiffs_page_ix;` But according to the documentation 

> // Page index type. Make sure the size of this type can hold
>// the highest page number of all pages - i.e. spiffs_file_system_size / log_page_size

For a flash of size 1MiB and logical page size of 256, shouldn't `u16_t` suffice?

pellepl, 2016-04-25:
Hi,

Good - but wierd. Yes 16 bits should suffice, well enough. Perhaps it is
some alignment thing that's fixed indirectly by setting page types to 32
bit. I am running almost same setup as you (1 Meg flash, 64 k blocks, 256
byte pages) without problems.

pellepl, 2016-04-25:
Btw what is your block size?
Den 25 apr 2016 17:25 skrev "Peter Andersson" <pelleplutt1976@gmail.com>:

> Hi,
>
> Good - but wierd. Yes 16 bits should suffice, well enough. Perhaps it is
> some alignment thing that's fixed indirectly by setting page types to 32
> bit. I am running almost same setup as you (1 Meg flash, 64 k blocks, 256
> byte pages) without problems.
>

arjun789, 2016-04-26:
>#define FLASH_ERASE_SIZE (32 * 1024)
#define FLASH_CHIP_CAPACITY (1 * 1024 * 1024)
#define FLASH_PAGE_SIZE 256

> #ifndef SPIFFS_BUFFER_HELP
>#define SPIFFS_BUFFER_HELP 0
>#endif

>#ifndef SPIFFS_CACHE
#define SPIFFS_CACHE 1
#endif

>#if SPIFFS_CACHE
#ifndef SPIFFS_CACHE_WR
#define SPIFFS_CACHE_WR 1
#endif

>#ifndef SPIFFS_CACHE_STATS
#define SPIFFS_CACHE_STATS 1
#endif
#endif

>#ifndef SPIFFS_PAGE_CHECK
#define SPIFFS_PAGE_CHECK 1
#endif

>#ifndef SPIFFS_GC_MAX_RUNS
#define SPIFFS_GC_MAX_RUNS 5
#endif

>#ifndef SPIFFS_GC_STATS
#define SPIFFS_GC_STATS 1
#endif

>#ifndef SPIFFS_GC_HEUR_W_DELET
#define SPIFFS_GC_HEUR_W_DELET (5)
#endif

>#ifndef SPIFFS_GC_HEUR_W_USED
#define SPIFFS_GC_HEUR_W_USED (-1)
#endif

>#ifndef SPIFFS_GC_HEUR_W_ERASE_AGE
#define SPIFFS_GC_HEUR_W_ERASE_AGE (50)
#endif

>#ifndef SPIFFS_OBJ_NAME_LEN
#define SPIFFS_OBJ_NAME_LEN (32)
#endif

>#ifndef SPIFFS_COPY_BUFFER_STACK
#define SPIFFS_COPY_BUFFER_STACK (256)
#endif

>#ifndef SPIFFS_USE_MAGIC
#define SPIFFS_USE_MAGIC (1)
#endif

>#if SPIFFS_USE_MAGIC
#ifndef SPIFFS_USE_MAGIC_LENGTH
#define SPIFFS_USE_MAGIC_LENGTH (0)
#endif
#endif

>#ifndef SPIFFS_LOCK
#define SPIFFS_LOCK(fs)
#endif
#ifndef SPIFFS_UNLOCK
#define SPIFFS_UNLOCK(fs)
#endif

>#ifndef SPIFFS_SINGLETON
#define SPIFFS_SINGLETON 1
#endif

>#if SPIFFS_SINGLETON

>#ifndef SPIFFS_CFG_PHYS_SZ
#define SPIFFS_CFG_PHYS_SZ(ignore) (FLASH_CHIP_CAPACITY)
#endif
#ifndef SPIFFS_CFG_PHYS_ERASE_SZ
#define SPIFFS_CFG_PHYS_ERASE_SZ(ignore) (FLASH_ERASE_SIZE)
#endif
#ifndef SPIFFS_CFG_PHYS_ADDR
#define SPIFFS_CFG_PHYS_ADDR(ignore) (0)
#endif
#ifndef SPIFFS_CFG_LOG_PAGE_SZ
#define SPIFFS_CFG_LOG_PAGE_SZ(ignore) (FLASH_PAGE_SIZE)
#endif
#ifndef SPIFFS_CFG_LOG_BLOCK_SZ
#define SPIFFS_CFG_LOG_BLOCK_SZ(ignore) (FLASH_ERASE_SIZE)
#endif
#endif

>#ifndef SPIFFS_ALIGNED_OBJECT_INDEX_TABLES
#define SPIFFS_ALIGNED_OBJECT_INDEX_TABLES 0
#endif

>#ifndef SPIFFS_HAL_CALLBACK_EXTRA
#define SPIFFS_HAL_CALLBACK_EXTRA 0
#endif

>#ifndef SPIFFS_FILEHDL_OFFSET
#define SPIFFS_FILEHDL_OFFSET 0
#endif

>#ifndef SPIFFS_READ_ONLY
#define SPIFFS_READ_ONLY 0
#endif

>#ifndef SPIFFS_TEST_VISUALISATION
#define SPIFFS_TEST_VISUALISATION 1
#endif
#if SPIFFS_TEST_VISUALISATION
#ifndef spiffs_printf
#define spiffs_printf(...) printf(__VA_ARGS__)
#endif
#ifndef SPIFFS_TEST_VIS_FREE_STR
#define SPIFFS_TEST_VIS_FREE_STR "_"
#endif
#ifndef SPIFFS_TEST_VIS_DELE_STR
#define SPIFFS_TEST_VIS_DELE_STR "/"
#endif
#ifndef SPIFFS_TEST_VIS_INDX_STR
#define SPIFFS_TEST_VIS_INDX_STR(id) "i"
#endif
#ifndef SPIFFS_TEST_VIS_DATA_STR
#define SPIFFS_TEST_VIS_DATA_STR(id) "d"
#endif
#endif

>typedef int32_t s32_t;
typedef uint32_t u32_t;
typedef int16_t s16_t;
typedef uint16_t u16_t;
typedef int8_t s8_t;
typedef uint8_t u8_t;

>typedef u16_t spiffs_block_ix;

>typedef u32_t spiffs_page_ix;

>typedef u32_t spiffs_obj_id;

>typedef u16_t spiffs_span_ix;

pellepl, 2016-04-26:
Aha! If you change spiffs_page_ix back to u16_t and actually check the
return code of SPIFFS_mount, you'll see that you have a configuration where
SPIFFS_MAGIC is not possible, and SPIFFS_mount throws a -10028
(SPIFFS_ERR_MAGIC_NOT_POSSIBLE) back at you.

I advise you to check the return codes - though I am a sinner in this sense
also :)

To remedy, you could for instance halve your page size or double your
logical block size.

Cheers / P

arjun789, 2016-04-27:
>SPIFFS_mount throws a -10028
No it doesn't. I do check the error code for everything. It is returning [zero](https://github.com/pellepl/spiffs/issues/85#issuecomment-211857520)

pellepl, 2016-04-27:
Ok, I'm at total loss then. My only guesses for the initial problem could be compiler things, multithreading without proper locks, stack corrupt ion or something else evil. Sorry.

arjun789, 2016-04-27:
>something else evil

Most likely.

pellepl, 2016-04-27:
😄

closed 2016-04-28 by pellepl