pellepl/spiffs #86: build/linux_spiffs_test fails (segfault) on first test

dpruessner, 2016-04-21:
I'm using gcc version 5.1.0 on Linux64 (arch).  Just running `make; and make test` fails with 

```
adding suites...
adding test evil_write
[...]
72 tests added
running tests...
fish: “./linux_spiffs_test” terminated by signal SIGSEGV (Address boundary error)
```

Looks like the `SUITE(check_tests)` macro creates a function, then the subsequent `TEST(...)` macros create nested functions.  However, it looks like pointers to these nested functions are assigned into a global variable.  As far as I can tell, these function pointers are accessed after the call to `__suite_check_tests(...)` is called.  I'm trying to piece together how this could ever run, given how I understand GCC creates nested functions with trampolines.

But from other comments, this test suite clearly works for other folks.  What toolchain successfully builds the test suite?

Thanks,
Daniel

pellepl, 2016-04-21:
Hi, on gcc 4.8.4 things work. But yes, trying on my newly installed laptop, about the same setup as yours, with gcc 5.2.1, I get  an ```Illegal instruction (core dumped)```.

To be honest, that test framework is a hack I made many years ago. As you say, it depends on pointing out nested functions. I never payed much attention to what gcc actually emitted in forms of assembly (had no insight in the nested-func trampolines) as it "just-seemed-to-work"(tm). I probably assumed that a pointer to a nested function is the same as a pointer to an ordinary function.

Although ugly, it has served me well, but is heavily compiler specific. I am not surprised gcc-5.x.x pukes at it. It is probably time to upgrade to a proper unit test framework. 

Any suggestions? :)

pellepl, 2016-04-21:
Cleaned up the testbench a bit, now it should work on gcc5+. And clang for that matter.

dpruessner, 2016-04-21:
Whoa-- you're fast!  Thanks!  Looks like a slick little filesystem.  I'm excited to test it.  We're considering using it for over-the-air updates and factory reset images with the Broadcom WICED stack.

closed 2016-04-21 by dpruessner

pellepl, 2016-04-22:
> Whoa-- you're fast! Thanks!
Heh, sometimes I actually do get the thumb out ;)

Sounds fun, as far as I know spiffs has never been used on WICED. Let me know how it works and if you stumble on any problems.

Cheers!

dpruessner, 2016-04-23:
Just got it working with the snippit...  Looking good for the application side.

It's gonna be *really* tight for the bootloader.   One of the really attractive things was to simplify selection of firmware images: to have one factory default, and one "update" image on the external flash.  So if things go bad, it's easy to get the factory default from the bootloader.

Looks like I can get the SPIFFS down to ~10kB.  But I only have 16kB for the bootloader, the way the STM32F MCU partitions flash banks.  Between 1.3k for libc, and 1.7k for the MCU host libraries (for SPI), leaves < 1k for the app.

pellepl, 2016-04-23:
That is a challenge indeed..! Do you have separate binaries for spiffs, one for the bootloader and one for the app? I guess you've already done this, but in the bootloader case, try stripping everything. You probably do not need cache, write capabilities, and so forth. Making spiffs singleton will cut down on flash usage. Consider using polling spi, without any stm lib, instead of fancy dma or irq.

What STM32 is it? I've written quite a few bootloaders for that processor by now :)

dpruessner, 2016-04-23:
Hah-- good thoughts.  You just described the rough path I am currently trying.   I already had non-IRQ based SPI due to some weirdness with the RTOS.

We're using STM32F205RG (EMW3162 board; switching to ISM43xx next).  

I switched to singleton through compile-flag defines for the bootloader.  I've stripped out a lot.  But the complexity of filesystem access and return-codes, I need to print some stuff; using some bare-bones UART `puts` and `atoi`.  I've gotten a bit down.  It would be nice to write-- to delete files or update a boot count (too many failed boots => restore factory default image).

It's cutting it close.  May not get to write to the FS--  I think it will all fit with read-only.

Thanks!

pellepl, 2016-04-23:
Perhaps you can store the file ids you want to delete in some shared ram between bootloader/app and delete it on app startup instead? Reg boot count, I've been using such a shared ram mechanism - counting number of bad resets from the app (e.g. assert, watchdog, hardfault before app is considered ok). If going beyond some max value, factory image is automatically flashed.

dpruessner, 2016-04-23:
Good thoughts!  I need to look more closely at the shared ram space-- how that could work with the linker scripts in the WICED world.

I'm pushing up on a testing deadline-- trying to figure what features here get locked down for functional/field test.  I'd love to have extra factory image safeguards...  worried about getting lost in linker land.

pellepl, 2016-04-23:
Hahaha, yes, one tend to spend quite some time there when tinkering.

Godspeed to you! :) Shout if you need help

dpruessner, 2016-04-23:
Ok... Since you're offering help... Mind giving a little light on how you're doing the reboot count with STM32F?-- I thought the watchdog reset would clear out all the SRAM, no?  I know you can look at a flag and see that WATCHDOG was the reason for reset; but I can't figure out how to track a count without writing to flash....

Just changed bootloader over to SPIFFS_READ_ONLY-- saves a couple kB-- thanks!

pellepl, 2016-04-23:
>  I thought the watchdog reset would clear out all the SRAM, no?

Well - the watchdog reset is just as any other reset really. You need to do linker magic. It is actually not that hard. I am using gnu linker myself, so you'd need to adapt to your own environment. There are many ways to do this, but I think this is the easiest and will put least modification to existing linker scripts.

Any case, the link file defines the ram and flash section, something like this:
```
MEMORY
{
  FLASH (rx)      : ORIGIN = 0x08000000, LENGTH = 128K
  RAM (xrw)       : ORIGIN = 0x20000000, LENGTH = 20*1024
}
``` 
Say you want 32 bytes of shared ram, so redefine this to:
```
MEMORY
{
  FLASH (rx)      : ORIGIN = 0x08000000, LENGTH = 128K
  RAM (xrw)       : ORIGIN = 0x20000000, LENGTH = 20*1024-32
  SH_RAM (rw)  : ORIGIN = 0x20000000+LENGTH(RAM), 32
}
``` 

Then, later down in your linker file, preferably after the ```.bss``` section, you define your shared memory section:
```
  /* Shared memory area, not wiped at reset */	
  .shmem: {
    __shared_memory_address__ = .;         /* define a global symbol at shared memory start */
    *(.shmem)
    *(.shmem.*)
    . = ALIGN(4);
  } >SH_RAM
```
The whole reason of doing this is to keep your shared memory away from the .bss section, as this is (normally) zeroed at reset - take a peek in your startup_stm***.s file.

Now, in your bootloader and your app, you define something like:
```
__attribute__(( section(.shmem) )) uint8_t shared_memory[32]; // or some other struct
```
When accessing ```shared_memory``` from either app or bootloader, this symbol will point to the same address in ram (0x20000000+20K-32 bytes), and, more importantly, will not be wiped at reset.

A very good idea is to put some kind of magic number in the shared memory to make sure it is valid. This is to cover two reasons: cold start, and inadvertent memory overwrites.

talon75, 2016-04-23:
Hi! Hope you don't bother if I try to help a bit...
Of course, @pellepl is absolutely right.
Anyway, on STM32, you could also use backup domain registers to keep your boot counter. Or, since you're on STM32F2, you have plenty of backup SRAM to play with.

pellepl, 2016-04-23:
Very true. Just beware, that if you're using tamper events/irq, the BKUP area will be cleansed. On the other hand, if you're sleeping in standby mode, the backup ram is your only choice.

dpruessner, 2016-04-23:
Oh wow-- great ideas, guys.  Thanks!

I forgot that it is the crt0 that zeros the memory-- that the core doesn't zero out SRAM on rest.  I guess anything left resident (assuming power is constant) will still be there after reset.

I'll parse the WICED linker scripts for a structure similar to what you described, @pellepl.  The folks doing the WICED stack have included a lot of nice-to-have's.  But not all the features/use-cases are documented.  I've seen SHM references in linker.

Really appreciate it!

dpruessner, 2016-04-26:
Shared memory worked like a champ, adapting what you suggested.  WICED didn't have anything built-in.  Got it tracking boot progress, crashes, firmware update failure, etc.  Did have to add in WICED make scripts to make `objdump` disregard the .shmem section.  It made the `.bin` file ~400MB otherwise!

I will have a few questions about preemptively erasing deleted pages...  Not sure if what I was seeing was expected behavior.  I'll re-prototype in code I can share.  I changed the erase blocks to 64kB, leaving logical pages at 256byte (using [Adesto AT25SF081](http://datasheet.octopart.com/AT25SF081-SSHD-B-Adesto-Technologies-datasheet-31984108.pdf) ).  It's working much better, with fewer delays while writing.

pellepl, 2016-04-27:
Great. Just ask on when you're read 😉

For cleaning up yourself, you have SPIFFS_gc and SPIFFS_gc_quick.