pellepl/spiffs #89: Strange behavior with GC

gabrielsan, 2016-05-04:
Hi there. I'm getting file corruption when calling SPIFFS_gc between writes in a file.

I'm in a multithread env. with lock configured for the fs , one thread is writing while the other calls SPIFFS_gc once every half a second or so.

I'm guessing this is not the right way to use GC. I read the wiki page for a clue where to put forced calls to GC but did not found it.

pellepl, 2016-05-05:
You're supposed to be able to call GC whenever you want. It is a pretty fat operation though (if it fires), so doing a gc twice a second seems much. Is this to make sure the write thread wont block?

Anyhow, of course it should never corrupt anything. Might be a lock problem, #87 #88, I'll look into it a soon as I can.

gabrielsan, 2016-05-05:
I'm aqua from work today. I'll check tomorrow and give you an update. 

I'm running with SS=LBS=4K, LPS=1K, with a total of 2048 blocks. It runs fast, but I'm not sure if I should use a larger logical block size. I've been using spiffs for the last 2 days only, :D

pellepl, 2016-05-05:
> I'm aqua from work today.

Me too ;)

> I'm running with SS=LBS=4K, LPS=1K, with a total of 2048 blocks.

Oh that is a veeeery large page size - you have only 4 pages / block where
1 is used for meta. I would recommend either larger block size >=16k or
smaller page size, e.g. 128 or 256 bytes.

Cheers / P

gabrielsan, 2016-05-09:
so, I've increased the logical block size to 64Kb and the problem is gone. And yes, calling GC half a sec is a bit to much. I have yet to figure where/when to call it.

I have to move to other stuff right now, but if I find something else I'll come back here.

Thanks for the help.

pellepl, 2016-05-10:
No problem. Glad it worked out!

closed 2016-05-10 by pellepl