pellepl/spiffs #90: file not visible in SPIFFS_readdir

benpicco-tmp, 2016-05-20:
Hi,
you' ve always been curious about 'interesting' spiffs states, so I've got this one for you: In this dump there is a file called `cli.bin.a32` but it does not show up when querying all files like this

```
void ls(void) {
	spiffs_DIR dir;
	struct spiffs_dirent file;

	SPIFFS_opendir(&spi_fs, 0, &dir);

	while (SPIFFS_readdir(&dir, &file)) {
		printf("\t%d %s\n", file.size, file.name);
	}

	SPIFFS_closedir(&dir);
}
```

```
> ls
2016-05-20 10:56:20,956 - INFO # 	0 .volaWare
2016-05-20 10:56:20,978 - INFO # 	68736 cli.bin
```

However, after deleting `cli.bin`…

```
> rm_all
2016-05-20 10:56:25,434 - INFO # 	deleting cli.bin…
> ls
2016-05-20 10:56:37,208 - INFO # 	0 .volaWare
2016-05-20 10:56:37,229 - INFO # 	4 cli.bin.a32
```

it shows up!

I have no trouble opening/reading the file at any time.

my config is
```
    cfg.phys_erase_block = BLOCK_4K;    // 4096
    cfg.log_block_size   = BLOCK_4K;
    cfg.log_page_size    = 128;
```
[hidden_file.spiffs](http://flauschlabor.de/foobar/hidden_file.spiffs)

pellepl, 2016-05-24:
Thanks, I will have a peek at it as soon as I can (somewhat chaotic right
now).

Cheers / Peter

pellepl, 2016-06-07:
Thanks for finding this. It was a corner case indeed, but could lead to potentially very weird behaviour. Please retest with the new patch.

Cheers!

pellepl, 2016-07-20:
Ok to close?

closed 2016-07-20 by benpicco-tmp