pellepl/spiffs #92: NullPointer dereferencing

niclash, 2016-05-28:
https://github.com/pellepl/spiffs/blob/master/src/spiffs_cache.c#L154

A few lines earlier there is a if(cp), which means that cp can be returned "0" from the previous function, yet the cp->ix is called on line 154, which I think (not 100% sure yet) cases segfault on my system sometimes.

pellepl, 2016-05-28:
Um yes, that particular piece of code was pretty ugly, sorry for that. I will clean it uo. 

Purely functionally, before the page allocation there is a call to spiffs_cache_page_remove_oldest - this ensures that a page always is available. So cp should never be null.

niclash, 2016-05-28:
Possibly... So, I think you suggest that my code is suffering from
concurrency problems, which is totally possible.

pellepl, 2016-06-05:
Might be. I've cleaned things up a bit, you could try this change also. Though I'd bet all butter in Smalland that it won't do anything.

Cheers for notifying me!

closed 2016-06-05 by pellepl