pellepl/spiffs #97: Add index memory mapping feature

pellepl, 2016-08-08:
Allow users to map first level indices to memory array, making reading faster. For e.g. streaming and random access with lseek.

someburner, 2016-08-08:
Hey @pellepl 

I see you've been hammering out some enhancements for bigger files. I haven't tried out any of the last few commits, but I was wondering if any of these would pertain to one of my current uses of SPIFFS. 

In order to optimize SPIFFS and also track my own custom flags for files, I generated an in-memory list containing pix, name, and size. To skip the extra step of looking for the filename I was using the **SPIFFS_open_by_page** method you conveniently provided. But I found that when creating/writing a SPIFFS object (file), once the file exceeded 64KBytes, the pix would change.

Not having a full understanding of SPIFFS, I was worried this would cause things to break so I reverted to just using open by name. Anyways, I'm curious if any of your recent developments would help with this? Or will this issue you've made here provide new functionality for larger files?

pellepl, 2016-08-08:
Hi there @someburner,

You not only need to build your file/obj.id/page.index map, you also need to set a callback using `SPIFFS_set_file_callback_func`. In this callback, you will be notified if any file header is reassigned to a new page. So in your callback, you need to check if the callbacked obj.id matches any of your in-memory obj.ids, and if so update the page with the argument in the callback.

I will describe this under the wiki [here](https://github.com/pellepl/spiffs/wiki/Performance-and-Optimizing) soon.

I guess you created a file, snatched the obj.id and then updated the file with more data. This will move the header to a new page index as the header needs to be rewritten and your old registered page index will point to a deleted page.

If you know that the file system will not be modified in any way, it is ok to get all obj.id/page.indices once everything is written. But, if modifying any file, its header will move to a new page index. If a gc is also triggered, many headers may move.

e04a67b (`SPIFFS_TEMPORAL_FD_CACHE`) is a general cache remembering the files accessed most frequently - if there are file handles enough. If you on the other hand know in beforehand which files will be frequently accessed, above approach is better.

All above is optimizations for _opening_ files. 

87856826 on the other hand is an optimization for _reading_ a file. Here is functionality for reading all necessary metadata up front, and then when the file is read memory is used instead. This applies only to (very) large files and I am not so sure you would benefit from it.

I will update the wiki soon (following days) with examples and explanations of all optimizations I can think of.

someburner, 2016-08-08:
Gotcha. That makes much more sense. `SPIFFS_set_file_callback_func` looks like fun :)

pellepl, 2016-08-08:
Oh, you couldn't imagine ;)

Just don't mess with the file system from within the callback, i.e. no
SPIFFS_ calls. All hell will break loose.

Happy hacking / P

closed 2016-08-08 by pellepl