pellepl/spiffs #103: Pointer dereferencing and alignment issues with working buffers

dmarcosgon, 2016-09-07:
Hi,
I'm porting this library to PIC32, which uses MIPS architecture. With the library as-is, it crashed due to memory alignment errors while accessing the working buffers, as this particular MIPS architecture doesn't support unaligned access as most ARM do.  The issue are lines like this one, where pointers are dereferenced and typecast to more strictly aligned types:

((spiffs_page_ix*)((u8_t *)objix_hdr + sizeof(spiffs_page_object_ix_header)))[data_spix] = data_page;

from nucleus, line 1326. I have solved the issue myself by substituting these lines for memcpy's, which is a more portable and less prone to error way of doing it I think. These are the two (crude) regex's I used to change all the lines for memcpys:

FIND: (.+) = (\(\(spiffs_page_ix\*.+)\[(.+)\]; 
REPLACE: myptr = $2;\r\nmyptr+=$3;\r\nmemcpy((void*)&$1,(const void*)myptr,sizeof(spiffs_page_ix));
FIND: (\(\(spiffs_page_ix\*.+)\[(.+)\] = (.+);
REPLACE: myptr=$1;\r\nmyptr+=$2;\r\nmemcpy((void*)myptr,(const void*)&$3,sizeof(spiffs_page_ix));

Also you need to add the spiffs_page_ix* myptr variable. I'm not sure of the implications that using memcpy has but if there's no performance impact or other issue could you please implement this modification to allow easy porting for less forgiving architectures?

Thanks,
David

pellepl, 2016-09-07:
Hi David,

Did you try enabling  `SPIFFS_ALIGNED_OBJECT_INDEX_TABLES` in your spiffs_config.h?

Cheers / Peter

valkuc, 2016-12-08:
> Did you try enabling  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES  in your spiffs_config.h
Helped for me on ESP8266

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl