pellepl/spiffs #104: Read fails if file size is multiple of 256

arjun789, 2016-09-29:
If i create a file with length in multiples of 256, reading fails with -10003. 

Config file:

`#define FLASH_ERASE_SIZE (32 * 1024)
#define FLASH_CHIP_CAPACITY (1 * 1024 * 1024)
#define FLASH_PAGE_SIZE 256
#define MAX_FILENAME_LEN 32

#ifndef SPIFFS_DBG
#define SPIFFS_DBG(...) //printf(__VA_ARGS__)
#endif
#ifndef SPIFFS_GC_DBG
#define SPIFFS_GC_DBG(...) //printf(__VA_ARGS__)
#endif
#ifndef SPIFFS_CACHE_DBG
#define SPIFFS_CACHE_DBG(...) // printf(__VA_ARGS__)
#endif
#ifndef SPIFFS_CHECK_DBG
#define SPIFFS_CHECK_DBG(...) //printf(__VA_ARGS__)
#endif

#ifndef SPIFFS_BUFFER_HELP
#define SPIFFS_BUFFER_HELP 0
#endif

#ifndef SPIFFS_CACHE
#define SPIFFS_CACHE 1
#endif
#if SPIFFS_CACHE
#ifndef SPIFFS_CACHE_WR
#define SPIFFS_CACHE_WR 1
#endif

#ifndef SPIFFS_CACHE_STATS
#define SPIFFS_CACHE_STATS 1
#endif
#endif

#ifndef SPIFFS_PAGE_CHECK
#define SPIFFS_PAGE_CHECK 1
#endif

#ifndef SPIFFS_GC_MAX_RUNS
#define SPIFFS_GC_MAX_RUNS 5
#endif

#ifndef SPIFFS_GC_STATS
#define SPIFFS_GC_STATS 1
#endif

#ifndef SPIFFS_GC_HEUR_W_DELET
#define SPIFFS_GC_HEUR_W_DELET (5)
#endif

#ifndef SPIFFS_GC_HEUR_W_USED
#define SPIFFS_GC_HEUR_W_USED (-1)
#endif

#ifndef SPIFFS_GC_HEUR_W_ERASE_AGE
#define SPIFFS_GC_HEUR_W_ERASE_AGE (50)
#endif

#ifndef SPIFFS_OBJ_NAME_LEN
#define SPIFFS_OBJ_NAME_LEN (32)
#endif

#ifndef SPIFFS_COPY_BUFFER_STACK
#define SPIFFS_COPY_BUFFER_STACK (256)
#endif

#ifndef SPIFFS_USE_MAGIC
#define SPIFFS_USE_MAGIC (1)
#endif

#if SPIFFS_USE_MAGIC
#ifndef SPIFFS_USE_MAGIC_LENGTH
#define SPIFFS_USE_MAGIC_LENGTH (0)
#endif
#endif

#ifndef SPIFFS_LOCK
#define SPIFFS_LOCK(fs)
#endif

#ifndef SPIFFS_UNLOCK
#define SPIFFS_UNLOCK(fs)
#endif

#ifndef SPIFFS_SINGLETON
#define SPIFFS_SINGLETON 1
#endif

#if SPIFFS_SINGLETON
#ifndef SPIFFS_CFG_PHYS_SZ
#define SPIFFS_CFG_PHYS_SZ(ignore) (FLASH_CHIP_CAPACITY)
#endif
#ifndef SPIFFS_CFG_PHYS_ERASE_SZ
#define SPIFFS_CFG_PHYS_ERASE_SZ(ignore) (FLASH_ERASE_SIZE)
#endif
#ifndef SPIFFS_CFG_PHYS_ADDR
#define SPIFFS_CFG_PHYS_ADDR(ignore) (0)
#endif
#ifndef SPIFFS_CFG_LOG_PAGE_SZ
#define SPIFFS_CFG_LOG_PAGE_SZ(ignore) (FLASH_PAGE_SIZE)
#endif
#ifndef SPIFFS_CFG_LOG_BLOCK_SZ
#define SPIFFS_CFG_LOG_BLOCK_SZ(ignore) (FLASH_ERASE_SIZE)
#endif
#endif

#ifndef SPIFFS_ALIGNED_OBJECT_INDEX_TABLES
#define SPIFFS_ALIGNED_OBJECT_INDEX_TABLES 0
#endif

#ifndef SPIFFS_HAL_CALLBACK_EXTRA
#define SPIFFS_HAL_CALLBACK_EXTRA 0
#endif

#ifndef SPIFFS_FILEHDL_OFFSET
#define SPIFFS_FILEHDL_OFFSET 0
#endif

#ifndef SPIFFS_READ_ONLY
#define SPIFFS_READ_ONLY 0
#endif

#ifndef SPIFFS_TEST_VISUALISATION
#define SPIFFS_TEST_VISUALISATION 1
#endif
#if SPIFFS_TEST_VISUALISATION
#ifndef spiffs_printf
#define spiffs_printf(...) printf(__VA_ARGS__)
#endif
#ifndef SPIFFS_TEST_VIS_FREE_STR
#define SPIFFS_TEST_VIS_FREE_STR "_"
#endif
#ifndef SPIFFS_TEST_VIS_DELE_STR
#define SPIFFS_TEST_VIS_DELE_STR "/"
#endif
#ifndef SPIFFS_TEST_VIS_INDX_STR
#define SPIFFS_TEST_VIS_INDX_STR(id) "i"
#endif
#ifndef SPIFFS_TEST_VIS_DATA_STR
#define SPIFFS_TEST_VIS_DATA_STR(id) "d"
#endif
#endif

typedef int32_t s32_t;
typedef uint32_t u32_t;
typedef int16_t s16_t;
typedef uint16_t u16_t;
typedef int8_t s8_t;
typedef uint8_t u8_t;

typedef u16_t spiffs_block_ix;
typedef u32_t spiffs_page_ix;
typedef u32_t spiffs_obj_id;
typedef u16_t spiffs_span_ix;
`

pellepl, 2016-09-30:
Hi @arjun789,

I cannot reproduce it here. I wrote following test:
```C
TEST(files_256bytes_boundary_bug_104) {
// use following config types for this test:
//  typedef u16_t spiffs_block_ix;
//  typedef u32_t spiffs_page_ix;
//  typedef u32_t spiffs_obj_id;
//  typedef u16_t spiffs_span_ix;
  fs_reset_specific(0, 0, 1*1024*1024, 32*1024, 32*1024, 256);
  int i;
  const int runs = 16;
  u8_t ref_data[runs*256];
  u8_t buf_data[runs*256];
  for (i = 0; i < runs; i++) {
    int res;
    spiffs_file fd;
    char fname[32];
    int len = 256 * (i+1);

    // create file, fill up with random data
    sprintf(fname, "file%i", len);
    memrand(ref_data, len);
    fd = SPIFFS_open(FS, fname, SPIFFS_O_CREAT | SPIFFS_O_TRUNC | SPIFFS_O_WRONLY, 0);
    TEST_CHECK_GE(fd, 0);
    TEST_CHECK_EQ(SPIFFS_write(FS, fd, ref_data, len), len);
    TEST_CHECK_EQ(SPIFFS_close(FS, fd), SPIFFS_OK);

    // read back, verify data
    fd = SPIFFS_open(FS, fname, SPIFFS_O_RDONLY, 0);
    TEST_CHECK_GE(fd, 0);
    TEST_CHECK_EQ(SPIFFS_read(FS, fd, buf_data, len), len);
    TEST_CHECK_EQ(memcmp(ref_data, buf_data, len), 0);
    TEST_CHECK_EQ(SPIFFS_close(FS, fd), SPIFFS_OK);
  }

  return TEST_RES_OK;
} TEST_END
```

but everything worked as expected. The only way to force a -10003 was to read beyond the actual filesize. Am I doing something differently?

Cheers / P

arjun789, 2016-10-01:
>The only way to force a -10003 was to read beyond the actual filesize. Am I doing something differently?

Aren't you supposed to return a zero?

pellepl, 2016-10-04:
SPIFFS_read will return number of bytes read, or a negative number being the error. 
Oh... Ok, if you have a file being 100 bytes and you try to read 101 bytes, you'll get -10003. When you should get 100 instead. Subsequent reads should return -10003.

Is this the problem you mean?

arjun789, 2016-10-04:
Yes

>Subsequent reads should return -10003.
Any reason why you don't want to return 0 here as well?

rojer, 2016-10-04:
yes, reads at/past EOF should return 0, that's the standard way of indicating eof.

pellepl, 2016-10-04:
Ok, thanks. Will fix.

closed 2017-03-02 by pellepl