pellepl/spiffs #107: How to truncate a file?

yiminglei, 2016-10-07:
How to truncate a file?

For example,I have a 100 bytes file.After it be used.It turn to 60 bytes.This file size is still 100 bytes. How to truncate a file?

Thank you.

pellepl, 2016-10-12:
In spiffs_nucleus.c, there is a `spiffs_object_truncate` function. This is an internal function, but it will do what you want. 

To call it, you need to expose it in spiffs.h - by adding e.g. `SPIFFS_truncate(spiffs *fs, const char *path, int newsize)`. Implement this new function in spiffs_hydrogen.c. You can regard functions `SPIFFS_remove` as a template for your new `SPIFFS_truncate`.

Do note, that `spiffs_object_truncate` only can make the file smaller. In posix, trunc can make files bigger. Which may seem a bit odd, but there you are.

Cheers / P

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl

VikOlliver, 2019-10-10:
Yeah, I'd find this handy in a real world implementation...