pellepl/spiffs #108: how to get filename from obj_id in file_cb()

everslick, 2016-10-11:
i would like to find out the filename of a file in the file_cb(). is this possible without too much of a performance hit?

pellepl, 2016-10-12:
Getting the filename from a file_cb will require some small coding from your side. In your callback, you'll get the page index for the updated file (argument `pix`). In spiffs_hydrogen.c, there is a static function called `spiffs_stat_pix`. You simply call `spiffs_stat_pix` with the `pix` you get in the callback. You need to expose the `spiffs_stat_pix`.
This will fail if the file was deleted though (as the page with the filename just was removed). 
As for the performance hit, this will be O(1) as you have the page index - no search is required. If you have the cache enabled, the page should be cached as spiffs just tampered with it, so the whole thing should basically result in one memcpy.

Getting filedescs from a callback is not feasible as there can be many filedescriptors open on the same file.

As for your other question, if you have many filedescs open on the same file and someone removes that file, all the filedescs related to that file will be automatically closed and invalidated.

Some pseudo code:
```C
spiffs_file fd1, fd2;
fd1 = SPIFFS_open(fs, "blabla", SPIFFS_O_RDWR, 0);
fd2 = SPIFFS_open(fs, "blabla", SPIFFS_O_RDWR, 0);

SPIFFS_fremove(fs, fd2); // or SPIFFS_remove(fs, "blabla");
uint8_t c;
SPIFFS_read(fs ,fd1, &c, 1); //<--- will return SPIFFS_ERR_FILE_CLOSED (-10008)
```

everslick, 2016-10-12:
thank you very much, for your answer.

because i'm especially interested in when a file was removed someplace else in the code, i thought to exploit the file_cb() mechanism. but as the filename is already deleted the moment the CB fires, this approach is misguided anyway. nevertheless I guess I can just rely on file operations to fail on removed files and reopen them as needed.

after digging through SPIFFS code for some time now, i have to say that i'm very impressed. your documentation is extraordinary! thank you!

pellepl, 2016-10-12:
Wow, thanks for the kind words! No one ever told me that in my two decades
of software business ;)

When I think of it, you could probably get away with reading the deleted
page actually. The callback for a deleted file comes directly after the
deletion iirc, so the page will be deleted but not overwritten at the time
of the callback. You could probably just simply call `spiffs_stat_pix` and
get the filename, regardless of operation.

Cheers / P

everslick, 2016-10-12:
hehe, you are welcome!

yes, it occurred to me as well, that i might just read the filename from
the deleted page. but:

this makes me a little uncomfortable, because it relies on internal
behavior that might change some day.
second i would have to mess with the API (make the static function non
static (and as i write a firmware for ESP8266, that uses the Arduino Core
for the ESP8266, i want to use the code that is already there, unaltered))
third, my workaround (trying to seek in the file and checking the return
code) works reasonable well for now.

so thanks for your suggestions, they are very much appreciated -
nevertheless!

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl