pellepl/spiffs #112: low  SPIFFS_open performance

christianeisendle, 2016-10-20:
In some situation I experienced bad performance when calling SPIFFS_open.
Normally, when calling SPIFFS_open it takes less than 100us for this function to open the file handle in return.
In some situations, however, I measured up to 30ms. I debugged this further and I saw that it is related to the search in spiffs_obj_lu_find_entry_visitor, since it starts searching at the position of the last opened file (cursor_obj_lu_entry points to that). Now if the file i want to open is located before the last opened file, the search, which is linear, takes maximum time, so I end up at 30ms. 

If I reset the pointer everytime to 0 before doing the search, i'm back at < 100us. 
I just have a few files stored on the FS, so it should be fine. But i'm wondering why cursor_obj_lu_entry is pointing to the last opened file. Is it done because of the assumption that one and the same file may be re-used? (Or, another case would be when creating new files, it's faster to find a new location). I don't know all the details yet about spiffs so there might be a good reason why it's done like this. (And maybe there is also something which could be optimized of course :smile: 

Thanks,
Christian.

pellepl, 2016-10-20:
Yes, this is designed this way. The idea was that if a file is modified, it is moved "forwards", and will be quicker to find. Also, having a rotating cursor makes the cost of a random file search even over the time. Then again, in your case it doesn't work that well. Have you tried enabling `SPIFFS_TEMPORAL_FD_CACHE`? This would probably make magic in your case.

christianeisendle, 2016-10-20:
I'm still using 0.3.5, temporal cache is only available with 0.3.6 (#95), right?
But if I understand right with temporal caching enabled first open still may take long but subsequent open are then cached and should then be faster. I will try that by tomorrow and will perform some performance measurements. Thanks for your quick response.

pellepl, 2016-10-20:
Ah, yes, it is a 0.3.6 feature. But you can also keep important files open. See [this](https://github.com/pellepl/spiffs/wiki/Performance-and-Optimizing#just-one-or-few-important-files), it should work for 0.3.5 also. Rolling your own cache (the section below) will work as well.

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl