pellepl/spiffs #114: Failed writing to file! -10012

ideepuraj, 2016-11-10:
Hi,
I have ported the spiffs to PIC32 MIPS platform and the basic FS is up and running.
But i am facing a frequent file system corruption while writing files continuously and the file system will never recover. After rebooting and mounting the file systems all the file write or file creation will fail until reformat the entire file system.

any help on this will be highly appreciated. 

file write

                fileUpload = SPIFFS_open(&fs, fileTxRx_name,
                                SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
                if (fileUpload < 0) {
                    DB_ERR("FileUpload: Failed to create %s %d\n", 
                                    fileTxRx_name, SPIFFS_errno(&fs));
                }
loop --
                int ret = SPIFFS_write(&fs, fileUpload,
                                appData.readBuffer, appData.cdcReadDataSize);
                if(ret < 0) {
                    DB_ERR("FILEUPLOAD: Failed writing to file! %d\n", ret);
                    fileTxRx_size = -1; /* hack to fail*/
                }
end loop --

fs
    spiffs_config cfg;
    cfg.phys_size = 4 * 1024 * 1024; /* define flash layout */
    cfg.phys_addr = 0;              /* start spiffs at start of spi flash */
    cfg.phys_erase_block = 65536;   /* block erase size */
    cfg.log_block_size = 65536;     
    cfg.log_page_size = LOG_PAGE_SIZE;

    cfg.hal_read_f = my_spiffs_read;
    cfg.hal_write_f = my_spiffs_write;
    cfg.hal_erase_f = my_spiffs_erase;

config
#define SPIFFS_BUFFER_HELP              0
#define SPIFFS_CACHE                    1
#define SPIFFS_CACHE_WR                 1
#define SPIFFS_CACHE_STATS              0
#define SPIFFS_PAGE_CHECK               1
#define SPIFFS_GC_MAX_RUNS              5
#define SPIFFS_GC_STATS                 0
#define SPIFFS_GC_HEUR_W_DELET          (5)
#define SPIFFS_GC_HEUR_W_USED           (-1)
#define SPIFFS_GC_HEUR_W_ERASE_AGE      (50)
#define SPIFFS_OBJ_NAME_LEN             (32)
#define SPIFFS_COPY_BUFFER_STACK        (64)
#define SPIFFS_USE_MAGIC                (1)
#define SPIFFS_USE_MAGIC_LENGTH         (0)
#define SPIFFS_ALIGNED_OBJECT_INDEX_TABLES       1
#define SPIFFS_HAL_CALLBACK_EXTRA         0
#define SPIFFS_FILEHDL_OFFSET                 0
#define SPIFFS_READ_ONLY                      0
#define SPIFFS_TEMPORAL_FD_CACHE              0
#define SPIFFS_TEMPORAL_CACHE_HIT_SCORE       4
#define SPIFFS_IX_MAP                         1
#define SPIFFS_TEST_VISUALISATION         1

pellepl, 2016-11-12:
Hi,
your config seems sane (though you probably want to enable ```SPIFFS_TEMPORAL_FD_CACHE``` also).
The -10012 ```SPIFFS_ERR_IS_FREE``` is a fatal error code in most cases. Do you get it while writing? It normally indicates some spi flash hal layer problem, at least when I've seen it. It could be worthwhile rechecking the write and erase functions of your spi flash drivers. What does ```SPIFFS_vis``` give you?

ideepuraj, 2016-11-13:
@pellepl ,
Thanks for the reply. I will double check my SPI HAL layers to ensure that all write and erase functions works correctly. 

I am using a Winbond w25Q128 flash for my project.  Do you have any working driver for the same ?

SPIFFS_TEMPORAL_FD_CACHE is disabled as just a debug option. i was using an older version of the SPIFFS and updated to latest just see any fix for my write failure.

Regards, 
Deepu Raj

ideepuraj, 2016-11-14:
```
my_spiffs_read(): addr: 0xf9200 size:0x100
my_spiffs_write(): addr: 0xf9400 size:0x100
my_spiffs_write(): addr: 0xf0124 size:0x2
my_spiffs_write(): addr: 0xf9404 size:0x1
my_spiffs_write(): addr: 0xf0120 size:0x2
my_spiffs_write(): addr: 0xf9204 size:0x1
my_spiffs_read(): addr: 0xf9300 size:0x100
my_spiffs_write(): addr: 0xf9376 size:0x8a
my_spiffs_write(): addr: 0xf0126 size:0x2
my_spiffs_write(): addr: 0xf9500 size:0x5
my_spiffs_write(): addr: 0xf9505 size:0x76
my_spiffs_write(): addr: 0xefa00 size:0x100
my_spiffs_read(): addr: 0xf9400 size:0x100
my_spiffs_write(): addr: 0xf9600 size:0x100
my_spiffs_write(): addr: 0xf0128 size:0x2
my_spiffs_write(): addr: 0xf9604 size:0x1
my_spiffs_write(): addr: 0xf0124 size:0x2
my_spiffs_write(): addr: 0xf9404 size:0x1
my_spiffs_read(): addr: 0xf9500 size:0x100
my_spiffs_write(): addr: 0xf957b size:0x85
my_spiffs_write(): addr: 0xf012a size:0x2
my_spiffs_write(): addr: 0xf9700 size:0x5
my_spiffs_write(): addr: 0xf9705 size:0x7b
my_spiffs_write(): addr: 0xefa00 size:0x100
my_spiffs_read(): addr: 0xf9600 size:0x100
my_spiffs_write(): addr: 0xf9800 size:0x100
my_spiffs_write(): addr: 0xf012c size:0x2
my_spiffs_write(): addr: 0xf9804 size:0x1
my_spiffs_write(): addr: 0xf0128 size:0x2
my_spiffs_write(): addr: 0xf9604 size:0x1
my_spiffs_read(): addr: 0xf9700 size:0x100
my_spiffs_write(): addr: 0xf9780 size:0x80
my_spiffs_write(): addr: 0xf012e size:0x2
my_spiffs_write(): addr: 0xf9900 size:0x5
my_spiffs_write(): addr: 0xf9905 size:0x80
my_spiffs_write(): addr: 0xefa00 size:0x100
my_spiffs_read(): addr: 0xf9800 size:0x100
my_spiffs_write(): addr: 0xf9a00 size:0x100
my_spiffs_write(): addr: 0xf0130 size:0x2
my_spiffs_write(): addr: 0xf9a04 size:0x1
my_spiffs_write(): addr: 0xf012c size:0x2
my_spiffs_write(): addr: 0xf9804 size:0x1
my_spiffs_read(): addr: 0xf9900 size:0x100
my_spiffs_write(): addr: 0xf9985 size:0x7b
my_spiffs_write(): addr: 0xf0132 size:0x2
my_spiffs_write(): addr: 0xf9b00 size:0x5
my_spiffs_write(): addr: 0xf9b05 size:0x85
my_spiffs_write(): addr: 0xefa00 size:0x100
my_spiffs_read(): addr: 0xf9a00 size:0x100
my_spiffs_write(): addr: 0xf9c00 size:0x100
my_spiffs_write(): addr: 0xf0134 size:0x2
my_spiffs_write(): addr: 0xf9c04 size:0x1
my_spiffs_write(): addr: 0xf0130 size:0x2
my_spiffs_write(): addr: 0xf9a04 size:0x1
my_spiffs_read(): addr: 0xf9b00 size:0x100
my_spiffs_write(): addr: 0xf9b8a size:0x76
my_spiffs_write(): addr: 0xf0136 size:0x2
my_spiffs_write(): addr: 0xf9d00 size:0x5
my_spiffs_write(): addr: 0xf9d05 size:0x8a
my_spiffs_write(): addr: 0xefa00 size:0x100
my_spiffs_read(): addr: 0xf9c00 size:0x100
my_spiffs_write(): addr: 0xf9e00 size:0x100
my_spiffs_write(): addr: 0xf0138 size:0x2
my_spiffs_write(): addr: 0xf9e04 size:0x1
my_spiffs_write(): addr: 0xf0134 size:0x2
my_spiffs_write(): addr: 0xf9c04 size:0x1
my_spiffs_read(): addr: 0xf9d00 size:0x100
my_spiffs_write(): addr: 0xf9d8f size:0x71
my_spiffs_write(): addr: 0xf013a size:0x2
[... 604 lines cut ...]
my_spiffs_read(): addr: 0x90000 size:0x100
my_spiffs_read(): addr: 0x90100 size:0x100
my_spiffs_read(): addr: 0xa0000 size:0x100
my_spiffs_read(): addr: 0xa0100 size:0x100
my_spiffs_read(): addr: 0xb0000 size:0x100
my_spiffs_read(): addr: 0xb0100 size:0x100
my_spiffs_read(): addr: 0xc0000 size:0x100
my_spiffs_read(): addr: 0xc0100 size:0x100
my_spiffs_read(): addr: 0xd0000 size:0x100
my_spiffs_read(): addr: 0xd0100 size:0x100
my_spiffs_read(): addr: 0xe0000 size:0x100
my_spiffs_read(): addr: 0xe0100 size:0x100
my_spiffs_read(): addr: 0xf0000 size:0x100
my_spiffs_read(): addr: 0xf0100 size:0x100
my_spiffs_read(): addr: 0x10000 size:0x100
my_spiffs_read(): addr: 0x10100 size:0x100
my_spiffs_erase(): addr: 0x1 size:0x10000
serial_flash_erase_block() addr: 0x1
my_spiffs_write(): addr: 0x101fe size:0x2
my_spiffs_write(): addr: 0x101fc size:0x2
my_spiffs_read(): addr: 0xff300 size:0x100
my_spiffs_read(): addr: 0x200 size:0x100
FILEUPLOAD: Failed writing to file! -10012
Transfer done
```

ideepuraj, 2016-11-14:
i have formatted the flash and tried a fresh file transfer with debug msg enabled. Please find the attached logs were it failed. 

Hope you will be able to help me on this.

ideepuraj, 2016-11-14:
Thanks for the help. from the above logs it clear that the chip erase block is not working and all crash logs points to an erase block few line before. I will check my flash driver to fix this

pellepl, 2016-11-15:
Sorry for not answering sooner. Yes, the serial_flash_erase_block() addr: 0x1
was my best bet also. Seems it could erase more than one logical block. Great you've found the problem.
Cheers / P

ideepuraj, 2016-11-15:
@pellepl  Thank you very much for your support and time. after the flash_erase_block fix (erase block address byte order was wrong and it only worked for block 0) everything is working fine so far. All my stress test are passed now.

Really appreciate your efforts. 
Cheers

closed 2016-11-15 by ideepuraj