pellepl/spiffs #115: SPIFFS_ERR_NOT_FOUND problem

Nicholas3388, 2016-11-16:
Hi, I use SPIFFS on ESP32. In the SPIFFS config file, I disable SPIFFS_USE_MAGIC, and mount file system successfully. But when I'm trying to call SPIFFS_open to create and open a new file, it returned failed. The function spiffs_object_find_object_index_header_by_name within SPIFFS_open return the errno SPIFFS_ERR_NOT_FOUND. Is there any solution for my problem? Thanks.

Nicholas3388, 2016-11-16:
If I enable SPIFFS_USE_MAGIC, the file system will mount failed.

pellepl, 2016-11-23:
Mounting without SPIFFS_USE_MAGIC will always succeed, as spiffs in this case will not check the fs at all. See [this](https://github.com/pellepl/spiffs/wiki/Using-spiffs#mounting).

What is the error code when enabling SPIFFS_USE_MAGIC? As far as I know, the ESP32 is uncharted territories, but if the spi hal layer resembles anything of that of the ESP8266, there are lot of alignment issues to think of.

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl