pellepl/spiffs #116: Does SPIFFs support directories?

nkolban, 2016-11-23:
The README.md page of the repository says that directories are not supported while the usage guide seems to show that directories are supported.  Is there an inconsistency here?

pellepl, 2016-11-24:
Directories are not supported. If you refer to the ```"/filename"``` in the examples, the initial slash will be a part of the filename. I think I included the slash in case spiffs would ever get repositories.

nkolban, 2016-12-03:
Many thanks for your response.  I guess I am confused over the APIs such as "SPIFFS_opendir" and "SPIFFS_readdir".   I had assumed that if I created a "file" called "/dir1/file1.txt" then that would have created a "directory" called "dir1" which if I'd opened it and then read it, would have resulted in finding "file1.txt".  Is that not the case?    How and when would I use SPIFFS_opendir and SPIFFS_readdir if directoris are not supported?

pellepl, 2016-12-03:
Spiffs is flat, so creating a file "/dir/foo.txt" would result in a file named "/dir/foo.txt" rather than a file "foo.txt" under "dir/". There is only root.
Still, if you want to list all your files under root you'll need opendir and readdir. The directory argument is ignored though, and is left for future expansion.

nkolban, 2016-12-03:
Thank you sir ... my dim light bulb went on.  I'm happily using your SPIFFS library under the ESP32 and about to map to the Virtual File System of their technology ... 

http://esp-idf.readthedocs.io/en/latest/api/vfs.html

Congratulations on the great library.

closed 2016-12-03 by nkolban

pellepl, 2016-12-03:
Heh, no problem - I suffer the same from time to time 😉 Ah cool, the ESP32 - haven't played with that myself yet. Hope your porting efforts are painless, I haven't heard of an ESP32 port of spiffs until now. Let me know if you run into trouble.
Cheers

jolivepetrus, 2017-03-09:
Hello @pellepl,

FYI: we are using a SPIFFS on ESP32, in our Lua RTOS project:

https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/lua_rtos/vfs/spiffs.c

pellepl, 2017-03-10:
@jolivepetrus, nice! Always fun to see new integrations. Nice feat with the quasi-directories!

sjlongland, 2017-11-15:
Just stumbled on this project and wondered the same thing… the API has calls for `SPIFFS_diropen` and friends… and they do take a directory path.

Looking at the implementation, I see it's ignored.  For now at least.
https://github.com/pellepl/spiffs/blob/master/src/spiffs_hydrogen.c#L1002

Is that likely to be the case long-term?  One neat feature I guess would be for this to filter out files that don't match a given prefix pattern, that would give you crude "directories".  I'm already looking at doing this in my own code.

So create files named `foo/a`, `foo/b`, `bar/a`, passing in `NULL` or `""` when listing directory contents would yield all files, but passing in `"foo/"` would just give you `foo/a` and `foo/b`.  Deleting a directory happens magically when you delete all the files that contain it, and the directory takes up no additional space other than in the names of the files contained within.

The directory separator could be optional too, a simple `strncmp`-based implementation would see files `foo/a`, `foo/b` and `foobar` as children of a directory named `foo`, which could help make that path stretch a little further.