pellepl/spiffs #117: Any chance this could be used with Arduino API ?

maichaell, 2016-12-02:
Hi Pelle,

This really is not an issue but more like a question..

I'm using STM32duino framework with STM32F103C8 and 32MB Windbond flash. How much effort it would be to have this library working with Arduino SPI API ?

Thanks

pellepl, 2016-12-02:
Hi,
Well - I have zero experience from the arduino, but given the big community I'd be surprised if there isn't a spi driver for the winbond (or some other compatible) somewhere on the net. All you need when it comes to spi flash functions are read, write and erase.
Regarding the effort it should be easier than to integrate it on e.g. an esp8266 or pic32, as the target platform for spiffs was the stm32 in the first place.

maichaell, 2016-12-03:
I found one sketch with these methods. I tried this and it seems to work okay. 

I also just found the 'Integrate spiffs' page and it seems I do need to implement three functions.

`
    s32_t (*spiffs_read)(u32_t addr, u32_t size, u8_t *dst)
    s32_t (*spiffs_write)(u32_t addr, u32_t size, u8_t *src)
    s32_t (*spiffs_erase)(u32_t addr, u32_t size)

Could you say are these following okay ? I'm not sure about the erase method because does your interface expect to erase one byte or sector ? 

Thanks !!

`

`uint16_t read(uint32_t addr,uint8_t *buf,uint16_t n){ 
  select();
  SPI.transfer(CMD_READ_DATA);
  SPI.transfer(addr>>16);          // A23-A16
  SPI.transfer((addr>>8) & 0xFF);  // A15-A08
  SPI.transfer(addr & 0xFF);       // A07-A00
 
  uint16_t i;
  for(i = 0; i<n; i++ ) {
    buf[i] = SPI.transfer(0x00);
  }
  
  deselect();
  return i;
}`

`uint16_t fastread(uint32_t addr,uint8_t *buf,uint16_t n) {
  select();
  SPI.transfer(CMD_FAST_READ);
  SPI.transfer(addr>>16);          // A23-A16
  SPI.transfer((addr>>8) & 0xFF);  // A15-A08
  SPI.transfer(addr & 0xFF);       // A07-A00
  SPI.transfer(0x00);              // dummy
  
  uint16_t i;
  for(i = 0; i<n; i++) {
    buf[i] = SPI.transfer(0x00);
  }
  
  deselect();  
  return i;
}
`

`//
// Erase sector units (erases the data in 4kb space units)
// sect_no (in) sector number (0 - 2048)
// flgwait(in) true:waits to be processed false:No wait
// Return value: true:Successful completion false:Failure
//  Note: in the data sheet is described as normally takes 30ms, 400ms maximum erasure
//         Significant 11 bits of the address 23-bit corresponds sector number. Lower 12 bits is the sector in the address.
//
boolean eraseSector(uint16_t sect_no, boolean flgwait) {
 uint32_t addr = sect_no;
 addr<<=12;

 WriteEnable();
 select(); 
 SPI.transfer(CMD_SECTOR_ERASE);
 SPI.transfer((addr>>16) & 0xff);
 SPI.transfer((addr>>8) & 0xff);
 SPI.transfer(addr & 0xff);
 deselect();
 
 // 処理待ち
[... 539 lines cut ...]
    Serial.print(buf);
    clm++;
    if (clm == 16) {
      sprintf(buf,"|%02x ",sum);
      Serial.print(buf);      
      Serial.println("");
      clm = 0;
    }    
  }
  Serial.println("----------------------------------------------------------");
  Serial.print("       ");
  for (int i=0; i<16;i++) {
    total+=vsum[i];
    sprintf(buf,"%02x ",vsum[i]);
    Serial.print(buf);
  }
  sprintf(buf,"|%02x ",total);
  Serial.print(buf);      
  Serial.println("");
  Serial.println("");
}

void loop() {

}`

pellepl, 2016-12-03:
By a quick glance it looks OK.  The erase hal function expects sector erase, spi flashes cannot be erased byte wise. The length will always be SPIFFS_PHYS_ERASE_SIZE. I recommend looking through the wiki thoroughly 😉

maichaell, 2016-12-03:
Thanks! Wiki is really good - I'll study that.

maichaell, 2016-12-06:
Just tried to integrate this. Run into this:

`
Compiling .pioenvs\bluepill_f103c8\src\spiffs-master\src\spiffs_cache.o
Compiling .pioenvs\bluepill_f103c8\src\spiffs-master\src\spiffs_check.o
Compiling .pioenvs\bluepill_f103c8\src\spiffs-master\src\spiffs_gc.o
spiffs-master\src\spiffs_gc.c: In function 'spiffs_gc_find_candidate':
spiffs-master\src\spiffs_gc.c:258:28: error: 'intptr_t' undeclared (first use in this function)
cand_scores = (s32_t*)(((intptr_t)cand_scores + sizeof(intptr_t) - 1) & ~(sizeof(intptr_t) - 1));
^
spiffs-master\src\spiffs_gc.c:258:28: note: each undeclared identifier is reported only once for each function it appears in
spiffs-master\src\spiffs_gc.c:258:37: error: expected ')' before 'cand_scores'
cand_scores = (s32_t*)(((intptr_t)cand_scores + sizeof(intptr_t) - 1) & ~(sizeof(intptr_t) - 1));
^
Compiling .pioenvs\bluepill_f103c8\src\spiffs-master\src\spiffs_hydrogen.o
*** [.pioenvs\bluepill_f103c8\src\spiffs-master\src\spiffs_gc.o] Error 1
spiffs-master\src\spiffs_hydrogen.c: In function 'SPIFFS_mount':
spiffs-master\src\spiffs_hydrogen.c:94:27: error: 'intptr_t' undeclared (first use in this function)
u8_t addr_lsb = ((u8_t)(intptr_t)fd_space) & (ptr_size-1);
^
spiffs-master\src\spiffs_hydrogen.c:94:27: note: each undeclared identifier is reported only once for each function it appears in
spiffs-master\src\spiffs_hydrogen.c:94:36: error: expected ')' before 'fd_space'
u8_t addr_lsb = ((u8_t)(intptr_t)fd_space) & (ptr_size-1);
^
spiffs-master\src\spiffs_hydrogen.c:103:31: error: expected ')' before 'cache'
addr_lsb = ((u8_t)(intptr_t)cache) & (ptr_size-1);
^
*** [.pioenvs\bluepill_f103c8\src\spiffs-master\src\spiffs_hydrogen.o] Error 1
 [ERROR] Took 1.17 seconds 
Process terminated with status 1 (0 minute(s), 1 second(s))
5 error(s), 0 warning(s) (0 minute(s), 1 second(s))
 
`

pellepl, 2016-12-06:
See this - http://www.cplusplus.com/reference/cstdint/. Seems you need to include stdint.h, or simply typedef intptr_t to s32_t in spiffs_config.h, given your target is using 32 bit addressing.

maichaell, 2016-12-06:
Getting there... mount return 0 but writing a file: -10010. I seem to have a problem in write function it does not write bytes but the whole buffer. The function is not my creation. I need to fix that. 

```
void winbondFlashClass::writePage(uint32_t addr_start,uint8_t *buf)
{
    char buff[64];
    sprintf(buff,">writePage: addr:%d\n",addr_start);
    Serial.println(buff);

  select();
  transfer(PAGE_PGM);
  transfer(addr_start>>16);
  transfer(addr_start>>8);
  transfer(0x00);
  uint8_t i=0;
  do {
    transfer(buf[i]);
    Serial.print((char)buf[i]);
    i++;
  }while(i!=0);
  deselect();

}
````

maichaell, 2016-12-07:
One question, does the write function send address in bytes or in page number?

``` 
my_spiffs_write(u32_t addr, u32_t size, u8_t *src) {
...
}
```

pellepl, 2016-12-07:
In bytes. The actual, physical address of  the spiflash.

maichaell, 2016-12-07:
Ok, Thanks for the info!

Sorry if this is basic info but I've never used SPI Flash chips before. So just to confirm that my write operations are correct..  

Does this 'Page Program' operation (WinBond 32Mbit) expect to have  the page number as a parameter or actual physical bytes in flash ? 

_**8.2.15 Page Program (02h)**
The Page Program instruction allows from one byte to 256 bytes (a page) of data to be programmed at
previously erased (FFh) memory locations. A Write Enable instruction must be executed before the device
will accept the Page Program Instruction (Status Register bit WEL= 1). The instruction is initiated by
driving the /CS pin low then shifting the instruction code “02h” followed by a **24-bit address (A23-A0)** and
at least one data byte, into the DI pin. The /CS pin must be held low for the entire length of the instruction
while data is being sent to the device. The Page Program instruction sequence is shown in Figure 29._

pellepl, 2016-12-07:
Heh, no problem, it can be confusing. It's address, not page index for the spiflash itself also. Do mind though, page boundaries cannot be crossed. Say your flash has 256 byte pages, and you want to write 512 bytes to address 8 and onwards. Then you have to issue a write command for each page, that is one write command for bytes 8-255, one for bytes 256-511, and one for bytes 512-520.

pellepl, 2016-12-07:
Forgot to mention that if spiffs is configured with a logical page size
being same or less as the actual page size of the spiflash, you do not have
to worry about page crossings in your driver. spiffs will handle that for
you. On the other hand it is of course nice having a spiflash driver coping
with general cases also.

maichaell, 2016-12-07:
Ok, thanks for the info!

Well I'm still having hard time to get it working. Basically I've used info from 'integrate' section in the wiki. And I'm using test file that creates file with 'hello world'. I have winbond 32Mbit chip

I've checked that write/read/erase spi functions work as supposed on my stm32.

In the program I do chip erase first. Then mounting fs (returns 0) and then running the test program. Write returns -10010.

I'm a bit out of ideas. What should I try next?

one question abut write. Even spiffs instructs to write 45 bytes should the code really write only 45 bytes? I've understood that it doesn't matter because whole page is written always in one operation and 'by design'.

pellepl, 2016-12-07:
Perhaps that is the problem - the file system very often (if not always)
writes only parts of pages. If you get a call to write 4 bytes of data to
0x123456, you must only write those four bytes.

maichaell, 2016-12-07:
Hey, got it working !! Seems to work okay! Thanks for all the help.

I got two problems:

- did not check for 'busy' status from flash before write/read/erase operations
- wrote always 256 bytes to the flash even spiffs requested less..

Here's some debug bytes for you ;-) it first mounts and then executes the test program

```spiffs_read: addr:254, size:2
spiffs_read: addr:33022, size:2
spiffs_read: addr:65790, size:2
spiffs_read: addr:98558, size:2
spiffs_read: addr:131326, size:2
spiffs_read: addr:164094, size:2
spiffs_read: addr:196862, size:2
spiffs_read: addr:229630, size:2
spiffs_read: addr:262398, size:2
spiffs_read: addr:295166, size:2
spiffs_read: addr:327934, size:2
spiffs_read: addr:360702, size:2
spiffs_read: addr:393470, size:2
spiffs_read: addr:426238, size:2
spiffs_read: addr:459006, size:2
spiffs_read: addr:491774, size:2
spiffs_read: addr:0, size:256
spiffs_read: addr:32768, size:256
spiffs_read: addr:65536, size:256
spiffs_read: addr:98304, size:256
spiffs_read: addr:131072, size:256
spiffs_read: addr:163840, size:256
spiffs_read: addr:196608, size:256
spiffs_read: addr:229376, size:256
spiffs_read: addr:262144, size:256
spiffs_read: addr:294912, size:256
spiffs_read: addr:327680, size:256
spiffs_read: addr:360448, size:256
spiffs_read: addr:393216, size:256
spiffs_read: addr:425984, size:256
spiffs_read: addr:458752, size:256
spiffs_read: addr:491520, size:256
mount res:0
TEST SPIFFS spiffs_read: addr:0, size:256
spiffs_read: addr:32768, size:256
spiffs_read: addr:65536, size:256
spiffs_read: addr:98304, size:256
spiffs_read: addr:131072, size:256
spiffs_read: addr:163840, size:256
spiffs_read: addr:196608, size:256
spiffs_read: addr:229376, size:256
spiffs_read: addr:262144, size:256
spiffs_read: addr:294912, size:256
spiffs_read: addr:327680, size:256
spiffs_read: addr:360448, size:256
spiffs_read: addr:393216, size:256
spiffs_read: addr:425984, size:256
spiffs_read: addr:458752, size:256
spiffs_read: addr:491520, size:256
spiffs_read: addr:0, size:256
spiffs_read: addr:32768, size:256
spiffs_read: addr:65536, size:256
spiffs_read: addr:98304, size:256
spiffs_read: addr:131072, size:256
spiffs_read: addr:163840, size:256
spiffs_read: addr:196608, size:256
spiffs_read: addr:229376, size:256
spiffs_read: addr:262144, size:256
spiffs_read: addr:294912, size:256
spiffs_read: addr:327680, size:256
spiffs_read: addr:360448, size:256
spiffs_read: addr:393216, size:256
spiffs_read: addr:425984, size:256
spiffs_read: addr:458752, size:256
spiffs_read: addr:491520, size:256
spiffs_read: addr:0, size:256
spiffs_write: addr:0, size:2
>writePageBytes: addr:0, bytes:2
1,80,<writepageBytes
spiffs_write: addr:256, size:45
>writePageBytes: addr:256, bytes:45
1,80,0,0,F8,0,0,0,FF,FF,FF,FF,1,6D,79,5F,66,69,6C,65,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,<writepageBytes
spiffs_read: addr:256, size:256
Write:
OK0
Close:
spiffs_write: addr:2, size:2
>writePageBytes: addr:2, bytes:2
1,0,<writepageBytes
spiffs_write: addr:512, size:5
>writePageBytes: addr:512, bytes:5
1,0,0,0,FC,<writepageBytes
spiffs_write: addr:517, size:12
>writePageBytes: addr:517, bytes:12
48,65,6C,6C,6F,20,77,6F,72,6C,64,0,<writepageBytes
spiffs_write: addr:256, size:256
>writePageBytes: addr:256, bytes:256
1,80,0,0,F8,0,0,0,C,0,0,0,1,6D,79,5F,66,69,6C,65,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,2,0,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,<writepageBytes
Open:spiffs_read: addr:512, size:256
READ OK
Hello world
DONE
```

pellepl, 2016-12-09:
Great! Good work! Do try and create and delete a lot of files also, just to make sure the erase works.

maichaell, 2016-12-09:
Good idea!

I have:
#define SPIFFS_CFG_PHYS_ERASE_SZ(ignore)  (32*1024)

and I get a call to erase 32kB
spiffs_erase: addr:327680, size:32768

.. so will spiffs always erase by the size what's defined  SPIFFS_CFG_PHYS_ERASE_SZ(ignore) and not any smaller ?

WinBond supports 4kB,32kB and 64kB erase operations. 32kB must be faster than 4kB but if I want best write performance and I'm not concerned about wasting extra bytes of space is 32/64kB better than 4kB for erase size ?

Thanks.

pellepl, 2016-12-09:
Yep, hal_erase will always be called with SPIFFS_CFG_PHYS_ERASE_SZ as size.

As you say, if you're not concerned about wasting bytes compared to
performance, bigger erase sizes are better than small.

maichaell, 2016-12-09:
Thanks! It works just fine. 

One question about directories. I understood there's no directory structure like we usually understand in Linux or Windows but I can see there are functions for directoreis. This is okay but I want to list available files somehow.

I've created files in "empty" directory. like: 

`SPIFFS_open(&fs, "my_file", SPIFFS_CREAT | SPIFFS_APPEND| SPIFFS_RDWR, 0);`

Can I list them or should I create them in '/' 

SPIFFS_open(&fs, "/my_file", SPIFFS_CREAT | SPIFFS_APPEND| SPIFFS_RDWR, 0);

?

Then I tried this: 

`SPIFFS_opendir(&fs, "", &d); `

and it seems to fault my board.

Thanks!

pellepl, 2016-12-09:
If you create a file named "/foo" you'll get a file with name "/foo".
Creating a file "foo" will be a new file named "foo", not related to "/foo"
in any way.
In SPIFFS_opendir, the directory argument is ignored, so it should work.
There is an example on the wiki, under Under spiffs.

maichaell, 2016-12-09:
Thanks, got it!

If I use RTOS.. can you say what operations are "thread-safe" ? For example can I open/read same file from two different tasks ? I found there's possibility to define a mutex through config

```
SPIFFS_LOCK(fs)
```

Should I implement these if using RTOS ? Thanks.

pellepl, 2016-12-09:
When using rtos you must define mutex in SPIFFS_LOCK and SPIFFS_UNLOCK. It
is OK to open same file in different threads as long as the spiffs stack
never calls into itself. Which the mutex will prevent.

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl