pellepl/spiffs #118: Read previous written file after re-run

mardanikamali, 2016-12-13:
Hi, 

I integrated SPIFFS with my own 16MB QSPI flash. When I run the following code, everything goes right. 

` 
>>
	char buf[15];
	spiffs_file fd = SPIFFS_open(&__fs, "my_file", SPIFFS_CREAT | SPIFFS_TRUNC | SPIFFS_RDWR, 0);
	printf("\nerrno after open = %i\n", SPIFFS_errno(&__fs));
	if (SPIFFS_write(&__fs, fd, (u8_t *)"123456789012345", 15) < 0) printf("err after write = %i\n", SPIFFS_errno(&__fs));
	printf("\nerrno after write = %i\n", SPIFFS_errno(&__fs));
	SPIFFS_close(&__fs, fd);
	printf("\nerrno after close = %i\n", SPIFFS_errno(&__fs));

	fd = SPIFFS_open(&__fs, "my_file", SPIFFS_RDONLY, 0);
	if (SPIFFS_read(&__fs, fd, (u8_t *)buf, 15) < 0) printf("err after read = %i\n", SPIFFS_errno(&__fs));
	printf("\nerrno after read = %i\n", SPIFFS_errno(&__fs));
	SPIFFS_close(&__fs, fd);
	printf("\n\n");
	printf("buf = %02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x", buf[0], buf[1], buf[2], buf[3], buf[4], buf[5], buf[6], buf[7], buf[8], buf[9], buf[10], buf[11], buf[12], buf[13], buf[14]);
	printf("\n\n");
`

I enabled SPIFFS_USE_MAGIC, and in this situation, run, mount, unmount, format, and monut are carried out well, respectively. 
Afterward, I try to rerun another test without write, and just read previous written file:

` 
>>
	char buf[15];
	
	spiffs_file fd = SPIFFS_open(&__fs, "my_file", SPIFFS_RDONLY, 0);
	if (SPIFFS_read(&__fs, fd, (u8_t *)buf, 15) < 0) printf("err after read = %i\n", SPIFFS_errno(&__fs));
	printf("\nerrno after read = %i\n", SPIFFS_errno(&__fs));
	SPIFFS_close(&__fs, fd);
	printf("\n\n");
	printf("buf = %02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x", buf[0], buf[1], buf[2], buf[3], buf[4], buf[5], buf[6], buf[7], buf[8], buf[9], buf[10], buf[11], buf[12], buf[13], buf[14]);
	printf("\n\n");
`
In this situation I got SPIFFS_ERR_INDEX_REF_LU! Also SPIFFS_USE_MAGIC is enabled in second scenario, which has a mass format! I think previous written file will be removed by SPIFFS_format. However, I disabled MAGIC in second scenario and I got SPIFFS_ERR_BAD_DESCRIPTOR. 

In fact, I don't know how I can SPIFFS_read a previous written file?

mardanikamali, 2016-12-13:
In second run, res after SPIFFS_open is 0, and after SPIFFS_read is -10016.

pellepl, 2016-12-13:
It sounds like your erase function in the HAL has some peculiarities in it. Have you verified that your erase function actually erases a block? You should get SPIFFS_ERR_NOT_FOUND. Error SPIFFS_ERR_INDEX_REF_LU indicates a broken structure, and it should not happen after a format.

pellepl, 2017-03-02:
feel free to reopen if necessary

closed 2017-03-02 by pellepl

jwgod, 2018-01-16:
@pellepl 

Hi, I have the exact same problem as @mardanikamali. I am using MSP430 with CCS V7.

In the attachment is the output from the first run (open, write, close, open read close)
[output_first_run.txt](https://github.com/pellepl/spiffs/files/1633400/output_first_run.txt)

Here is the output from the rerun (reading previously written file^)
[output_rerun.txt](https://github.com/pellepl/spiffs/files/1633398/output_rerun.txt)

and here is the output from performing mount, unmount, format and mount
[mnt_unmnt_fmt_mnt_output.txt](https://github.com/pellepl/spiffs/files/1631151/mnt_unmnt_fmt_mnt_output.txt)

Also, I have tested my erase function which does erase specified blocks. 
My erase function in the HAL is as follows,

```
void my_spi_erase(uint32_t addr, uint32_t num_byte) {
    uint8_t num_block = 0, num_single = 0;

    num_block = num_byte / WB_BLOCK_SIZE_64; // number of full blocks
    num_single = num_byte % WB_BLOCK_SIZE_64; // number of leftovers

    while (num_block--) {
        block_erase_64KB(addr);
        addr += WB_BLOCK_SIZE_64;
        num_byte -= WB_BLOCK_SIZE_64;
    }

    if (num_single > 0) {
        block_erase_64KB(addr);
    }
}
```
Since in the integration section you said "... the **address** and **size** on spiffs_erase requests will always be on physical block size boundaries ..." so I thought my erase function should be good enough or maybe I misunderstood something?

jwgod, 2018-01-16:
I had a similar issue. I use MSP430 with CCS 7.

The problem was resolved by switching on the SPIFFS_ALIGNED_OBJECT_INDEX_TABLES option. I forgot that MSP430 is not a Cortex processor

iceberg1369, 2018-05-27:
@jwgod you saved me! I almost pulled all my hairs until I see your comment